bblanchon/ArduinoJson #1938: is<TEnum>() function returns 1 even though the given Enum does not contain the supplied value

divStar, 2023-06-25:
**Describe the bug**  
Please excuse me if this is supposed to be this way.
When using `someJsonObject["someValue"].is<SomeEnum>` it returns `1` even though the actual `int` value **does not exist** in the enum `SomeEnum`.

**Troubleshooter report**  
Here is the report generated by the [ArduinoJson Troubleshooter](https://arduinojson.org/v6/troubleshooter/):  
1. The issue happens at run time
2. The issue concerns deserialization
3. `deserializeJson()` returns `Ok`
4. `JsonDocument` returns garbage
5. Input type is `char*`
6. Input doesn't come from a stream
7. Casting the pointer doesn't fix the issue

**Environment**  
Here is the environment that I used:
* Microcontroller: Arduino Zero SAMD21G18 (not the actual Arduino Zero board)
* Core/runtime: I am not sure
* IDE:
  * CLion 2023.1.3 (Build #CL-231.9011.31, built on May 15, 2023)
  * PlatformIO for CLion v231.8109.141
  * PlatformIO Core 6.1.7
  * PlatformIO Home 3.4.4

**Reproduction**  
Here is a small snippet that reproduces the issue.

```c++
enum SomeEnum {
   VALUE0 = 0,
   VALUE1 = 1,
   VALUE2 = 2
}

StaticJsonDocument<256> doc;

DeserializationError error = deserializeJson(doc, "{\"hello\":15}", 256);

bool helloIsOfSomeEnum = doc["hello"].is<SomeEnum>();
```

**Compiler output**    
Not relevant I think.

**Program output**  
Not relevant I think

Expected output:
From my example `helloIsOfSomeEnum` should return `0` since the JSON value for `hello` is `15` and thus not defined in `SomeEnum`.

```
bool helloIsOfSomeEnum = doc["hello"].is<SomeEnum>();
// helloIsOfSomeEnum should equal 0
```

Actual output:
Not relevant I think.

**Workaround**
My workaround is to check the value manually against a zero-based max-value. This of course only works for enums with consecutive indices.
I am trying to validate, that a given index corresponds to an existing LED.

bblanchon, 2023-06-26:
Hi @divStar,

As far as I know, there is no way to do this in C++.
So, you're right, it's a bug, but I cannot do anything about it.

As a workaround, you can use an [`enum class`](https://www.stroustrup.com/C++11FAQ.html#enum) with a [custom converter](https://arduinojson.org/news/2021/05/04/version-6-18-0/).

Best regards,
Benoit

divStar, 2023-06-26:
Thank you very much for your reply! Perhaps not allowing `TEnum` OR forcing a custom-converter (or giving a warning if no custom converter is used) could be a good idea? The workaround with the custom converter is great though - thank you!

closed 2023-06-26 by divStar