bblanchon/ArduinoJson #1940: How to limit float with  2 decimal point when pack into JSON?

KiraVerSace, 2023-06-28:
Hi
For example, I want to upload the temperature to the server, usually like this,

`"temperature":"24.56"`

and the code like this:

```
float temp = getTemperature();
doc["temperature"]=String(temp, 2);
```

But if I want to upload the float directly like:
`"temperature":24.56`

and the code will be like 
`doc["temperature"]=temp;`

and the true in JSON is like blow:
`"temperature":24.560012034`,

though it is right, but looks not beautiful, is there anyway to slove it?

Thank you!

bblanchon, 2023-06-29:
Hi @KiraVerSace,

Please use the [ArduinoJson Troubleshooter](https://arduinojson.org/v6/troubleshooter/) before opening an issue.
You can also find answers to most questions in the [documentation](https://arduinojson.org/v6/how-to/configure-the-serialization-of-floats/):

> If you find that there are too many digits, you can round the floating point value.
> Use the function below to have a maximum of two decimal places.
>
>     // rounds a number to 2 decimal places
>     // example: round(3.14159) -> 3.14
>     double round2(double value) {
>        return (int)(value * 100 + 0.5) / 100.0;
>     }

Alternatively, you can also use [serialized](https://arduinojson.org/v6/api/misc/serialized/)<

Best regards,
Benoit

KiraVerSace, 2023-06-30:
> 

Interesting, I do it as your suggestion before, the difference is only I use float instead of double, it do not work, but your can work well.

Thank you !

closed 2023-06-30 by bblanchon