bblanchon/ArduinoJson #1941: Measure json doc optimal size before creating it.

zekageri, 2023-06-29:
Hi! First of all, thank you for this amazing library. I love it.

I have a problem where i must measure the a doc before creating it. Here is my case:

```cpp
/*
    Use: redirect("Test","message");
*/
boolean HsH_Socket::redirect(const char* message, const char* controll){
    if( !fileSys.config.serverCommEnabled || !webSocket.isConnected() ){ return false; }
    MSG_Struct msgStruct;
    msgStruct.message = message;
    msgStruct.controll = controll;
    xSemaphoreTake(messageSemaphore, portMAX_DELAY);
    messages.push_back( msgStruct );
    xSemaphoreGive(messageSemaphore);
    return true;
}
```

### LOOP:

```cpp
void loop(){
    consumeRedirects();
}
```

### consumeRedirects()

```cpp
void HsH_Socket::consumeRedirects(){
    if( messages.empty() || !webSocket.isConnected() ){ return; }
    xSemaphoreTake(messageSemaphore, portMAX_DELAY);
    MSG_Struct msgStruct = messages.front();

    size_t msgLength = msgStruct.message.length() + 15500;
    SpiRamJsonDocument frame(msgLength);
    JsonObject header = frame.createNestedObject("header");

    header["from"]  = "hsh";
    header["to"]    = "client";
    header["key"]   = fileSys.config.hshKey;
    header["controll"]  = msgStruct.controll;
    header["stamp"]     = hsh_timeSystem.getUnixMS();

    frame["data"] = msgStruct.message;

    size_t finalLength = serializeJson(frame, messageString, msgLength);

    if (finalLength) {
        if( !webSocket.sendTXT(messageString,finalLength) ){
            Serial.printf("\n[SOCKET] - Failed to send %d bytes.\n", finalLength);
        }
    }
    messages.pop_front();
    xSemaphoreGive(messageSemaphore);
}
```

In my `consumeRedirects` function i have this line: `size_t msgLength = msgStruct.message.length() + 15500;`
The `15500` is a magic number, i don't know why that much. It depends on my message's length and if i put a small message, a small + size is enough, but as my message grows i must increase this number.

I have two problems with it.

- If i add a small number to my msgLength the messages arrive but broken. If i add too much its going to eat too much memory. 
- Second problem that this entire thing can freeze. Meaning that i can not send or receive messages anymore for some reason. Maybe it is because of the port max delay in the semaphore take? But this is unrelated to arduinoJson.

**MSG_Struct looks like this:**

```cpp
struct MSG_Struct {
    std::string message;
    std::string controll;
};
char* messageString;
std::deque<MSG_Struct> messages;
```

bblanchon, 2023-06-30:
Hi @zekageri,

Please read [How to determine the capacity of the JsonDocument?](https://arduinojson.org/v6/how-to/determine-the-capacity-of-the-jsondocument/)

Best regards,
Benoit

zekageri, 2023-06-30:
Thank you for your response.
This is a really good article.

Unfortunatelly my data can be any size and any shape.
I'm sending messages from different components trought websockets to a node.js server.
Data can be a file's content ( with several kb ) or a simple two-three word message.
And if i decide to add a new component with completely different data sets i don't want to modify this json doc size again.

My struct contains all of my data in the `std::string message` variable as a serialized json. But it is not sufficent to use `message.length()` and i just want to understand why `ArduinoJson` needs more than `message.length() + controll.length()`.

If message contains a simple data like "Hello World" it is okay for the doc to be message.length() + 500 but if it contains a file's content i must add a huge number like 15000. I don't understand that part.

zekageri, 2023-06-30:
The frame that i'm building in this function looks like this

```js
frame = {
    header: {
        from: "hsh",
        to: "client",
        key: "hshKey",
        controll: "setHeader" || "message" ( message is the default )
    },
    data: message ( serialized object or array or any serialized json )
}
```

Where data can be any size ( like a file content or a simple hello world )

bblanchon, 2023-07-03:
> Where data can be any size 

If you think about it, data cannot be of *any* size; it can be between 0 and whatever your microcontroller will be able to handle.
Therefore, this size is limited by the amount of RAM that is available for this task.
You can do a back-of-the-envelope calculation to estimate how much RAM should be available and use it as the capacity of the `JsonDocument`.
This is [Technique 3: set a memory budget](https://arduinojson.org/v6/how-to/determine-the-capacity-of-the-jsondocument/#technique-3-set-a-memory-budget).

zekageri, 2023-07-05:
Thank you very much. I appreciate your help!
It seesms that i have no other option but to use Technique 3.

bblanchon, 2023-07-06:
In ArduinoJson 7, `JsonDocument` will have an elastic capacity, so you won't have to think about this problem anymore.
I'm actively working on it, but there is still a long way to go.