bblanchon/ArduinoJson #1944: Deserializing unsigned int fails

Ry-Kode, 2023-07-07:
Hi,

I'm having trouble reading an unsigned int value when deserializing items in an array.

The troubleshooter report outcome was : "talk to a human" :-)

Environment: Raspberry Pi OS, c++17

Json:  
  {"frames":[
    {"id":1,"frame-name":"Frame_1",
    "items":[{"id":2,"name":"Obj1", "color":4290953763}]}
                 ]}

Code (simplified):
  DeserializationError lError = deserializeJson(lJsonDoc, lShowFile);
  JsonArray arr = lJsonDoc["frames"];

  for (JsonVariant lJsonFrame: arr) {

    JsonArray JsonItemsArr = lJsonFrame["items"];

      for (JsonObject KV: JsonItemsArr) {
        lItemIdx = -1;
        int lItemID = KV["id"];   //-> OK, returns 2
        unsigned int lItemColor  = KV["color"] | 0;    // --> NOT OK, returns 0 
      }
  }

Expected output: 4290953763
Actual output: 0

What am I missing here?

PS. Thanks for your great library!! 
I have been using it for over 3 years now and it has proven to be 100% stable.

bblanchon, 2023-07-08:
Hi @Ry-Kode,

[JsonVariant::operator|()](https://arduinojson.org/v6/api/jsonvariant/or/) aligns its return type to the default value type.
In your case, the default value is `0`, which is an `int`.
Since `4290953763` is too large to fit on an `int`, [JsonVariant::operator|()](https://arduinojson.org/v6/api/jsonvariant/or/) returns the default value.

To fix this program, you need to change the default value type to `unsigned long long`.
For example:

```c++
auto lItemColor  = KV["color"] | 0ULL;
```

Notice the `ULL` suffix that tells the compiler that this is an `unsigned long long` literal.

See an online demo: https://wandbox.org/permlink/HEXxFtqDiVTF2cw9

Please share the Troubleshooter report so I know where to insert a new page.

Best regards,
Benoit

Ry-Kode, 2023-07-09:
Works like a charm. Many thanks Benoît!

PS. Because the troubleshooter report gave no result, I did not copy or save it.

closed 2023-07-10 by bblanchon