bblanchon/ArduinoJson #1946: std::string_view - pro's and con's

vortigont, 2023-07-11:
with c++17 compatible compilers availability using `std::string_view` could become handy. But it's use cases are almost missing in documentation, so just want to rise some questions to get more details.

1) is there any difference in handling literals with `const char*` and `std::string_view` with ArduinoJson?
2) Does using `std::string_view` as. i.e. object keys implies any zero-copy or thin storage of any kind?

for example

```
std::string_view key1 = "my_str_key1";
std::string_view val1 = "foo_bar";

StaticJsonDocument<256> doc;
JsonObject obj = doc.to<JsonObject>();

obj[key1] = 123;    // will this copy "my_str_key1" into doc's mem pool?

String str("another_key");
obj[std::string_view(str.c_str())] = std::string_view("some_value");      // it might look ugly, but just interested in what happens here with string literals and doc memory

```

bblanchon, 2023-07-12:
Hi @vortigont,

If you insert a `std::string_view` in a `JsonDocument` (either as a key or a value), ArduinoJson will make a copy.
For example, in the following snippet, statement `B` consumes 12 more bytes than statement `A`:

```c++
doc["hello"] = "world";  // A
doc[std::string_view("hello")] = std::string_view("world");  // B
```

There is no duplication when extracting a `std::string_view` from a `JsonDocument`, so it's better than using a `std::string`.
The only benefit compared to `const char*` is that a `std::string_view` can contain NUL characters.

Best regards,
Benoit

vortigont, 2023-07-13:
now I see that makes it clear. Thanks @bblanchon for your clarification!

closed 2023-07-13 by vortigont