bblanchon/ArduinoJson #1949: Serialization trunking the data

ronit-str8bat, 2023-07-21:
Hi,

I want to serialize the data (6850 bytes) stored in the structure. But when I'm calling the serializeJson function, it's only emitting the 1247 bytes. is there a limit on size from ArduinoJson or esp32. 

```
 void socketIO_connect_ackshot(Device* device) 
{
  DynamicJsonDocument SI_connect_ack(20000);
  JsonArray json_array               = SI_connect_ack.to<JsonArray>();
  JsonObject sIO_shotdata_obj = json_array.createNestedObject();
  JsonArray ack_key                  = sIO_shotdata_obj.createNestedArray("ack_key");  

  for (int i = 0; i < 6800; i++) {
    ack_key.add(device->acc[i]);
  }
  
  for (int i = 0; i < 6; i++) {
    ack_key.add(device->mac_id[i]);
  }
  for (int i = 0; i < 24; i++) {
    ack_key.add(device->sx[i]);
  }
  for (int i = 0; i < 15; i++) {
    ack_key.add(device->lx[i]);
  }

  ack_key.add(device->receive_data);
  ack_key.add(device->size);
  ack_key.add(device->upload);
  ack_key.add(device->lx);
  ack_key.add(device->sx);

  String json_ackdata;
  Serial.print("measure json:");
  Serial.println(measureJson(SI_connect_ack));

  serializeJson(SI_connect_ack, json_ackdata);

  Serial.print("json data:");
  Serial.println(json_ackdata);

  SI_connect_ack.clear();
}

```

Printing the **json_ackdata**, giving trunked data of around 600 bytes. 
And if I increase the size of 
>   DynamicJsonDocument SI_connect_ack(10000);

to 20000, it string **json_ackdata** gets 1247 bytes but making the size 100,000 giving the empty string 

Can someone suggest what can be the reason?

For reference :-
I'm using ESP32-Wroom-32, 
Arduino JSON version - 6.21.2,
compiling the code giving the message 
```
Sketch uses 905357 bytes (69%) of program storage space. Maximum is 1310720 bytes.
Global variables use 60736 bytes (18%) of dynamic memory, leaving 266944 bytes for local variables. Maximum is 327680 bytes.

```

bblanchon, 2023-07-22:
Hi @ronit-str8bat,

What does the [ArduinoJson Troubleshooter](https://arduinojson.org/v6/troubleshooter/) say?

Best regards,
Benoit

ronit-str8bat, 2023-07-24:
Hi @bblanchon ,
After answering ArduinoJson Troubleshooter, it went out of ideas. 

I increased the document size to the 110550 and in for this space I'm able to send the 6891 bytes out of 6900 and doc.overflowed() returning 0 means I have the sufficient space..

But as soon as I increased the doc size to 110570 and again it's sending the same data 6891 bytes and doc.overflowed() returning 0 but esp32 restarts
Below is the message 
`
13:12:36.088 -> ELF file SHA256: ff4d61e1f092c7fb
13:12:36.213 -> Rebooting...
13:12:36.213 -> ets Jun  8 2016 00:22:57
13:12:36.213 -> rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
13:12:36.213 -> configsip: 0, SPIWP:0xee
13:12:36.213 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
13:12:36.260 -> mode:DIO, clock div:1
13:12:36.260 -> load:0x3fff0030,len:1344
13:12:36.260 -> load:0x40078000,len:13924
13:12:36.260 -> ho 0 tail 12 room 4
13:12:36.260 -> load:0x40080400,len:3600
13:12:36.260 -> entry 0x400805f0
`

But when I increase the size to 110600, I get empty string as output and doc.overflowed() returns 1 and esp32 didn't restart.

So I think there are 2 problem, why increasing the memory size restarting the device and increasing it further not restarting the device and giving the empty string?

bblanchon, 2023-07-25:
One thing we can tell is that your RAM is either full or highly fragmented; maybe you should address this first.

I don't think ArduinoJson caused the crash; it is likely due to a failed allocation elsewhere.
Please add calls to [`TRACE()`](https://github.com/bblanchon/ArduinoTrace) to find the line that causes the crash.

If you still think the problem comes from ArduinoJson (hint: it doesn't), please provide an [MCVE](https://stackoverflow.com/help/mcve).

ronit-str8bat, 2023-07-25:
Sure, allow me some time I will add trace and get back to you. 
Thanks a lot for making ArduinoTrace library and uploading a video on youtube regarding how to use it, it has helped a lot in debugging.

closed 2023-10-09 by bblanchon