bblanchon/ArduinoJson #1950: JsonPair key value empty

softwarecrash, 2023-07-21:
Hello,
i need a little hint or help,
i try to iterate over a nested json obect, so that working, but when i set a key to a number int or float, the output is empty

init the Json:
```
DynamicJsonDocument Json(1024);
JsonObject liveData= Json.createNestedObject("DeviceData");
```

set the Values:
```
  liveData["notWorking"] = 12345;
  liveData["Working"] = "12345";
```

iterate:
```
for (JsonPair kv : liveData) {
    Serial.print(kv.key().c_str());
    Serial.print(": ");
    Serial.println(kv.value().as<const char*>());
}
```

and the serial output:
```
11:48:43.052 > notWorking:
11:48:43.052 > Working: 12345
```

so i cant figure out my issue,
thanks for your time

bblanchon, 2023-07-22:
Hi @softwarecrash,

The second value, `"12345"`, is a `const char*`, so ArduinoJson has no difficulty returning a `const char*`.
The first value, `1234`, is an `int`, so ArduinoJson cannot return a `const char*`.

However, ArduinoJson can convert `int` to strings objects, so you could fix your program by replacing `as<const char*>()` with `as<String>()`.

This being said, if all you need is printing the value to the serial port, you'd better call `serializeJson(kv.value(), Serial)` which doesn't require the temporary `String`.

Best regards,
Benoit

softwarecrash, 2023-07-22:
@bblanchon 
thank you so much for pointing out that,
my fault to think it can convert it directly to const char* as output. the code was an example, in use i have tried to send every key with pubsubclient, so String is not the best option, but it works now.

```
for (JsonPair kv : liveData) {
    DEBUG_PRINT(kv.key().c_str());
    DEBUG_PRINT(": ");
    DEBUG_PRINTLN(kv.value().as<String>());
    mqttclient.publish(topicBuilder(buff, kv.key().c_str()), kv.value().as<String>().c_str());
}
```
Thank you for your time

bblanchon, 2023-07-22:
You're welcome, @softwarecrash!
Don't forget to [cast a star](https://github.com/bblanchon/ArduinoJson/stargazers) to support the project :wink:
You may be interested in reading [How to use ArduinoJson with PubSubClient?](https://arduinojson.org/v6/how-to/use-arduinojson-with-pubsubclient/)

softwarecrash, 2023-07-22:
@bblanchon i dont have a credit card, so i used another way :)

sorry but i need another hint, 
i have a buch of nested objects

```
DynamicJsonDocument Json(1024);
JsonObject staticData = Json.createNestedObject("DeviceData"); // battery package data
JsonObject liveData = Json.createNestedObject("LiveData");     // battery package data
```
filled with
```
DeviceData["abc"] = 123;
liveData["abc"] = 123;
```

i have tried it with this example to iterate as structure for mqtt
https://github.com/bblanchon/ArduinoJson/issues/1655

```
for(JsonPair i : Json.as<JsonObject>()) {
  for(JsonPair k : i.value().as<JsonObject>()) {
    mqttclient.publish(i.key().c_str() +"/"+ k.key().c_str()), k.value().as<String>().c_str());
  }
}
```
the topic with +"/"+ dont work in this way, its only for example

in test it compile fine, but run in a runtime error, i think its a verry simple fault by me that i cant see

bblanchon, 2023-07-23:
```c++
StaticJsonDocument<512> root;
char msgBuffer1[256];
char msgBuffer1[256];
```

Maybe that's just a stack overflow...
Can you switch to a `DynamicJsonDocument`?

softwarecrash, 2023-07-23:
> ```c++
> StaticJsonDocument<512> root;
> char msgBuffer1[256];
> char msgBuffer1[256];
> ```
> 
> Maybe that's just a stack overflow... Can you switch to a `DynamicJsonDocument`?

yes, same behavior with `DynamicJsonDocument root(1024);`

bblanchon, 2023-07-24:
Recreated your program on wandbox.org, and it works as expected.
See: https://wandbox.org/permlink/bNL2gDtQfQW8AvzD

Full program:

```c++
#include <iostream>
#include <string>
#include "ArduinoJson.h"

int main() {
  DynamicJsonDocument root(1024);
  JsonObject nest1 = root.createNestedObject("nest1");
  JsonObject nest2 = root.createNestedObject("nest2");

  nest1["nest1_1"] = 1;
  nest1["nest1_2"] = 2;
  nest1["nest1_3"] = 3;

  nest2["nest2_1"] = 1;
  nest2["nest2_2"] = 2;
  nest2["nest2_3"] = 3;

  for (JsonPair i : root.as<JsonObject>()) {
    std::cout << i.key().c_str() << "\n";
    for (JsonPair k : i.value().as<JsonObject>()) {
      std::cout << k.key().c_str() << ": " << k.value().as<std::string>() << "\n";
    }
  }
}
```

Output:

```
nest1
nest1_1: 1
nest1_2: 2
nest1_3: 3
nest2
nest2_1: 1
nest2_2: 2
nest2_3: 3
```

I bet the problem lies between the blocks you shared 😉

softwarecrash, 2023-07-24:
Thanks for your time again.
i think i must rebuild the whole code around.

closed 2023-07-24 by softwarecrash

bblanchon, 2023-07-25:
You're welcome, @softwarecrash.
Remember that [`JsonObject`](https://arduinojson.org/v6/api/jsonobject/) refers to an object stored in the [`JsonDocument`](https://arduinojson.org/v6/api/jsondocument/), so the `JsonDocument` must still be alive when you enumerate the keys in the `JsonObject`.
Also, some operations, like [`shrinkToFit()`](https://arduinojson.org/v6/api/basicjsondocument/shrinktofit/) or [`garbageCollect()`](https://arduinojson.org/v6/api/jsondocument/garbagecollect/), invalidate all the existing references.