bblanchon/ArduinoJson #1952: error: 'class ArduinoJson::V6212PB::JsonObject' has no member named 'read'

mounilsankar, 2023-07-23:
**Description**
Im trying to compile my code which fetches a json string via a http client and parses the json string into C++ code and I am getting compilation error

**Troubleshooter's report**
1. The issue happens at compile time
2. The error is not in the list

**Environment**
* Microcontroller: ESP8266
* IDE: Arduino IDE 2.1.1

**Reproduction code**
```c++
#include <ArduinoJson.h>
#include <ESP8266WiFi.h>
#include <ESP8266WebServer.h>

const char* ssid = "newveena";
const char* password = "98246192";

ESP8266WebServer server(80);

void parseJson(JsonObject o) {
    const int id = o["id"];
    Serial.print(id);
    switch (id) {
        // id 1 corresponds to the on Start block
        case 1:
            {
                for (JsonObject child : o["contains"].as<JsonArray>()) {
                    parseJson(child);
                }
                break;
            }
        // id 2 corresponds to a forever block
        case 2:
            {
                for (JsonObject child : o["contains"].as<JsonArray>()) {
                    parseJson(child);
                }
                break;
            }
        // id 5 corresponds to a set pwm pin block
        case 5:
            {
                JsonObject arguments = o["args"][0];
                JsonObject pinBlock = arguments["pin_block"];
                JsonObject valueBlock = arguments["value_block"];
                int pin, value;
                if (pinBlock.isNull()) {
                    pin = arguments["pin"];
                } else {
                    pin = evaluateBlock(pinBlock);
                }
                if (valueBlock.isNull()) {
                    value = arguments["value"];
                } else {
                    value = evaluateBlock(valueBlock);
                }
                analogWrite(pin, value);
                pinMode(pin, OUTPUT);
                break;
            }
        // id 6 corresponds to set pin block
        case 6:
            {
                JsonObject arguments = o["args"][0];
                JsonObject pinBlock = arguments["pin_block"];
                JsonObject valueBlock = arguments["value_block"];
                int pin, value;
                if (pinBlock.isNull()) {
                    pin = arguments["pin"];
                } else {
                    pin = evaluateBlock(pinBlock);
                }
                if (valueBlock.isNull()) {
                    value = arguments["value"];
                } else {
                    value = evaluateBlock(valueBlock);
                }
                digitalWrite(pin, value);
                pinMode(pin, OUTPUT);
                break;
            }
    }
}

int evaluateBlock(JsonObject pinBlock) {
    DynamicJsonDocument doc(128);
    deserializeJson(doc, pinBlock);
    const int id = doc["id"];
    switch (id) {
        // id 3 corresponds to a state of pin block
        case 3:
            {
                JsonObject childBlock = doc["child_block"];
                int pin;
                if (childBlock.isNull()) {
                    pin = doc["pin"];
                } else {
                    pin = evaluateBlock(childBlock);
                }
                return digitalRead(pin);
            }
        // id 4 corresponds to a read analogue pin block
        case 4:
            {
                JsonObject childBlock = doc["child_block"];
                int pin;
                if (childBlock.isNull()) {
                    pin = doc["pin"];
                } else {
                    pin = evaluateBlock(childBlock);
                }
                return analogRead(pin);
            }
        default:
            return 0;
    }
}

void handleJSONData() {
  if (server.method() == HTTP_POST) {
    String json = server.arg("plain"); // Get the JSON data from the request body
    // Parse and process the received JSON data here
    Serial.println(json);
    StaticJsonDocument<384> doc;
    DeserializationError error = deserializeJson(doc, json);
    if (error) { 
      Serial.print(F("deserializeJson() failed: "));
      server.send(400, "text/plain", "deserializeJson() failed: ");
      Serial.println(error.f_str());
    return;
    }
    server.send(200, "application/json", "{\"status\":\"ok\"}");
    parseJson(doc.as<JsonObject>());
  } else {
    server.send(400, "text/plain", "Invalid request");
  }
}

void setup() {
  // put your setup code here, to run once:
  Serial.begin(9600);
  WiFi.begin(ssid, password);
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  Serial.println("WiFi connected");
  IPAddress ip = WiFi.localIP();
  Serial.print("IP Address: ");
  Serial.println(ip);

  server.on("/jsondata", handleJSONData);
  server.begin();
  Serial.println("HTTP server started");

}
void loop() {

 server.handleClient();
}
```

mounilsankar, 2023-07-23:
The error message is as follows 

In file included from /Users/mounilsankar/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson/Deserialization/deserialize.hpp:9,
                 from /Users/mounilsankar/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson/Json/JsonDeserializer.hpp:7,
                 from /Users/mounilsankar/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson.hpp:46,
                 from /Users/mounilsankar/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson.h:9,
                 from /Users/mounilsankar/Documents/Arduino/ConvertBlock/ConvertBlock.ino:1:
/Users/mounilsankar/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson/Deserialization/Reader.hpp: In instantiation of 'int ArduinoJson::V6212PB::detail::Reader<TSource, Enable>::read() [with TSource = ArduinoJson::V6212PB::JsonObject; Enable = void]':
/Users/mounilsankar/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson/Json/Latch.hpp:38:25:   required from 'void ArduinoJson::V6212PB::detail::Latch<TReader>::load() [with TReader = ArduinoJson::V6212PB::detail::Reader<ArduinoJson::V6212PB::JsonObject, void>]'
/Users/mounilsankar/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson/Json/Latch.hpp:30:7:   required from 'char ArduinoJson::V6212PB::detail::Latch<TReader>::current() [with TReader = ArduinoJson::V6212PB::detail::Reader<ArduinoJson::V6212PB::JsonObject, void>]'
/Users/mounilsankar/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson/Json/JsonDeserializer.hpp:48:26:   required from 'char ArduinoJson::V6212PB::detail::JsonDeserializer<TReader, TStringStorage>::current() [with TReader = ArduinoJson::V6212PB::detail::Reader<ArduinoJson::V6212PB::JsonObject, void>; TStringStorage = ArduinoJson::V6212PB::detail::StringCopier]'
/Users/mounilsankar/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson/Json/JsonDeserializer.hpp:72:13:   required from 'ArduinoJson::V6212PB::DeserializationError::Code ArduinoJson::V6212PB::detail::JsonDeserializer<TReader, TStringStorage>::parseVariant(ArduinoJson::V6212PB::detail::VariantData&, TFilter, ArduinoJson::V6212PB::DeserializationOption::NestingLimit) [with TFilter = ArduinoJson::V6212PB::detail::AllowAllFilter; TReader = ArduinoJson::V6212PB::detail::Reader<ArduinoJson::V6212PB::JsonObject, void>; TStringStorage = ArduinoJson::V6212PB::detail::StringCopier]'
/Users/mounilsankar/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson/Json/JsonDeserializer.hpp:36:23:   required from 'ArduinoJson::V6212PB::DeserializationError ArduinoJson::V6212PB::detail::JsonDeserializer<TReader, TStringStorage>::parse(ArduinoJson::V6212PB::detail::VariantData&, TFilter, ArduinoJson::V6212PB::DeserializationOption::NestingLimit) [with TFilter = ArduinoJson::V6212PB::detail::AllowAllFilter; TReader = ArduinoJson::V6212PB::detail::Reader<ArduinoJson::V6212PB::JsonObject, void>; TStringStorage = ArduinoJson::V6212PB::detail::StringCopier]'
/Users/mounilsankar/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson/Deserialization/deserialize.hpp:48:13:   required from 'ArduinoJson::V6212PB::DeserializationError ArduinoJson::V6212PB::detail::deserialize(ArduinoJson::V6212PB::JsonDocument&, TStream&&, Args ...) [with TDeserializer = ArduinoJson::V6212PB::detail::JsonDeserializer; TStream = ArduinoJson::V6212PB::JsonObject&; Args = {}; <template-parameter-1-4> = void]'
/Users/mounilsankar/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson/Json/JsonDeserializer.hpp:680:39:   required from 'ArduinoJson::V6212PB::DeserializationError ArduinoJson::V6212PB::deserializeJson(ArduinoJson::V6212PB::JsonDocument&, Args&& ...) [with Args = {ArduinoJson::V6212PB::JsonObject&}]'
/Users/mounilsankar/Documents/Arduino/ConvertBlock/ConvertBlock.ino:77:34:   required from here
/Users/mounilsankar/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson/Deserialization/Reader.hpp:21:21: error: 'class ArduinoJson::V6212PB::JsonObject' has no member named 'read'
   21 |     return source_->read();  // Error here? You passed an unsupported input type
      |            ~~~~~~~~~^~~~

exit status 1

Compilation error: exit status 1

bblanchon, 2023-07-23:
Hi @mounilsankar,

You cannot pass a [`JsonObject`](https://arduinojson.org/v6/api/jsonobject/) as the second argument to [`deserializeJson()`](https://arduinojson.org/v6/api/json/deserializejson/).
Indeed [`deserializeJson()`](https://arduinojson.org/v6/api/json/deserializejson/) expects a string or a stream, so it doesn't make sense to pass a [`JsonObject`](https://arduinojson.org/v6/api/jsonobject/).

See [‘class Xxx’ has no member named ‘read’](https://arduinojson.org/v6/error/no-member-named-read/)

Best regards,
Benoit

mounilsankar, 2023-07-23:
Thank you so much! Was stuck on this for hours!

bblanchon, 2023-07-24:
You're welcome!
Thank you for using ArduinoJson.

closed 2023-07-24 by bblanchon