bblanchon/ArduinoJson #1953: Arrays aren't output (6.20.x and 6.21.3)

dannybackx, 2023-07-28:
**Describe the bug**  
With source code that resembles the serialization examples, array output seems to be omitted. Observed both with 6.21.3 and (for comparison 6.20.x).

Note full source at https://sourceforge.net/p/esp32s3-4827s043/code/HEAD/tree/trunk/https/components/acmeclient/Acme.cpp

**Troubleshooter report**  
Here is the report generated by the [ArduinoJson Troubleshooter](https://arduinojson.org/v6/troubleshooter/):  
1. The issue happens at run time
2. The issue concerns serialization
3. Output is incomplete
4. Program doesn't call `serializeJson()` in a loop
5. `JsonDocument::overflowed()` returns `false`
6. No string contains a NUL

**Environment**  
Here is the environment that I used:
* Microcontroller: esp32s3
* Core/runtime: esp-idf-v5.1
* IDE: esp-idf-v5.1

**Reproduction**  
Here is a small snippet that reproduces the issue.

```c++
void Acme::WriteOrderInfo() {
  if (order == NULL) {
    ESP_LOGE(acme_tag, "%s: NULL order", __FUNCTION__);
    return;
  }

  char *fn = (char *)malloc(strlen(order_fn) + 5 + strlen(filename_prefix));
  sprintf(fn, "%s/%s", filename_prefix, order_fn);
  CreateDirectories(fn);
  FILE *f = fopen(fn, "w");
  if (f == NULL) {
    ESP_LOGE(acme_tag, "Couldn't write order info into %s, %s", fn, strerror(errno));
    free(fn);
    return;
  }

  ESP_LOGI(acme_tag, "Writing order info into %s", fn);
  free(fn);

  DynamicJsonDocument jo(10240);
  // DynamicJsonDocument jo(1024);
  if (order->status) jo[acme_json_status] = order->status;
  if (order->status) jo[acme_json_expires] = order->expires;
  if (order->finalize) jo[acme_json_finalize] = order->finalize;
  if (order->certificate) jo[acme_json_certificate] = order->certificate;

  if (order->identifiers) {
    // identifiers array must be NULL terminated
    DynamicJsonDocument jia = jo.createNestedArray(acme_json_identifiers);
    for (int i=0; order->identifiers[i]._type != 0 || order->identifiers[i].value != 0; i++) {
      DynamicJsonDocument jie = jia.createNestedObject();
      jie[acme_json_type] = order->identifiers[i]._type;
      jie[acme_json_value] = order->identifiers[i].value;
      jie["index"] = i;

      ESP_LOGI(acme_tag, "%s: identifiers[%d] = %s %s", __FUNCTION__, i,
        order->identifiers[i]._type, order->identifiers[i].value);
    }
  }

  if (order->authorizations) {
    // authorizations array must be NULL terminated
    DynamicJsonDocument jaa = jo.createNestedArray(acme_json_authorizations);
    for (int i=0; order->authorizations[i]; i++) {
      jaa.add(order->authorizations[i]);
      ESP_LOGI(acme_tag, "%s: authorizations[%d] = %s", __FUNCTION__, i, order->authorizations[i]);
    }
  }

  size_t sz = measureJson(jo);
  char *output = (char *)malloc(sz + 1);
  size_t nb = serializeJson(jo, output, sz+1);
  fprintf(f, "%s", output);
  fclose(f);

  ESP_LOGE(acme_tag, "%s: arduinojson overflow ? %d", __FUNCTION__, jo.overflowed());
  ESP_LOGI(acme_tag, "Wrote %d bytes of JSON order info", nb);
  ESP_LOGI(acme_tag, "Order info : %s", output);
  free(output);
}

```

**Compiler output**   
**Program output**  
If relevant, include the repro program output.

Expected output:
Expected is for the output not to have empty arrays for identifiers and authorizations.
You can see the expected output in the debugging output - there should be one array entry in each, something like 
.
```
"identifiers" : [ { "dns" , " s3panel2.dannybackx.dns-cloud.net"} ],
"authorizations : [ { "https://acme-staging-v02.api.letsencrypt.org/acme/authz-v3/7488615384" } ],
```

Actual output:

```
I (16:28:19.756) Acme: PerformWebQuery -> {
  "status": "ready",
  "expires": "2023-08-04T14:28:19Z",
  "identifiers": [
    {
      "type": "dns",
      "value": "s3panel2.dannybackx.dns-cloud.net"
    }
  ],
  "authorizations": [
    "https://acme-staging-v02.api.letsencrypt.org/acme/authz-v3/7488615384"
  ],
  "finalize": "https://acme-staging-v02.api.letsencrypt.org/acme/finalize/112659444/9995645104"
}
E (16:28:19.784) Acme: RequestNewOrder: reply_status ready
I (16:28:19.790) Acme: ReadOrder : read status as ready
I (16:28:19.796) Acme: ReadOrder : read expires as 2023-08-04T14:28:19Z
I (16:28:19.803) Acme: ReadOrder : read finalize as https://acme-staging-v02.api.letsencrypt.org/acme/finalize/112659444/9995645104
I (16:28:19.816) Acme: ReadOrder : 1 identifiers
I (16:28:19.821) Acme: Identifiers 0 - dns s3panel2.dannybackx.dns-cloud.net
I (16:28:19.828) Acme: ReadOrder : 1 authorizations
I (16:28:19.834) Acme: Auth 0 - https://acme-staging-v02.api.letsencrypt.org/acme/authz-v3/7488615384
I (16:28:19.859) Acme: Writing order info into /fs/acme/staging/order.json
I (16:28:19.860) Acme: WriteOrderInfo: identifiers[0] = dns s3panel2.dannybackx.dns-cloud.net
I (16:28:19.866) Acme: WriteOrderInfo: authorizations[0] = https://acme-staging-v02.api.letsencrypt.org/acme/authz-v3/7488615384
E (16:28:19.888) Acme: WriteOrderInfo: arduinojson overflow ? 0
I (16:28:19.888) Acme: Wrote 181 bytes of JSON order info
I (16:28:19.890) Acme: Order info : {"status":"ready","expires":"2023-08-04T14:28:19Z","finalize":"https://acme-staging-v02.api.letsencrypt.org/acme/finalize/112659444/9995645104","identifiers":[],"authorizations":[]}
```

bblanchon, 2023-07-29:
Hi @dannybackx,

The following line creates an empty array in the original document and then makes a copy of this empty array in a new document:

```c++
DynamicJsonDocument jia = jo.createNestedArray(acme_json_identifiers);
```

So, when you insert values in the array, you insert them in this second document (`jia`) and not in the original one (`jo`).

Instead, you must use a `JsonArray`, which points to the array without making a copy:

```c++
JsonArray jia = jo.createNestedArray(acme_json_identifiers);
```

You could also use `auto`, of course.

Best regards,
Benoit

dannybackx, 2023-07-29:
Thanks, looks like I made a couple of mistakes.

closed 2023-07-29 by dannybackx