bblanchon/ArduinoJson #1954: Is it possible to have pointer to JsonDocument as struct member?

hansmbakker, 2023-07-31:
[Disclaimer: I am not an expert on C++ and my question might be caused by missing knowledge]

## Describe the issue
I am trying to create some reusable functions with HTTPClient and ArduinoJson for ESP32. Basically my goal is to wrap the logic to
- make HTTP requests
- deserialize their responses
- return
  - the status code
  - json document
  - some result enum

An example of such a function looks like this:
```c++
template <int jsonDocumentSize>
JSONResult<jsonDocumentSize> RestClient::makeGetRequest(const std::string &uri)
```

The issue I have is this: given that I want to return multiple things from these functions, I thought to return a struct with members. Initially that worked using a template with a StaticJsonDocument (see the code below).

However, since there is a [different type recommended for large and small JSON documents](https://arduinojson.org/v6/api/jsondocument/#staticjsondocument-vs-dynamicjsondocument) I thought to try adding their common base class as a pointer instead.

While rewriting the code I had, I tried to do this:
```c++
auto doc = *result.jsonDoc;
```
to try using the JsonDocument from the pointer member as a normal JsonDocument variable.

This resulted in the error
> use of deleted function 'ArduinoJson::V6213PB2::JsonDocument::JsonDocument(const ArduinoJson::V6213PB2::JsonDocument&)'

I cannot find any example using JsonDocument as a class or struct member, and I'm wondering whether it is possible at all to do this? (Also given that e.g. DynamicJsonDocument uses its own allocator - is it meant not to be used otherwise?)

I did find https://arduinojson.org/v6/how-to/reuse-a-json-document/ noting that it is best to get the values from a JsonDocument as quickly as possible and then drop the JsonDocument so as to not let it leak, but if possible I would still be interested to understand whether this is possible. 

## Troubleshooter report
Not applicable

## Environment
Here is the environment that I'm using':
* Microconroller: ESP32
* Core/runtime: [e.g. ESP8266 core for Arduino v3.0.2]
* IDE: [e.g. Arduino IDE 1.8.16]

## Sample code
```c++
#ifndef RESULTS_H
#define RESULTS_H

#include <ArduinoJson.h>

enum ResultType
{
    Success,
    NoAccessToken,
    ConnectionError,
    HTTPError
};

struct HTTPResult
{
    ResultType resultType;
    int httpStatusCode;
};

template <int jsonDocumentSize>
struct JSONResult : HTTPResult
{
    DeserializationError deserializeError;
    // This worked:
    // StaticJsonDocument<jsonDocumentSize> jsonDoc;

    // This gives me trouble
    JsonDocument *jsonDoc;
};

#endif // RESULTS_H
```

bblanchon, 2023-08-01:
Hi @hansmbakker,

In the case studies of [Mastering ArduinoJson](https://arduinojson.org/book/), I use this technique to implement a reusable JSON-RPC client similar to your REST client.

In my case, I used a `StaticJsonDocument<256>` member variable because I knew that all my messages would fit, and I could not afford much more on an AVR microcontroller.
In your case, however, I think you should simply use a `DynamicJsonDocument` because an ESP32 has much more RAM:

```c++
struct JSONResult : HTTPResult
{
   JSONResult(int jsonDocumentSize)  : jsonDoc(jsonDocumentSize) {]

   DynamicJsonDocument jsonDoc;
};
```

Notice that I had to add a constructor to pass the capacity.

> However, since there is a [different type recommended for large and small JSON documents](https://arduinojson.org/v6/api/jsondocument/#staticjsondocument-vs-dynamicjsondocument) I thought to try adding their common base class as a pointer instead.

You're right, but if you try to create a class that does both, you would probably lose the small benefit of `StaticJsonDocument`, so you'd better use a `DynamicJsonDocument` anyway.

By the way (and I'm telling you this in total confidence 🤫), there will be neither `StaticJsonDocument` nor `DynamicJsonDocument` in [ArduinoJson 7](https://github.com/bblanchon/ArduinoJson/tree/7.x); only a `JsonDocument` with an elastic capacity.
So, don't worry too much about this issue because it'll be gone as soon.

Best regards,
Benoit

hansmbakker, 2023-08-01:
@bblanchon thank you! This is a super helpful answer!

closed 2023-08-01 by hansmbakker