bblanchon/ArduinoJson #1957: .add doesn't keep the order, last becomes first.

gotfredsen, 2023-08-10:
**Describe the issue**  
I am creating an array with groups of elements, but as I add the groups, the last group becomes first group.

**Troubleshooter report**  
1. The issue happens at run time
2. The issue concerns serialization
3. Output is incomplete
4. Program doesn't call `serializeJson()` in a loop
5. `JsonDocument::overflowed()` returns `'class ArduinoJson::V6213PB2::JsonObject' has no member named 'overflowed'`
6. No string contains a NUL

**Environment**  
Here is the environment that I'm using':
* Microconroller: ESP32
* Core/runtime: ESP32 in PlatformIO
* IDE: PlatformIO 6.1.9
* ArduinoJson @ 6.21.3

**Reproduction**  
Here is a small snippet that demonstrate the problem.

```c++
    JsonArray dataArray = doc.createNestedArray("IGP");

    JsonObject dataSet = dataArray.createNestedObject();  // First data set
    dataSet["name"] = "PM1 Atm";
    dataSet["value"] = pm1a;
    dataSet["unit"] = "µg/m3";

    dataArray.add(dataSet);  // Subsequent data sets
    dataSet["name"] = "PM2.5 Atm";
    dataSet["value"] = pm25a;
    dataSet["unit"] = "µg/m3";

    dataArray.add(dataSet);  // Subsequent data sets
    dataSet["name"] = "PM10 Atm";
    dataSet["value"] = pm10a;
    dataSet["unit"] = "µg/m3";

    dataArray.add(dataSet);  // Subsequent data sets
    dataSet["name"] = "PM1 Std";
    dataSet["value"] = pm1s;
    dataSet["unit"] = "µg/m3";

    dataArray.add(dataSet);  // Subsequent data sets
    dataSet["name"] = "PM2.5 Std";
    dataSet["value"] = pm25s;
    dataSet["unit"] = "µg/m3";

    dataArray.add(dataSet);  // Subsequent data sets
    dataSet["name"] = "PM10 Std";
    dataSet["value"] = pm10s;
    dataSet["unit"] = "µg/m3";
```

**Program output**  
If relevant, include the program output.

Expected output:
```JSON
{
  "IGP": [
    {
      "name": "PM1 Atm",
      "value": 4,
      "unit": "µg/m3"
    },
    {
      "name": "PM2.5 Atm",
      "value": 20,
      "unit": "µg/m3"
    },
    {
      "name": "PM10 Atm",
      "value": 20,
      "unit": "µg/m3"
    },
    {
      "name": "PM1 Std",
      "value": 4,
      "unit": "µg/m3"
    },
    {
      "name": "PM2.5 Std",
      "value": 20,
      "unit": "µg/m3"
    },
    {
      "name": "PM10 Std",
      "value": 20,
      "unit": "µg/m3"
    }
  ]
}
```

Actual Output:
```JSON
{
  "IGP": [
    {
      "name": "PM10 Std",
      "value": 20,
      "unit": "µg/m3"
    },
    {
      "name": "PM1 Atm",
      "value": 4,
      "unit": "µg/m3"
    },
    {
      "name": "PM2.5 Atm",
      "value": 20,
      "unit": "µg/m3"
    },
    {
      "name": "PM10 Atm",
      "value": 20,
      "unit": "µg/m3"
    },
    {
      "name": "PM1 Std",
      "value": 4,
      "unit": "µg/m3"
    },
    {
      "name": "PM2.5 Std",
      "value": 20,
      "unit": "µg/m3"
    }
  ]
}
```

bblanchon, 2023-08-11:
Hi @gotfredsen,

You're mutating the first object and adding copies of it, which might give the illusion that elements are out of order, but it's only the first one that is incorrect.

Instead, you should call `createNestedObject()` for each nested object.
Demo: https://wandbox.org/permlink/iww2dwfq4HBUvBpj

Best regards,
Benoit

gotfredsen, 2023-08-11:
Thank you, that solved it.