bblanchon/ArduinoJson #1962: failed: NoMemory

stspanhe, 2023-08-22:
**Description**
I'm trying to deserialize a high amount of data with deserializeJson(doc, http.getStream());
I used the assistant deserializeJson filter stream and calculated the amount of memory needed. The assistant tells me 256 since I try only with two parameters. My MaxAllocHeap is = 110580.

**Troubleshooter's report**
1. The issue happens at run time
2. The issue concerns deserialization
3. `deserializeJson()` returns `NoMemory`
4. Increasing the capacity of the `JsonDocument` doesn't fix the issue
5. The `JsonDocument` is a `DynamicJsonDocument`
6. `doc.capacity()` returns the right value
7. Changing the Assistant settings doesn't solve the issue

**Environment**
* Microcontroller: ESP32
* Core/Framework: DOIT ESP32 DEVKIT V1
* IDE: Arduinio IDE 2.1.1

**Reproduction code**
```c++
String forcast_url = "http://opendata-download-metfcst.smhi.se/api/category/pmp3g/version/2/geotype/point/lon/11.5482/lat/57.9911/data.json";

        http.useHTTP10(true);               // Remember to call useHTTP10(true) when you use getStream(), or deserializeJson() will return InvalidInput.
        Serial.print("[HTTP] begin...\n");
        http.begin (client,forcast_url);

// Stream& input;

        StaticJsonDocument<64> filter;
        filter["approvedTime"] = true;
        filter["referenceTime"] = true;

        DynamicJsonDocument doc(500);
        // ReadLoggingStream loggingStream(http.getStream(), Serial);
        // deserializeJson(doc,loggingStream); // Parse response

        WiFiClient client;
        HTTPClient http;
        http.useHTTP10(true);               // Remember to call useHTTP10(true) when you use getStream(), or deserializeJson() will return InvalidInput.
        Serial.print("[HTTP] begin...\n");
        http.begin (client,forcast_url);
        http.GET();
        Serial.print("doc.capacity()=");
        Serial.println(doc.capacity());
        DeserializationError error = deserializeJson(doc, http.getStream());
        if (error) {
          Serial.print("deserializeJson() failed: ");
          Serial.println(error.c_str());
          return;
        }
        http.end();
        
        
        const char* approvedTime = doc["approvedTime"]; // "2023-08-22T16:07:27Z"
        const char* referenceTime = doc["referenceTime"]; // "2023-08-22T16:00:00Z"
```

bblanchon, 2023-08-23:
Hi @stspanhe,

You forgot to pass the filter to [`deserializeJson()`](https://arduinojson.org/v6/api/json/deserializejson/):

```c++
deserializeJson(doc, http.getStream(), DeserializationOption::Filter(filter));
```

Best regards,
Benoit

stspanhe, 2023-08-25:
Thanks @bblanchon, how stupid of me to miss that. :D
Now I've  got the filter to work properly, but still I'm out of memory with the filters I want. So currently looking into Deserializing in chunks. Haven't got it work yet though, the Stream::find() (response. find("parameters:[")) want find anything in my stream

bblanchon, 2023-09-22:
@stspanhe, can I close this issue?

closed 2023-10-09 by bblanchon