bblanchon/ArduinoJson #1964: Feature request: JsonPair.key().as<Type>()

razvanphp, 2023-09-11:
Hey, thank you for your great work!

It would be a lot cleaner (and cleaner) if this functionality would be implemented 🙂 

Currently, only the `.value()` has it as it is [implemented in JsonVariant](https://arduinojson.org/v6/api/jsonvariant/as/).

My code snippet:

```
for (JsonPair el : json.as<JsonObject>()) {
    Serial.println(atoi(el.key().c_str()));
    Serial.println(el.value().as<int>());
}
```

Cheers,
R

bblanchon, 2023-09-12:
Hi @razvanphp,

`key()` is a string (a `JsonString` to be exact), why would you expect it to have the same interface as a variant?

Best regard,
Benoit

bblanchon, 2023-09-13:
So, you're asking for `JsonString::as<int>()`, right?

I must say that I'm not fond of this idea:

1. It blurs the message that `JsonString` is a string and not a `JsonVariant`.
2. It would work for `double` and `bool` as well, but I doubt anyone will ever use it.
3. It would not work for `JsonArray` and `JsonObject` even if the key contains a JSON array/object, and even if it worked, who would do that?

I'm sorry, but I'll need to see some very convincing code examples before adding something like that in the library.

razvanphp, 2023-09-13:
Well, I was using it just for int, as you can see above in my code snippet, because I was setting registers in arduino and the key in json is not valid as integer, it needs to be string.

Indeed `JsonString` is a string representation, but one would still need to call `c_str()` to really use it as a string, why wouldn't it be another convenience method to convert to any other type?

Obviously It's your call and feel free to discard my suggestion 🙂