bblanchon/ArduinoJson #1965: JsonArduino 6.15.2: Add a non-null terminated string to a jsonDocument

NicoMayer, 2023-09-12:
Hello everyone, does anyone know how to add a char string without null character in ArduinoJson 6.15.2 to a JsonDocument?

I found this SizedRamStringAdapter in the code. But I don't know how to use this.
```c
...
ARDUINOJSON_NAMESPACE::SizedRamStringAdapter dataAdapter =ARDUINOJSON_NAMESPACE::adaptString(dataBuffer, dataBufferLen);
jsonResult["data"] = dataAdapter;
...
```

Would anyone have an idea or tip for me?

Thank you very much and best regards
Nico

bblanchon, 2023-09-12:
Hi Nico,

String adapters are part of the internal API.
You should use [`JsonString`](https://arduinojson.org/v6/api/jsonstring/) instead:

```c++
jsonResult["data"] = JsonString(dataBuffer, dataBufferLen);
```

Best regards,
Benoit

NicoMayer, 2023-09-13:
Hello, thank you for your answer.

I tried your lines of code. The string constructor with the size parameter was not yet available in ArduinoJson 6.15.2

I will now simply manually add a null character to the end of the data buffer

Best regards
Nico

bblanchon, 2023-09-14:
Hi Nico,

Why don't you upgrade to 6.21?

Best regards,
Benoit

NicoMayer, 2023-09-14:
Yeah, that's a complicated story.

The ArduinoJson 6.15 is part of my company's embedded framework. I am not involved in the development of this framework and therefore cannot update ArduinoJson.

I will now talk to the development team and suggest an upgrade of ArduinoJson.

Thank you for your support and best regards,
Nico

bblanchon, 2023-09-15:
You're welcome :-)
Be aware that there are some breaking changes between 6.15 and 6.21.
See: https://arduinojson.org/v6/revisions/

closed 2023-09-22 by bblanchon