bblanchon/ArduinoJson #1966: Strange issue.

zekageri, 2023-09-12:
I have two methods. `saveAll()` and `getPinsJson(JsonObject modulePins);`

```cpp
void HwConfig::saveAll(){
    DynamicJsonDocument doc(MODULES_DOC_SIZE);
    for (size_t i = 0; i < modules.size(); i++){
        JsonObject moduleObj = doc.createNestedObject( String(modules[i].address) );
        JsonObject modulePins = moduleObj.createNestedObject("pins");
        moduleObj["name"]   = modules[i].name;
        modules[i].getPinsJson( modulePins );
        // Below condition is for debug
        if( modules[i].id == 20 ){
            for (JsonPair kv : modulePins) {
                JsonObject pinObj = kv.value().as<JsonObject>();
                Serial.printf("(save) - %s\n", pinObj["name"].as<std::string>().c_str());
            }
        }
    }
}
```

**saveAll() calls getPinsJson()**

```cpp
void Module::getPinsJson(JsonObject pinsObj){
    for (size_t i = 0; i < pins.size(); i++){
        JsonObject pinObj = pinsObj.createNestedObject(String(pins[i].readablePinNumber));
        pinObj["name"] = pins[i].name;
        // Below condition is for debug
        if( id == 20 ){
            Serial.printf("[PIN_LOAD] - %s\n",pinObj["name"].as<std::string>().c_str());
        }
    }
}
```

In the `getPinsJson()` method i got the correct names

```cpp
[PIN_LOAD] - M8-IN 1
[PIN_LOAD] - M8-IN 2
[PIN_LOAD] - M8-IN 3
[PIN_LOAD] - M8-IN 4
[PIN_LOAD] - M8-IN 5
[PIN_LOAD] - M8-IN 6
[PIN_LOAD] - M8-IN 7
[PIN_LOAD] - M8-IN 8
[PIN_LOAD] - M8-IN 9
[PIN_LOAD] - M8-IN 10
[PIN_LOAD] - M8-IN 11
[PIN_LOAD] - M8-IN 12
[PIN_LOAD] - M8-IN 13
[PIN_LOAD] - M8-IN 14
[PIN_LOAD] - M8-IN 15
[PIN_LOAD] - M8-IN 16
[PIN_LOAD] - M8-OUT 1 (A)
[PIN_LOAD] - M8-OUT 1 (B)
[PIN_LOAD] - M8-OUT 2 (A)
[PIN_LOAD] - M8-OUT 2 (B)
[PIN_LOAD] - M8-OUT 3 (A)
[PIN_LOAD] - M8-OUT 3 (B)
[PIN_LOAD] - M8-OUT 4 (A)
[PIN_LOAD] - M8-OUT 4 (B)
[PIN_LOAD] - M8-OUT 5 (A)
[PIN_LOAD] - M8-OUT 5 (B)
[PIN_LOAD] - M8-OUT 6 (A)
[PIN_LOAD] - M8-OUT 6 (B)
[PIN_LOAD] - M8-OUT 7 (A)
[PIN_LOAD] - M8-OUT 7 (B)
[PIN_LOAD] - M8-OUT 8 (A)
[PIN_LOAD] - M8-OUT 8 (B)
```

The above log is correct.
In the `saveAll()` method i got this

```cpp
(save) - M8-OUT 1 (B)
(save) - M8-OUT 2 (B)
(save) - M8-OUT 3 (B)
(save) - M8-OUT 4 (B)
(save) - M8-OUT 5 (B)
(save) - M8-OUT 6 (B)
(save) - M8-OUT 7 (B)
(save) - M8-OUT 8 (B)
(save) - M8-IN 9
(save) - M8-IN 10
(save) - M8-IN 11
(save) - M8-IN 12
(save) - M8-IN 13
(save) - M8-IN 14
(save) - M8-IN 15
(save) - M8-IN 16
```

And only this. Nothing more.
What can be the problem?

pins is a _**vector**_ of `Pin` classes
`std::vector<Pin> pins;`

Thank you for your help.

zekageri, 2023-09-13:
I have placed `isNull()` checks but they all pass.

```cpp
JsonObject pinObj = pinsObj.createNestedObject(String(pins[i].readablePinNumber));
if( pinObj.isNull() ){
    Serial.printf("[PIN_LOAD] - Failed to create %s\n",pins[i].name.c_str());
    continue;
}

if( moduleObj.isNull() ){
    if( debug ){
        Serial.printf("[HwConfig] - Failed to create %s\n",moduleKey.c_str());
    }
    continue;
}
if( modulePins.isNull() ){
    if( debug ){
        Serial.printf("[HwConfig] - Failed to create pins for %s\n",moduleKey.c_str());
    }
    continue;
}
```

zekageri, 2023-09-13:
Oh it seems to me that here is my problem

```cpp
DynamicJsonDocument doc(MODULES_DOC_SIZE);
if( doc.isNull() ){
    Serial.printf("[HwConfig] - Failed to create json doc for save.\n");
}
```

It logs out. Why can't it create the json doc i wonder...

bblanchon, 2023-09-13:
Hi @zekageri,

I spent quite some time trying to understand your program, but it's a bit too convoluted for me.
Could you narrow down the issue to a simpler example?

BTW, did you consult the ~~oracle~~ [troubleshooter](https://arduinojson.org/v6/troubleshooter/)?

Best regards,
Benoit

zekageri, 2023-09-13:
I consulted the troubleshooter but i don't have a problem with neither serialization or deserialization

I'm creating a DynamicJsonDocument and trying to fill it with data.
After it is filled, i want to save it to LittleFS.
That's all.

Here is a simple example

```cpp
#include <ArduinoJson.h>

#define MODULES_DOC_SIZE 10000

void fillPins(JsonObject pinsObj){
  for (size_t i = 0; i < 5; i++){
    String pinKey = "Pin-" + String(i);
    JsonObject pinObj = pinsObj.createNestedObject(pinKey);
    pinObj["name"] = pinKey;
  }
}

void save(){
  DynamicJsonDocument doc(MODULES_DOC_SIZE);
  for (size_t i = 0; i < 10; i++){
    String moduleKey = String(i);
    JsonObject moduleObj = doc.createNestedObject( moduleKey );
    JsonObject modulePins = moduleObj.createNestedObject("pins");
    moduleObj["name"] = "Test-" + String(i);
    fillPins(modulePins);
  }
  serializeJsonPretty(doc,Serial);
}

void setup() {
  Serial.begin(115200);
  save();
}

void loop() {
  delay(10);
}

```

zekageri, 2023-09-13:
It is workking on WokWi so clearly i screw up something...

https://wokwi.com/projects/375736094955068417

zekageri, 2023-09-13:
I will try with this allocator

```cpp
struct SpiRamAllocator {
    void* allocate(size_t size) {return heap_caps_malloc(size, MALLOC_CAP_SPIRAM);}
    void deallocate(void* pointer) {heap_caps_free(pointer);}
    void* reallocate(void* ptr, size_t new_size) {return heap_caps_realloc(ptr, new_size, MALLOC_CAP_SPIRAM);}
};
using SpiRamJsonDocument = BasicJsonDocument<SpiRamAllocator>;
```

Maybe i have not enough heap and the arduino core is not working. ( They said that the malloc should automatically allocate ram from the ps ram now )

zekageri, 2023-09-13:
Well, it wasn't the insufficent internal heap.

I still get strange results.
First of all, this logs out **_Failed to create json doc for save_**, and it doesnt matter if i use `StaticJsonDocument` or `SpiRamJsonDocument` or `DynamicJsonDocument`.

```cpp
SpiRamJsonDocument doc(MODULES_DOC_SIZE);
if( doc.isNull() ){
    logger.write("[HwConfig] - Failed to create json doc for save.\n");
}
```

`MODULES_DOC_SIZE` is also doesnt matter. I have tried with several different sizes.

zekageri, 2023-09-13:
Something is wrong in how i pass a `JsonObject` to `getPinsJson()` method.
I have tried to pass it by reference. I read in the documentation that i dont have to do this.

zekageri, 2023-09-13:
So

```cpp
void Module::getPinsJson(JsonObject pinsObj);
void Module::getPinsJson(JsonObject &pinsObj);
```

Doesnt matter

zekageri, 2023-09-13:
It is a little convoluted because i have two class.

One class is for managing "modules". These module classes are in a vector. ` std::vector<Module> modules; `
`HwConfig` is iterating over this vector, trying to gather every module's data into a document.

```cpp
DynamicJsonDocument doc(MODULES_DOC_SIZE);
for (size_t i = 0; i < modules.size(); i++){
    // Accessing a module by modules[i]
}
```

Every module has input and output pins.
These pins are in separate classes which only the `module` class knows about.
So there is a method in `module` class to retrive the information about it's pins.
These pins in the `module` class are also in a vector ` std::vector<Pin> pins; `

`module` class will iterate over it's pins and gather information about them

```cpp
void Module::getPinsJson(JsonObject pinsObj){
    for (size_t i = 0; i < pins.size(); i++){
        // Access a pin with pins[i]
    }
}
```

When i print out the details about the pins in `getPinsJson()` method, i get the correct data.
Names, numbers etc. It seems fine.
But when i print the same pin `Object` inside the `saveAll()` method they are total garbage.

zekageri, 2023-09-13:
### It is something like this

[image]

zekageri, 2023-09-13:
When a module is iterating over it"s pins and i log out the gatheret information from the document in that method, it logs out fine. When i log out in the outer loop it is garbage and it saved garbage to the file system.

zekageri, 2023-09-13:
I'm totally lost. I have tried to pass the `JsonObject` by reference and as is.
I have tried `DynamicJsonDocument`, `SpiRamJsonDocument` and `StaticJsonDocument`
I have tried to create the nested objects like this ` doc[moduleKey] ` instead of `createNestedObject` ( that was silly )
```cpp
JsonObject moduleObj;
JsonObject modulePins;
// fill moduleObj and modulePins
moduleObj["pins"] = modulePins;
doc[moduleKey] = moduleObj;
```
I have tried to decrease and increase the size of the document ( from 5k to 150k )

Nothing seems to help.

zekageri, 2023-09-13:
Sometimes it puts some pins inside but it is totally garbage
For example, here is a portion of the saved file after this saveAll() call

```json
"147": {
        "pins": {
            "1": {
                "name": "CR16-IN 1",
                "type": 4,
                "value": 0,
                "state": false,
                "regNumber": 0,
                "pinNumber": 0,
                "readablePinNumber": 1,
                "faultCode": 0,
                "forced": false
            },
            "2": {
                "name": "R32-IN 2",
                "type": 4,
                "value": 0,
                "state": false,
                "regNumber": 1,
                "pinNumber": 1,
                "readablePinNumber": 2,
                "faultCode": 0,
                "forced": false
            }
        },
        "id": 179,
        "serial": 102,
        "address": 147,
        "name": "TH02B_1",
        "used": false
    },
```

Module 147 doesn't even have pins and the pins vector is empty on that module.
Even it had something it shouldn't be CR16-IN 1 and R32-IN 2. What is happening in here.

zekageri, 2023-09-13:
Here is a wokwi
https://wokwi.com/projects/375743724075751425
It works fine on there....

zekageri, 2023-09-13:
Here is the actual method i use for generating the pins for the M8 module

```cpp
void Module::generateM8OutputPins(int outputPinCount) {
    for (int i = 0; i < outputPinCount; i += 2) {
        correctPin pinOptA = getCorrectPin(i);
        correctPin pinOptB = getCorrectPin(i + 1);

        for (int j = 0; j < 2; j++) {
            Pin outputPin;
            outputPin.name = name + "-OUT " + std::to_string(i / 2 + 1) + (j == 0 ? " (A)" : " (B)");
            outputPin.regNumber = (j == 0) ? pinOptA.regNum : pinOptB.regNum;
            outputPin.pinNumber = (j == 0) ? pinOptA.pin : pinOptB.pin;
            outputPin.readablePinNumber = i / 2 + 1;
            outputPin.type = PIN_TYPE_MOTOR;
            pins.push_back(outputPin);

            logger.write("[PIN_GENERATE] - %s\n",outputPin.name.c_str());
        }
    }
    if( debug ){
        logger.write("[MODULE] - %d output pins generated for %s\n",outputPinCount,name.c_str());
    }
}
```

This method is generating the following pattern:

```cpp
[PIN_GENERATE] - M8-OUT 1 (A)
[PIN_GENERATE] - M8-OUT 1 (B)
[PIN_GENERATE] - M8-OUT 2 (A)
[PIN_GENERATE] - M8-OUT 2 (B)
[PIN_GENERATE] - M8-OUT 3 (A)
[PIN_GENERATE] - M8-OUT 3 (B)
[PIN_GENERATE] - M8-OUT 4 (A)
[PIN_GENERATE] - M8-OUT 4 (B)
[PIN_GENERATE] - M8-OUT 5 (A)
[PIN_GENERATE] - M8-OUT 5 (B)
[PIN_GENERATE] - M8-OUT 6 (A)
[PIN_GENERATE] - M8-OUT 6 (B)
[PIN_GENERATE] - M8-OUT 7 (A)
[PIN_GENERATE] - M8-OUT 7 (B)
[PIN_GENERATE] - M8-OUT 8 (A)
[PIN_GENERATE] - M8-OUT 8 (B)
```

And it generating fine. My pins vector contains these pins as expected.

zekageri, 2023-09-13:
But in the saved file:

```json
"22": {
"pins": {
  "1": {
      "name": "M8-OUT 1 (B)",
      "type": 3,
      "value": 0,
      "state": false,
      "regNumber": 0,
      "pinNumber": 1,
      "readablePinNumber": 1,
      "faultCode": 0,
      "forced": false
  },
  "2": {
      "name": "M8-OUT 2 (B)",
      "type": 3,
      "value": 0,
      "state": false,
      "regNumber": 0,
      "pinNumber": 3,
      "readablePinNumber": 2,
      "faultCode": 0,
      "forced": false
  },
  "3": {
      "name": "M8-OUT 3 (B)",
      "type": 3,
      "value": 0,
      "state": false,
      "regNumber": 0,
      "pinNumber": 5,
      "readablePinNumber": 3,
      "faultCode": 0,
      "forced": false
  },
  "4": {
      "name": "M8-OUT 4 (B)",
      "type": 3,
      "value": 0,
      "state": false,
      "regNumber": 0,
      "pinNumber": 7,
      "readablePinNumber": 4,
      "faultCode": 0,
      "forced": false
  },
  "5": {
      "name": "M8-OUT 5 (B)",
      "type": 3,
      "value": 0,
      "state": false,
      "regNumber": 1,
      "pinNumber": 1,
      "readablePinNumber": 5,
      "faultCode": 0,
      "forced": false
  },
  "6": {
      "name": "M8-OUT 6 (B)",
      "type": 3,
      "value": 0,
      "state": false,
      "regNumber": 1,
      "pinNumber": 3,
      "readablePinNumber": 6,
      "faultCode": 0,
      "forced": false
  },
  "7": {
      "name": "M8-OUT 7 (B)",
      "type": 3,
      "value": 0,
      "state": false,
      "regNumber": 1,
      "pinNumber": 5,
      "readablePinNumber": 7,
      "faultCode": 0,
      "forced": false
  },
  "8": {
      "name": "M8-OUT 8 (B)",
      "type": 3,
      "value": 0,
      "state": false,
      "regNumber": 1,
      "pinNumber": 7,
      "readablePinNumber": 8,
      "faultCode": 0,
      "forced": false
  },
  "9": {
      "name": "M8-IN 9",
      "type": 4,
      "value": 0,
      "state": false,
      "regNumber": 8,
      "pinNumber": 8,
      "readablePinNumber": 9,
      "faultCode": 0,
      "forced": false
  },
  "10": {
      "name": "M8-IN 10",
      "type": 4,
      "value": 0,
      "state": false,
      "regNumber": 9,
      "pinNumber": 9,
      "readablePinNumber": 10,
      "faultCode": 0,
      "forced": false
  },
  "11": {
      "name": "M8-IN 11",
      "type": 4,
      "value": 0,
      "state": false,
      "regNumber": 10,
      "pinNumber": 10,
      "readablePinNumber": 11,
      "faultCode": 0,
      "forced": false
  },
  "12": {
      "name": "M8-IN 12",
      "type": 4,
      "value": 0,
      "state": false,
      "regNumber": 11,
      "pinNumber": 11,
      "readablePinNumber": 12,
      "faultCode": 0,
      "forced": false
  },
  "13": {
      "name": "M8-IN 13",
      "type": 4,
      "value": 0,
      "state": false,
      "regNumber": 12,
      "pinNumber": 12,
      "readablePinNumber": 13,
      "faultCode": 0,
      "forced": false
  },
  "14": {
      "name": "M8-IN 14",
      "type": 4,
      "value": 0,
      "state": false,
      "regNumber": 13,
      "pinNumber": 13,
      "readablePinNumber": 14,
      "faultCode": 0,
      "forced": false
  },
  "15": {
      "name": "M8-IN 15",
      "type": 4,
      "value": 0,
      "state": false,
      "regNumber": 14,
      "pinNumber": 14,
      "readablePinNumber": 15,
      "faultCode": 0,
      "forced": false
  },
  "16": {
      "name": "M8-IN 16",
      "type": 4,
      "value": 0,
      "state": false,
      "regNumber": 15,
      "pinNumber": 15,
      "readablePinNumber": 16,
      "faultCode": 0,
      "forced": false
  }
},
"id": 20,
"serial": 2,
"address": 22,
"name": "M8",
"used": false
}
```

It should also have 16 input pins in this vector.

zekageri, 2023-09-13:
If i log out the pins vector right after i created it

```cpp
for (size_t i = 0; i < pins.size(); i++){
   logger.write("[PIN_GENERATED] - %s\n",pins[i].name.c_str());
}
```
```cpp
[PIN_GENERATED] - M8-IN 1
[PIN_GENERATED] - M8-IN 2
[PIN_GENERATED] - M8-IN 3
[PIN_GENERATED] - M8-IN 4
[PIN_GENERATED] - M8-IN 5
[PIN_GENERATED] - M8-IN 6
[PIN_GENERATED] - M8-IN 7
[PIN_GENERATED] - M8-IN 8
[PIN_GENERATED] - M8-IN 9
[PIN_GENERATED] - M8-IN 10
[PIN_GENERATED] - M8-IN 11
[PIN_GENERATED] - M8-IN 12
[PIN_GENERATED] - M8-IN 13
[PIN_GENERATED] - M8-IN 14
[PIN_GENERATED] - M8-IN 15
[PIN_GENERATED] - M8-IN 16
[PIN_GENERATED] - M8-OUT 1 (A)
[PIN_GENERATED] - M8-OUT 1 (B)
[PIN_GENERATED] - M8-OUT 2 (A)
[PIN_GENERATED] - M8-OUT 2 (B)
[PIN_GENERATED] - M8-OUT 3 (A)
[PIN_GENERATED] - M8-OUT 3 (B)
[PIN_GENERATED] - M8-OUT 4 (A)
[PIN_GENERATED] - M8-OUT 4 (B)
[PIN_GENERATED] - M8-OUT 5 (A)
[PIN_GENERATED] - M8-OUT 5 (B)
[PIN_GENERATED] - M8-OUT 6 (A)
[PIN_GENERATED] - M8-OUT 6 (B)
[PIN_GENERATED] - M8-OUT 7 (A)
[PIN_GENERATED] - M8-OUT 7 (B)
[PIN_GENERATED] - M8-OUT 8 (A)
[PIN_GENERATED] - M8-OUT 8 (B)
```
It shows the correct pins, so the problem has to be when i create JsonObjects on save.

zekageri, 2023-09-13:
I narrowed down the problem to the `getPinsJson(JsonObject &pinsObj)` method.

```cpp
void Module::getPinsJson(JsonObject &pinsObj){
    for(Pin pin : pins){
        String key = String(pin.readablePinNumber);
        JsonObject pinObj = pinsObj.createNestedObject(key);
        pinObj["name"] = pin.name;
        pinObj["type"] = pin.type;
        pinObj["value"] = pin.value;
        pinObj["state"] = pin.state;
        pinObj["regNumber"] = pin.regNumber;
        pinObj["pinNumber"] = pin.pinNumber;
        pinObj["readablePinNumber"] = pin.readablePinNumber;
        pinObj["faultCode"] = pin.faultCode;
        pinObj["forced"] = pin.forced;
        logger.write("(1)[PIN_LOAD] - %s\n",pinObj["name"].as<std::string>().c_str());
    }
    // Iterate over pinsObj
    for(JsonPair kv : pinsObj){
        JsonObject pinObj = kv.value().as<JsonObject>();
        logger.write("(2)[PIN_LOAD] - %s\n",pinObj["name"].as<std::string>().c_str());
    }
}
```

Here is the log result:

```cpp
(1)[PIN_LOAD] - M8-IN 1
(1)[PIN_LOAD] - M8-IN 2
(1)[PIN_LOAD] - M8-IN 3
(1)[PIN_LOAD] - M8-IN 4
(1)[PIN_LOAD] - M8-IN 5
(1)[PIN_LOAD] - M8-IN 6
(1)[PIN_LOAD] - M8-IN 7
(1)[PIN_LOAD] - M8-IN 8
(1)[PIN_LOAD] - M8-IN 9
(1)[PIN_LOAD] - M8-IN 10
(1)[PIN_LOAD] - M8-IN 11
(1)[PIN_LOAD] - M8-IN 12
(1)[PIN_LOAD] - M8-IN 13
(1)[PIN_LOAD] - M8-IN 14
(1)[PIN_LOAD] - M8-IN 15
(1)[PIN_LOAD] - M8-IN 16
(1)[PIN_LOAD] - M8-OUT 1 (A)
(1)[PIN_LOAD] - M8-OUT 1 (B)
(1)[PIN_LOAD] - M8-OUT 2 (A)
(1)[PIN_LOAD] - M8-OUT 2 (B)
(1)[PIN_LOAD] - M8-OUT 3 (A)
(1)[PIN_LOAD] - M8-OUT 3 (B)
(1)[PIN_LOAD] - M8-OUT 4 (A)
(1)[PIN_LOAD] - M8-OUT 4 (B)
(1)[PIN_LOAD] - M8-OUT 5 (A)
(1)[PIN_LOAD] - M8-OUT 5 (B)
(1)[PIN_LOAD] - M8-OUT 6 (A)
(1)[PIN_LOAD] - M8-OUT 6 (B)
(1)[PIN_LOAD] - M8-OUT 7 (A)
(1)[PIN_LOAD] - M8-OUT 7 (B)
(1)[PIN_LOAD] - M8-OUT 8 (A)
(1)[PIN_LOAD] - M8-OUT 8 (B)
(2)[PIN_LOAD] - M8-OUT 1 (B)
(2)[PIN_LOAD] - M8-OUT 2 (B)
(2)[PIN_LOAD] - M8-OUT 3 (B)
(2)[PIN_LOAD] - M8-OUT 4 (B)
(2)[PIN_LOAD] - M8-OUT 5 (B)
(2)[PIN_LOAD] - M8-OUT 6 (B)
(2)[PIN_LOAD] - M8-OUT 7 (B)
(2)[PIN_LOAD] - M8-OUT 8 (B)
(2)[PIN_LOAD] - M8-IN 9
(2)[PIN_LOAD] - M8-IN 10
(2)[PIN_LOAD] - M8-IN 11
(2)[PIN_LOAD] - M8-IN 12
(2)[PIN_LOAD] - M8-IN 13
(2)[PIN_LOAD] - M8-IN 14
(2)[PIN_LOAD] - M8-IN 15
(2)[PIN_LOAD] - M8-IN 16
```

The (1)[PIN_LOAD] prints are correct. But as soon as it leaves the for loop it becomes corrupt.
I have allocated plenty of space for the document.

bblanchon, 2023-09-14:
Even if all `pin.name`s are different, many `pin.readablePinNumber`s are identical, so `pinsObj.createNestedObject(String(pin.readablePinNumber))` overrides the previously created objects.

Solution: don't use `pin.readablePinNumber` as the key.

zekageri, 2023-09-14:
Ooooooohhhh. God when i read your comment i almost fainted. This might be it. Thank you thank you. How was i that stupid. I will test it.

zekageri, 2023-09-15:
Okay, it was that. All keys had to be unique. Thank you again and sorry for the issue.

closed 2023-09-15 by zekageri