bblanchon/ArduinoJson #1970: Cannot compile code that used to work

viktak, 2023-09-14:
**Describe the issue**  
I have an ESP8266 based project that I created a couple of years ago, using this wonderful library (like in all my projects:)) Now I wanted to make a small change in my original code, but I cannot even compile it as it is. I am getting a lot of compiler errors, but all of them seem to originate from ArduinoJSON. I think the problem is that over the year(s) I haven't touched the project, some of the other libraries, platform, etc. got updated, and silly me, I haven't originally specify the version numbers required in `platformio.ini`.

**Troubleshooter report**  
Here is the report generated by the [ArduinoJson Troubleshooter](https://arduinojson.org/v6/troubleshooter/):  
[Paste the report here]

**Environment**  
Here is the environment that I'm using':
* Microconroller: ESP8266
* Core/runtime: ESP8266 core for Arduino v4.2.1
* IDE: PlatformIO 3.3.1 (2023-07-28)

**Reproduction**  
Unfortunately, I don't even know which part of my code uses the offending bits.

Even if I comment out all the ArduinoJson related stuff from my code, but leave `#include <ArduinoJson.h>` I get the following errors:
```
[{
	"resource": "/d:/Projects/Archlight/archlight/.pio/libdeps/esp12e/ArduinoJson/src/ArduinoJson/Variant/ConverterImpl.hpp",
	"owner": "cpp",
	"severity": 8,
	"message": "no matching function for call to 'convertToJson(const char&, ArduinoJson::V6213PB2::JsonVariant&)'",
	"startLineNumber": 17,
	"startColumn": 18,
	"endLineNumber": 17,
	"endColumn": 18
},{
	"resource": "/d:/Projects/Archlight/archlight/.pio/libdeps/esp12e/ArduinoJson/src/ArduinoJson/Variant/ConverterImpl.hpp",
	"owner": "cpp",
	"severity": 8,
	"message": "conversion to non-const reference type 'using string_view = class std::basic_string_view<char>&' {aka 'class std::basic_string_view<char>&'} from rvalue of type 'std::basic_string_view<char>' [-fpermissive]",
	"startLineNumber": 24,
	"startColumn": 20,
	"endLineNumber": 24,
	"endColumn": 20
},{
	"resource": "/d:/Projects/Archlight/archlight/.pio/libdeps/esp12e/ArduinoJson/src/ArduinoJson/Variant/ConverterImpl.hpp",
	"owner": "cpp",
	"severity": 8,
	"message": "invalid user-defined conversion from 'char' to 'std::string_view&' {aka 'std::basic_string_view<char>&'} [-fpermissive]",
	"startLineNumber": 24,
	"startColumn": 26,
	"endLineNumber": 24,
	"endColumn": 26
},{
	"resource": "/d:/Projects/Archlight/archlight/.pio/libdeps/esp12e/ArduinoJson/src/ArduinoJson/Variant/ConverterImpl.hpp",
	"owner": "cpp",
	"severity": 8,
	"message": "no type named 'type' in 'struct ArduinoJson::V6213PB2::detail::enable_if<false, bool>'",
	"startLineNumber": 156,
	"startColumn": 1,
	"endLineNumber": 156,
	"endColumn": 1
}]
```
This tells me that it is not my code that has the issue. Also, I don't think ArduinoJson causes the problem. It must be some configuration problem. That's why I logged this question here. Maybe someone with more experience knows the problem right away..

I have tried deleting my `.platformio` directory, `full clean`ing the project, but it still won't compile...

Any idea where to look next?

bblanchon, 2023-09-15:
Hi @viktak,

You copied these errors for the "Problem" tab, but you'd get way more information if you looked a the "Terminal" tab.

From what you shared, I can only tell that your program uses a `char`, which is deprecated since [6.18](https://arduinojson.org/news/2021/05/04/version-6-18-0/) and unsupported since [6.20](https://arduinojson.org/news/2022/12/26/arduinojson-6-20-0/).

You'll have to look at the "Terminal" tab to find out where this `char` is.
I invite you to consult the [Troubleshooter](https://arduinojson.org/troubleshooter); it will tell you how to fix this error.

Best regards,
Benoit

viktak, 2023-09-15:
Thank you Benoit, this version number helped. Reverting to `bblanchon/ArduinoJson @ 6.18.0` solved my issue.

This, however, still doesn't answer why compilation fails even if I don't call it at all in my code.
Also, I just noticed, that using `bblanchon/ArduinoJson @ ^6.18.0` also fails. I thought this was the safest option to use: "Accept new functionality in a backwards compatible manner and patches"

Anyway, thanks for putting me in the right direction!!!

closed 2023-09-15 by viktak