bblanchon/ArduinoJson #1971: ArduinoJson USB Packet buffering in STM32Cube C application

samspencer5991, 2023-09-18:
**Describe the issue**  
I am currently using ArduinoJSON in a native C application using PlatformIO for an STM32Cube project. I am using a custom writer to invoke TinyUSB CDC output for the serialization. My documents are fairly large, often around 8-12KB. Everything is running well however the performance and related issues of ArduinoJson invoking the writer for heavily fragmented write calls. Because I am not using Arduino, I cannot use the suggestion of StreamUtils to buffer the write. Something like that is an ideal solution though.

Are there any suggestions for being able to: Write the serialization in consecutive chunks (i.e. write it to either an output buffer or my custom writer in 64 byte USB packet chunks until the entire serialization is complete)?

**Environment**  
Here is the environment that I'm using':
* Microconroller: STM32G491
* Core/runtime: STM32Cube
* IDE: PlatformIO

bblanchon, 2023-09-19:
Hi @samspencer5991,

Indeed, `serializeJson()` doesn't buffer the characters before sending them to the writer.
To improve the performance, you'll have to add a buffer in your writer class.

Best regards,
Benoit

samspencer5991, 2023-09-19:
Sure, that doesn't address my question specifically though. One issue is there doesn't seem to be a mechanism for the writer to know how many bytes are in the way. Of particular interest is the 64 byte USB packet size which would be ideal for transmitting a full packet over USB. But it can't afford to wait for too long or risk missing the next USB timing frame.

bblanchon, 2023-09-20:
Why don't you use a 64-byte buffer in your custom writer class and send it as soon as it's full?

samspencer5991, 2023-09-20:
I thought about that, but the problem is the case where there are less than 64 bytes. Any delay will cause huge delays in sending consecutive large documents, and also risk missing the USB timing frame. Ideally, a dummy serialise function which returns the number of bytes required, then the custom wittier serialise would solve this.

bblanchon, 2023-09-21:
I would simply send the remaining bytes right after calling `serializeJson()`.
For example, you could add a `flush()` function to your custom writer:

```c++
MyBufferedWriter writer;
serializeJson(doc, writer);
writer.flush();
```

There shouldn't be any delay there.

samspencer5991, 2023-09-21:
I'm really not sure how that helps sorry. The use case is this. Let's assume that a JSON document consists of several thousand bytes. For optimum efficiency, they should be packed into 64-byte packets, and sent. If the writer functions simply collect a number of bytes, put them into the buffer, and then transmit that as a packet once the buffer is full, what mechanism is there for the writer to know the difference between 38 (for example) bytes in the buffer with more on the way, vs. the last 38 bytes that can be transmitted right away?

bblanchon, 2023-09-22:
Well, that's what the `flush()` function is here for: it tells the writer to send the last 38 bytes.

samspencer5991, 2023-09-26:
This perfectly solved my problem, I was completely overthinking things in the trenches of a big code-base re-write. Thanks @bblanchon !

closed 2023-09-26 by samspencer5991