bblanchon/ArduinoJson #1973: User exception (panic/abort/assert) Nodemcu

BimoSora2, 2023-09-19:
I have a problem, my code panics when using json to install to the webserver. I do not know why?

`// Import required libraries
#include <Arduino.h>
#ifdef ESP32
#include <WiFi.h>
#include <AsyncTCP.h>
#elif defined(ESP8266)
#include <ESP8266WiFi.h>
#include <ESPAsyncTCP.h>
#endif
#include <ESPAsyncWebServer.h>
#include <FS.h>
#include <ArduinoJson.h>

int temp, pressure, range;
float lux;

String temp1 = "";
String pressure1 = "";
int range1;
String rain = "";
float lux1;
String light = "";

// Timer variables
unsigned long lastTime = 0;  
unsigned long timerDelay = 20;

/* Put your SSID & Password */
const char* ssid = "Weather Station";  // Enter SSID here
const char* password = "12345678";  //Enter Password here

/* Put IP Address details */
IPAddress local_ip(192,168,1,1);
IPAddress gateway(192,168,1,1);
IPAddress subnet(255,255,255,0);

// Create AsyncWebServer object on port 80
AsyncWebServer server(80);

void setup() {
  // Initialize Serial port
  Serial.begin(9600);
  while (!Serial) continue;

  if(SPIFFS.begin()==true) {
    Serial.println("SPIFFS initialised OK");
  }

  WiFi.softAP(ssid, password);
  WiFi.softAPConfig(local_ip, gateway, subnet);
  delay(100);
}

void loop() {
  const size_t capacity = JSON_OBJECT_SIZE(128);
  DynamicJsonDocument doc(capacity);
  
  DeserializationError error = deserializeJson(doc, Serial);
  if (error) {
    Serial.print(F("deserializeJson() failed: "));
    Serial.println(error.c_str());
    return;
  }
  
  //serializeJson(doc, Serial);

  temp1 = doc["temperature"].as<String>();
  pressure1 = doc["pressure"].as<String>();
  range1 = doc["range"];
  switch (range1){
    case 0:
      rain = "RAIN";
    break;
    case 1:
      rain = "RAIN WARNING;
    break;
    case 2:
      rain = "NO RAIN";
    break;
  }
  
  lux1 = doc["lux"];
  if(lux1 < 100){
    light = "NIGHT";
  }
  else if(lux1 > 100){
    light = "LIGHT";
  }

  server.on("/", HTTP_GET, [](AsyncWebServerRequest *request){
    request->send(SPIFFS, "/index.html", "text/html");
  });

  server.on("/assets/css/foundation.css", HTTP_GET, [](AsyncWebServerRequest *request){
    request->send(SPIFFS, "/assets/css/foundation.css", "text/css");
  });

  server.on("/assets/js/vendor.js", HTTP_GET, [](AsyncWebServerRequest *request){
    request->send(SPIFFS, "/assets/js/vendor.js", "text/js");
  });

  server.on("/assets/js/foundation.js", HTTP_GET, [](AsyncWebServerRequest *request){
    request->send(SPIFFS, "/assets/js/foundation.js", "text/js");
  });
  
  server.on("/temp1", HTTP_GET, [](AsyncWebServerRequest *request){
    request->send_P(200, "text/plain", temp1.c_str());
  });
  server.on("/pressure1", HTTP_GET, [](AsyncWebServerRequest *request){
    request->send_P(200, "text/plain", pressure1.c_str());
  });
  server.on("/rain", HTTP_GET, [](AsyncWebServerRequest *request){
    request->send_P(200, "text/plain", rain.c_str());
  });
  server.on("/light", HTTP_GET, [](AsyncWebServerRequest *request){
    request->send_P(200, "text/plain", light.c_str());
  });
  
  Serial.print("Recieved Temperature:  ");
  temp = doc["temperature"];
  Serial.println(temp);
  
  Serial.print("Recieved Pressure:  ");
  pressure = doc["pressure"];
  Serial.println(pressure);

  Serial.print("Recieved Range:  ");
  range = doc["range"];
  Serial.println(range);

  Serial.print("Recieved Lux:  ");
  lux = doc["lux"];
  Serial.println(lux);
  Serial.println("-----------------------------------------");

  server.begin();
  //end
}`
`
<!doctype html>
<html class="no-js" lang="en">
  <head>
    <meta charset="utf-8" />
    <meta http-equiv="x-ua-compatible" content="ie=edge">
    <meta name="viewport" content="width=device-width, initial-scale=1.0">
    <title>Welcome to ESP Server</title>
    <link rel="stylesheet" href="assets/css/foundation.css">
  </head>
<body>

    <div class="grid-container text-center">
      <div class="grid-x grid-padding-x">
        <div class="large-12 cell">
          <h1>Welcome to ESP Server</h1>
        </div>
      </div>

      <div class="grid-x grid-padding-x">
        <div class="large-12 cell">
          <div class="grid-x grid-padding-x">
            <div class="large-12 medium-12 small-12 cell">
              <div class="callout">
                <h3>TEMPERATURE</h3>
                <div class="grid-x grid-padding-x">
                  <div class="large-12 medium-12 small-12 cell">
                    <p>SUHU UDARA<br />
                      <span id="temp1">
                        %TEMP1%
                      </span>
                        &deg;C
                    </p>
                  </div>
                  <div class="large-12 medium-12 small-12 cell">
                    <p>TEKANAN UDARA<br />
                      <span id="pressure1">
                        %PRESSURE1%
                      </span>
                        hPa
                    </p>
                  </div>
                  <div class="large-12 medium-12 small-12 cell">
                    <p>CURAH HUJAN<br />
                      <span id="rain">
                        %RAIN%
                      </span>
                    </p>
                  </div>
                  <div class="large-12 medium-12 small-12 cell">
                    <p>KECERAHAN<br />
                      <span id="light">
                        %LIGHT%
                      </span>
                    </p>
                  </div>
                </div>
              </div>
            </div>
          </div>
        </div>
      </div>
    </div>

<script>
  setInterval(function () {
    var xhttp = new XMLHttpRequest();
    xhttp.onreadystatechange = function() {
      if (this.readyState == 4 && this.status == 200) {
        document.getElementById("temp1").innerHTML = this.responseText;
      }
    };
    xhttp.open("GET", "/temp1", true);
    xhttp.send();
  }, 500) ;
  setInterval(function () {
    var xhttp = new XMLHttpRequest();
    xhttp.onreadystatechange = function() {
      if (this.readyState == 4 && this.status == 200) {
        document.getElementById("pressure1").innerHTML = this.responseText;
      }
    };
    xhttp.open("GET", "/pressure1", true);
    xhttp.send();
  }, 500) ;
  setInterval(function () {
    var xhttp = new XMLHttpRequest();
    xhttp.onreadystatechange = function() {
      if (this.readyState == 4 && this.status == 200) {
        document.getElementById("rain").innerHTML = this.responseText;
      }
    };
    xhttp.open("GET", "/rain", true);
    xhttp.send();
  }, 500) ;
  setInterval(function () {
    var xhttp = new XMLHttpRequest();
    xhttp.onreadystatechange = function() {
      if (this.readyState == 4 && this.status == 200) {
        document.getElementById("light").innerHTML = this.responseText;
      }
    };
    xhttp.open("GET", "/light", true);
    xhttp.send();
  }, 500) ;
</script>
<script src="assets/js/vendor.js"></script>
<script src="assets/js/foundation.js"></script>
</body>
</html>`

bblanchon, 2023-09-20:
Hi @BimoSora2,

Please try the [ArduinoJson Assistant](https://arduinojson.org/troubleshooter/).

Regards,
Benoit

BimoSora2, 2023-09-20:
@bblanchon i see the error and don't know which one is wrong
Decoding stack results
````
0x4020af5c: operator new (unsigned int) at /home/gonit/.arduino15/packages/esp8266 /hardware/esp8266/3.1.2/cores/esp8266/abi.cpp line 44

8x40208604: AsyncWebServer::on (char const*, unsigned char, std::function) at

/home/gonit/Arduino/libraries/ESPAsyncWebServer/src/WebServer.cpp line 149

0x40214208: std::_Function_handler >::_M_manager(std::_Any_data &, const std::_Any_data &, std::_Manager_operation) at /home/gonit/.arduino15/packages /esp8266/tools/xtensa-lx106-elf-gcc/3.1.0-gcc10.3-e5f9fec/xtensa-1x106-elf/include /c++/10.3.0/bits/std_function.h line 268

8x402011f8: std:: Function_handler >::_M_invoke(const std::_Any_data &, AsyncWebServerRequest *&&) at /home/gonit/.arduino15/packages/esp8266/tools/xtensa- 1x106-elf-gcc/3.1.0-gcc10.3-e5f9fec/xtensa-1x106-elf/include/c++/18.3.0 /bits/std_function.h line 289

0x4020a38d: String::operator= (char const*) at /home/gonit/. arduino15/packages /esp8266/hardware/esp8266/3.1.2/cores/esp8266/WString.cpp line 319

8x482027b2: Loop() at /home/gonit/.arduino15/packages/esp8266/tools/xtensa-lx106- elf-gcc/3.1.0-gcc10.3-e5f9fec/xtensa-lx106-elf/include/c++/10.3.0 /bits/std_function.h line 303

0x402142d4: std:: Function_handler >::_M_manager(std::_Any_data &, const std::_Any_data &, std::_Manager_operation) at /home/gonit/.arduino15/packages /esp8266/tools/xtensa-lx106-elf-gcc/3.1.0-gcc10.3-e5f9fec/xtensa-lx106-elf/include /c++/10.3.0/bits/std_function.h line 268

0x402011a0: std::_Function_handler >::_M_invoke(const std::_Any_data &, AsyncWebServerRequest *&&) at /home/gonit/.arduino15/packages/esp8266/tools/xtensa- 1x106-elf-gcc/3.1.0-gcc10.3-e5f9fec/xtensa-lx106-elf/include/c++/10.3.0 /bits/std_function.h line 289

8x40202a77: ESP8266WiFiAPClass::softAPConfig(IPAddress, IPAddress, IPAddress) at /home/gonit/.arduino15/packages/esp8266/hardware/esp8266/3.1.2/libraries /ESP8266WiFi/src/ESP8266WiFiAP.cpp line 259

8x4820b2f8: Loop_wrapper() at /home/gonit/.arduino15/packages/esp8266/hardware /esp8266/3.1.2/cores/esp8266/core_esp8266_main.cpp line 258
````

bblanchon, 2023-09-22:
I don't see ArduinoJson in the stack trace; what makes you think the library is responsible for the crash?
Also, what did the troubleshooter say?

closed 2023-10-09 by bblanchon