bblanchon/ArduinoJson #1974: this constant expression has type "const char *" instead of the required "unsigned int" type

DuCLeK65t, 2023-09-19:
This is my code:

<pre>
const size_t capacity = JSON_OBJECT_SIZE(2) + JSON_OBJECT_SIZE(3);
DynamicJsonDocument jsonBuffer(capacity);

doc["datatype"] = "string";
</pre>

I do not understand why this cause
<pre>this constant expression has type "const char *" instead of the required "unsigned int" type</pre>

one the line
<pre>doc["datatype"] = "string";</pre>

Can anyone help me? Thank you!

bblanchon, 2023-09-20:
Hi @DuCLeK65t,

I think you mixed up the names `jsonBuffer` and `doc`.

Best regards,
Benoit

DuCLeK65t, 2023-09-20:
> Hi @DuCLeK65t,
> 
> I think you mixed up the names `jsonBuffer` and `doc`.
> 
> Best regards, Benoit

Oh thank you @bblanchon, my mistake
I've modifty it but actually it has the same error:
[image]

DuCLeK65t, 2023-09-20:
Oh i've understand, those code lines didn't work because i put it in *.h file

closed 2023-09-20 by DuCLeK65t