bblanchon/ArduinoJson #1975: Arduino Uno Wifi R4 - RTC Clock causing crashes

jimemo, 2023-09-22:
**Describe the bug** 
Arduino Uno Wifi R4 
ArduinoJSON crashes when also using the following RTC library: https://docs.arduino.cc/tutorials/uno-r4-wifi/rtc
This same code is working fine on other boards, EG ESP32 & Raspberry Pico/W

**Troubleshooter report**  
Here is the report generated by the [ArduinoJson Troubleshooter](https://arduinojson.org/v6/troubleshooter/):  
1. The program uses ArduinoJson 6
2. The issue happens at run time
3. The issue concerns deserialization
4. The program crashes
5. Program crashes inside `deserializeJson()`
6. Removing dangling pointers doesn't prevent the crash
7. Program doesn't use `StaticJsonDocument`
8. Reducing stack usage doesn't prevent the crash

**Environment**  
Here is the environment that I used:
* Microcontroller: Arduino UNO R4 Wifi
* Core/runtime: renesas-ra
* IDE: N/A

**Reproduction**  

Deserialization crashes when including the following two lines:
```
#include "RTC.h"
RTCTime currentTime;
```

*** Input JSON ***
{"device":{"name":"Neopixels","time":{"connect_freq":2,"epoch":1695368382,"display":"22\/09\/23 08:39:42"},"inputs":[{"uuid":"63f0d76da7034","name":"Neopixel-number","value":"105,13,888"},{"uuid":"64ea7a002aa6c","name":"Neopixel-bool","value":1},{"uuid":"64ea7a7b675a3","name":"Neopixel-double","value":"Test message"}]}}
```
String res_str;
  while (wificlient.available()) {
    char c = wificlient.read();
    res_str = res_str + (char)c;
  }
  if (res_str.length() > 0)
DynamicJsonDocument server_input(1024);
  DeserializationError json_error = deserializeJson(server_input, res_str);
}
```
**Program output**  
Firmware name: PROJECT_NAME, compiled on: Sep 20 2023
Fault on interrupt or bare metal(no OS) environment
===== Thread stack information =====
  addr: 20007dd8    data: 00000000
  addr: 20007ddc    data: 00000003
  addr: 20007de0    data: 00000001
  addr: 20007de4    data: 00000000
  addr: 20007de8    data: 20000048
  addr: 20007dec    data: 000000a9
  addr: 20007df0    data: 20002520
  addr: 20007df4    data: 00000001
  addr: 20007df8    data: 00000001
  addr: 20007dfc    data: 20002528
  addr: 20007e00    data: 00000001
  addr: 20007e04    data: 00000001
  addr: 20007e08    data: 20002528
  addr: 20007e0c    data: 00000001
  addr: 20007e10    data: 20003540
  addr: 20007e14    data: 20003688
  addr: 20007e18    data: 20003ba0
  addr: 20007e1c    data: 20003d40
  addr: 20007e20    data: 00000000
  addr: 20007e24    data: 00000000
  addr: 20007e28    data: 20003d30
  addr: 20007e2c    data: 20003d30
  addr: 20007e30    data: 20000420
  addr: 20007e34    data: ffffffff
  addr: 20007e38    data: 00000000
  addr: 20007e3c    data: 00000025
  addr: 20007e40    data: 0000000d
  addr: 20007e44    data: 0000001e
  addr: 20007e48    data: 00000005
  addr: 20007e4c    data: 0000007b
  addr: 20007e50    data: 00000003
  addr: 20007e54    data: 0000a75f
  addr: 20007e58    data: 00000001
  addr: 20007e5c    data: 0000001e
  addr: 20007e60    data: 00000205
  addr: 20007e64    data: 0000007b
  addr: 20007e68    data: 0000000d
  addr: 20007e6c    data: 00000025
  addr: 20007e70    data: 00000000
  addr: 20007e74    data: 00000103
  addr: 20007e78    data: 200003d0
  addr: 20007e7c    data: 00006bad
  addr: 20007e80    data: 200025d8
  addr: 20007e84    data: 00000105
  addr: 20007e88    data: 00000105
  addr: 20007e8c    data: 20002990
  addr: 20007e90    data: 000002a5
  addr: 20007e94    data: 000002a5
  addr: 20007e98    data: 5445472b
  addr: 20007e9c    data: 54415453
  addr: 20007ea0    data: 003a5355
  addr: 20007ea4    data: 20002c40
  addr: 20007ea8    data: 000002a5
  addr: 20007eac    data: 000002a5
  addr: 20007eb0    data: 200003d0
  addr: 20007eb4    data: 00000001
  addr: 20007eb8    data: 0000a500
  addr: 20007ebc    data: 00006c3d
  addr: 20007ec0    data: 20001000
  addr: 20007ec4    data: 00006c65
  addr: 20007ec8    data: 20000ffc
  addr: 20007ecc    data: 20007f40
  addr: 20007ed0    data: 00000000
  addr: 20007ed4    data: 0000aaa1
  addr: 20007ed8    data: 00015271
  addr: 20007edc    data: 40046f00
  addr: 20007ee0    data: 00000000
  addr: 20007ee4    data: 0000aae3
  addr: 20007ee8    data: 00015271
  addr: 20007eec    data: 00014623
  addr: 20007ef0    data: 00015271
  addr: 20007ef4    data: 0000eecf
  addr: 20007ef8    data: 0000eec5
  addr: 20007efc    data: 00002599
====================================
=================== Registers information ====================
  R0 : 00000001  R1 : 00000000  R2 : 00000006  R3 : 00000000
  R12: 0000000a  LR : 00005db5  PC : 000000c0  PSR: 01070000
==============================================================
Usage fault is caused by attempts to execute an undefined instruction
Show more call stack info by run: addr2line -e PROJECT_NAME.elf -a -f 000000c0 00005db4 0000a75e 00006bac 00006c3c 00006c64 0000aaa0 0000aae2 00014622 0000eece 0000eec4

bblanchon, 2023-09-24:
Hi @jimemo,

> Usage fault is caused by attempts to execute an undefined instruction

This sounds like a stack corruption to me.
I would suspect the RTC library; what makes you think ArduinoJson is at fault?

I don't have a UNO R4 to test your program.
I just ordered one from the Arduino store, it might take a few days to get there.

Best regards,
Benoit

jimemo, 2023-09-24:
Thanks for responding. I did some more testing and found that another one of my libraries is causing problems with the RTC clock. This sketch proves that the RTC clock and ArduinoJSON is working fine:
```
#include <Arduino.h>
#include <ArduinoJSON.h>
#include "RTC.h"

void setup() {
  Serial.begin(9600);
  delay(5000);
  Serial.println("START PROGRAM");
  RTC.begin();
  delay(500);
  RTCTime startTime(24, Month::SEPTEMBER, 2023, 22, 58, 00, DayOfWeek::WEDNESDAY, SaveLight::SAVING_TIME_ACTIVE);
  RTC.setTime(startTime);
}

void loop() {

  RTCTime currentTime;
  RTC.getTime(currentTime);

  Serial.print(currentTime.getDayOfMonth());
  Serial.print("/");
  Serial.print(Month2int(currentTime.getMonth()));
  Serial.print("/");
  Serial.print(currentTime.getYear());
  Serial.print(" - ");
  Serial.print(currentTime.getHour());
  Serial.print(":");
  Serial.print(currentTime.getMinutes());
  Serial.print(":");
  Serial.println(currentTime.getSeconds());

  String json_str = "{\"device\": {\"name\": \"Neopixels\",\"time\": {\"connect_freq\": 2,\"epoch\": 1695587321,\"display\": \"24\\/09\\/23 21:28:41\"},\"inputs\": [{\"uuid\": \"63f0d76da7034\",\"name\": \"Neopixel-number\",\"value\": \"55\"},{\"uuid\":\"64ea7a002aa6c\",\"name\": \"Neopixel-bool\",\"value\": \"55\" },{\"uuid\": \"64ea7a7b675a3\",\"name\": \"Neopixel-double\",\"value\": \"55\"}]}}";
  DynamicJsonDocument json_obj(1024);
  DeserializationError json_error = deserializeJson(json_obj, json_str );
  if (json_error) {
    Serial.print("deserializeJson() failed: ");
    Serial.println(json_error.f_str());
  }
  Serial.println(json_obj["device"]["name"].as<const char*>());
  Serial.println("END PROGRAM");
  while (1) {}
}
```

closed 2023-09-24 by jimemo

bblanchon, 2023-09-25:
Thanks for the update, @jimemo.
What was causing this crash?

jimemo, 2023-09-25:
@bblanchon it was my own setRTCTime() function. I had it as a bool and it wasn't returning true or false.

bblanchon, 2023-09-25:
You answered "Program crashes inside `deserializeJson()`" to the Troubleshooter.
How this error in `setRTCTime()` could affect `deserializeJson()`?

jimemo, 2023-09-25:
@bblanchon - this bug was raised in error. I hadn't put sufficient break
points and incorrectly assumed was caused by ArduinoJSON