bblanchon/ArduinoJson #1977: The microcontroller stops working during DynamicJsonDocument processing

SamShekhovtsov, 2023-09-30:
**Description**
I'm trying to use the ArduinoJson library to construct the JSON message. I need to construct and send multiple JSON messages. When building the first message, everything is fine, the message is composed and sent over the WiFi network. When trying to build the second JSON message the microcontroller just hangs and stops working, I have to reset the device, disconnect from the USB port and connect again. So the problem is that the first JSON message can be composed, but other subsequent JSON messages can't be buit, the device just stops working. I have tried to create instance of the JSON document is several ways:

DynamicJsonDocument firstJsonDoc(1024);
// construct the firstJsonDoc
DynamicJsonDocument secondJsonDoc(1024);
// the microcontroller crashes here

This is another way that I have tried to achieve the same thing:

DynamicJsonDocument* firstJsonDoc = new DynamicJsonDocument(1024);
// construct the firstJsonDoc
DynamicJsonDocument* secondJsonDoc = new DynamicJsonDocument(1024);
// the microcontroller crashes here

initially I was blaming on the insufficient memory, however after checking the memory capacity of the MKR device, this shouldn't be a problem, as the MKR WiFi 1010 has 32 KBytes of the SRAM memory and after allocating memory for all the global and local variables and other program - related things, at least 10000 bytes should remain for the application needs (dynamic program variables).

**Troubleshooter's report**
1. The program uses ArduinoJson 6
2. The issue happens at run time
3. The issue concerns serialization
4. Program crashes
5. Program uses `DynamicJsonDocument`
6. Program doesn't use `PROGMEM`
7. Program inserts string pointers into the `JsonDocument`
8. Replacing string pointers with literals doesn't fix the crash
9. Output type is `Stream`

**Environment**
* Microcontroller: MKR WiFi 1010
* IDE: Arduino IDE 2.1.1

**Reproduction code**
```c++
DynamicJsonDocument doc(2048);
doc["fields"] = "some data";
.... filling here a lot of data, the document is large....
mqttClient.beginMessage(anyTopic, (unsigned long)measureJson(doc));
serializeJson(doc, mqttClient);
mqttClient.endMessage();
```

**Remarks**
the code, that I'm using is straight - forward. I create a large JSON document and send it via network. I need to comose multiple large documents. From my point of view, it looks like the DynamicJsonDocument is using one allocated memory dump for all documents that I'mn trying to create.

mathieucarbou, 2023-09-30:
If you remove any mqtt calls and create the 2 documents back to back, does it work ? I suspect you have to let the mqtt loop run before the second one, to let the mqtt complete all related com regarding the first buffer. Just an hypothesis though.

bblanchon, 2023-10-01:
Hi @mathieucarbou,

Which MQTT library are you using?
Are you having this issue in the context of an MQTT callback, as @mathieucarbou implied?
Can you provide an [MCVE](https://stackoverflow.com/help/mcve)?

Best regards,
Benoit

SamShekhovtsov, 2023-10-01:
this issue is absolutely not related to the MQTT.  This is realted to the ArduinoJson library, it doesn't allocate a separate space in memory (as it should) for each separate document. From my understanding, about what is going on, the library uses a single shared memory space for all JSON document instances, which is a bug.

bblanchon, 2023-10-03:
How are we supposed to help you if you refuse to tell us anything about your program?

SamShekhovtsov, 2023-10-03:
I do not refuse to provide description of my program. Let me tell you:

The microcontroller program continuously sends JSON messages via MQTT to the server part, with few seconds interval. Each JSON document that is built or composed within the application is about 1500 bytes in size.
My device is: Arduino MKR WiFi 1010,  the board name is "Arduino® MKR WiFi 1010", the microcontroller name is "SAMD21 Cortex®-M0+ 32bit low power ARM MCU"

So far, I have tried to use the DynamicJsonDocument in the following way:

```c++
void sendJsonMessage() {
    DynamicJsonDocument doc(2048);
    doc["fields"] = "some data";
    .... filling here a lot of data, the document is large....
    mqttClient.beginMessage(anyTopic, (unsigned long)measureJson(doc));
    serializeJson(doc, mqttClient);
    mqttClient.endMessage();
}
```

this function is called systematically with some interval from the loop() system function.

the microcontroller device hangs and stops working after composing three or two JSON messages, it runs out of the available memory.

mathieucarbou, 2023-10-03:
> The microcontroller program continuously sends JSON messages via MQTT to the server part, with few seconds interval. Each JSON document that is built or composed within the application is about 1500 [...]

My suspicion is that you have too little memory on this device and you are probably looping too fast allocating more and more memory faster than it can be garbage-collected.

You have a small memory device so you should be careful with memory allocation. 

Here, I would either destroy the json doc or have it outside of the function, static, and be allocated once and reused by calling clear()

@bblanchon 

See:
- `JsonDocument::clear()`
- `JsonDocument::garbageCollect()`
- https://arduinojson.org/v6/how-to/reuse-a-json-document

(sorry I bumped randomly into your issue - @bblanchon will probably help more precisely 0 just wanted to throw some leads)

bblanchon, 2023-10-05:
@SamShekhovtsov, the code snippet you shared doesn't reproduce the issue.
Since you believe it's a bug in ArduinoJson, you should be able to reproduce it without the MQTT part, right?
I have an MKR1000, so if you share an [MCVE](https://stackoverflow.com/help/mcve), I'll be able to run it.

@mathieucarbou, there is no garbage collector in Arduino (at least not in this case)
The `DynamicJsonDocument` frees its buffer when it's destructed (i.e., when the function exists) so it cannot leak memory.

I suspect that @SamShekhovtsov's program leaks somewhere else and quickly runs out of RAM.
Also, we still don't know which MQTT library it uses; there could be a bug in it.

Tybeevdroovaart, 2023-10-15:
I moved my project from one laptop to another and had to install all the libraries. Because of this, I installed the latest version of ArduinoJson, and my ESP8266 got into a bootloop. I found out it had to do with the Json library. The latest version which seems to be working for me is 6.19.4. I am also using it in combination with MQTT and make use of the PubSubClient libarary by Nick O'Leary.

bblanchon, 2023-10-16:
@StrandmonYellow, just because it works with an older version of the library doesn't mean the new one is buggy; it could also be an undefined behavior (UB) that was previously silent.
Please provide an [MCVE](https://stackoverflow.com/help/mcve) so we can sort this out.