bblanchon/ArduinoJson #1978: ArduinoJSON unable to be compiled

dirkt68, 2023-10-03:
**Description**
Hello everyone,

I am currently using the single header file version of the library to try and do some basic serialization, and unfortunately I am having trouble with some errors coming up. I am assuming it is Visual Studio Code's fault, as the header file compiles fine on Programiz's Online Compiler. There are three spots marked as "errors" but which are valid syntax:

line 589 -> ```SizedRamString(const char* str, size_t sz) : str_(str), size_(sz)``` {} -> "expected identifier" (squiggle on the "s" in ```str_```)
line 527 -> ```ZeroTerminatedRamString(const char* str) : str_(str) {}``` -> same problem
line 888/889 -> ```FlashString(const __FlashStringHelper* str, size_t sz)
      : str_(reinterpret_cast<const char*>(str)), size_(sz) {}``` -> same problem

Thank you all for your help!

**Troubleshooter's report**
1. The program uses ArduinoJson 6
2. The issue happens at compile time
3. The error is not in the list

**Environment**
* Microcontroller: RAK4631
* Core/Framework: Latest offical Arduino plugin on Visual Studio Code, WisBlock RUI3 RAK4361 Core
* IDE: Visual Studio Code Version 1.82.3(user setting)

**Reproduction code**
```c++
#include <ArduinoJson.h>

void setup(){}

void loop(){}
```

**Remarks**
This also occurs in a blank project simply by including the header, so I do not believe it is code related. There are also a lot of undefined functions and other errors around the header file, but those three cause red squiggles. As a test, adding this header to a CLion PlatformIO project shows no errors whatsoever and compiles correctly.

bblanchon, 2023-10-04:
Hi @dirkt68,

Please share the complete compiler output.

Best regards,
Benoit

bblanchon, 2023-10-04:
I compiled `JsonGeneratorExample.ino` successfully with the following setup:

* Arduino IDE 2.2.1
* Core [RAK-nRF52-Arduino](https://github.com/RAKWireless/RAK-nRF52-Arduino) ([board manager url](https://raw.githubusercontent.com/RAKWireless/RAKwireless-Arduino-BSP-Index/main/package_rakwireless_index.json))
* Board WisBlock RAF4631 (FQBN: `rakwireless:nrf52:WisCoreRAK4631Board`)

I only had to add `#include <Adafruit_TinyUSB.h>` at the top of the file.

dirkt68, 2023-10-04:
Benoit,

Thank you for responding so quickly, I am attaching the compiler output from the most recent one, apologies for the length.
[compiler_output.txt](https://github.com/bblanchon/ArduinoJson/files/12803665/compiler_output.txt)

I am using the same version of the Arduino IDE, and I think there might be a difference between your test and my environment, as my FQBN is ```rak_rui:nrf52:WisCoreRAK4631Board```, not ```rakwireless:nrf52:WisCoreRAK4631Board```. RAK Wireless has two different methods of using their boards in Arduino, so maybe the rui version could be causing this issue? I tried compiling this file in a random ESP32 sketch and that worked fine on the same Arduino IDE.

Thank you for your time,
Dirk

bblanchon, 2023-10-05:
I could not install the RUI boards on the Arduino IDE.
I used the following boards url, but it didn't add any new boards to the IDE:
https://github.com/RAKWireless/RAKwireless-Arduino-BSP-Index/blob/main/package_rakwireless_index.json

bblanchon, 2023-10-05:
I was able to reproduce the issue by adding the following line at the top of the file:

```c++
#define str_(s) #s
```

I found this macro in:

* [RAK-STM32-RUI](https://github.com/RAKWireless/RAK-STM32-RUI/blob/cb8b6bd97b418c77f908e3a987b9f81d7c457733/cores/STM32WLE/component/service/mode/cli/atcmd_queue.h#L21)
* [RAK-nRF52-RUI](https://github.com/RAKWireless/RAK-nRF52-RUI/blob/00ff6b17a273a7e6483996bcf6093794ad6725b0/cores/nRF5/component/service/mode/cli/atcmd_queue.h#L21)
* [RAK-APOLLO3-RUI](https://github.com/RAKWireless/RAK-APOLLO3-RUI/blob/0e6e009193fb5eff9bb486cdf46e0a6272984e6f/cores/apollo3/component/service/mode/cli/atcmd_queue.h#L21C1-L21C1)

dirkt68, 2023-10-05:
Oh wow, would adding a check and #undef solve the issue if put in the header?

bblanchon, 2023-10-05:
As a workaround, you can `#undef`, but it's better to rename the macro.
I'll open issues on the corresponding repos.

dirkt68, 2023-10-05:
Benoit,

Thanks so much for your quick help with all of this, renaming all instance of ```str_``` in ```ArduinoJson.h``` to ```str_json_``` solves the problem for me. 

Sincerely,
Dirk

closed 2023-10-05 by dirkt68