bblanchon/ArduinoJson #1982: Strange JsonArray loop result.

zekageri, 2023-10-18:
Hi!

First of all, thank you for this amaazing library.
I can't stress enough how thankfull we are for you.

I have a strange problem.

Here is my method

```cpp
void ComponentHandler::turnOffAllLights(JsonDocument &idArray, int size, AsyncWebServerRequest *request){
    serializeJsonPretty(idArray, Serial);
    Serial.println();

    for (JsonObject idObj : idArray.as<JsonArray>()) {
        std::string lightID = idObj["id"].as<std::string>();
        Serial.printf("Searching for %s\n", lightID.c_str() );

        Component* light = getComponent( lightID );
        if( light ){
            Serial.printf("Found light: %s\n", lightID.c_str());
            idObj["state"] = light->setState(false);
        }
    }

    char* buffer;
    if( !hshSystem.alloc(buffer,size) ){
        request->send(500);
        return;
    }
    request->onDisconnect([buffer](){
        free(buffer);
    });
    
    serializeJsonPretty(idArray, Serial);

    if( serializeJson(idArray, buffer, size) > 0 ){
        request->send(200, "application/json", buffer);
    }else{
        request->send(500);
    }
}
```

The serial outputs:

```cpp
[
  {
    "state": true,
    "id": "Light_o_lnszq98t_1qw3682qaofa8asc8btbv"
  },
  {
    "state": true,
    "id": "Light_n_lnu94wka_e867xt80vvbnyqayi96nc"
  },
  {
    "state": true,
    "id": "Light_a_lnu94xdl_vtt6drws7fer4rdtb5cke"
  },
  {
    "state": true,
    "id": "Light_a_lnu94yc1_siu1h1yrf7mut17fho6jm"
  },
  {
    "state": true,
    "id": "Light_q_lnu986tv_ioi7ok0r17ugdw1bul3jc"
  }
]Searching for Light_o_lnszq98t_1qw3682qaofa8asc8btbv
Found light: Light_o_lnszq98t_1qw3682qaofa8asc8btbv
[Light] - (Tesztr) is OFF
Searching for null
Searching for null
Searching for null
Searching for null
[
  {
    "id: 1697614054\r\nevent: LightState\r\ndata: {\"id\":\"Light_o_lnszq98t_1qw3682qaofa8asc8btbv\",\"zoneid\":\"j_lnszpqgb_1xdsgv2kgv1yxkd6k59d5\",\"name\":\"Tesztr\",\"type\":\"Light\",\"state\":false}\r\n\r\n": true,
    "97614054\r\nevent: LightState\r\ndata: {\"id\":\"Light_o_lnszq98t_1qw3682qaofa8asc8btbv\",\"zoneid\":\"j_lnszpqgb_1xdsgv2kgv1yxkd6k59d5\",\"name\":\"Tesztr\",\"type\":\"Light\",\"state\":false}\r\n\r\n": "14054\r\nevent: LightState\r\ndata: {\"id\":",        
    "state": false
  },
  {
    "Light_o_lnszq98t_1qw3682qaofa8asc8btbv\",\"zoneid\":\"j_lnszpqgb_1xdsgv2kgv1yxkd6k59d5\",\"name\":\"Tesztr\",\"type\":\"Light\",\"state\":false}\r\n\r\n": true,
    "o_lnszq98t_1qw3682qaofa8asc8btbv\",\"zoneid\":\"j_lnszpqgb_1xdsgv2kgv1yxkd6k59d5\",\"name\":\"Tesztr\",\"type\":\"Light\",\"state\":false}\r\n\r\n": "nszq98t_1qw3682qaofa8asc8btbv\",\"zoneid"
  },
  {
    ":\"j_lnszpqgb_1xdsgv2kgv1yxkd6k59d5\",\"name\":\"Tesztr\",\"type\":\"Light\",\"state\":false}\r\n\r\n": true,
    "szpqgb_1xdsgv2kgv1yxkd6k59d5\",\"name\":\"Tesztr\",\"type\":\"Light\",\"state\":false}\r\n\r\n": "qgb_1xdsgv2kgv1yxkd6k59d5\",\"name\":\"Tes"
  },
  {
    "tr\",\"type\":\"Light\",\"state\":false}\r\n\r\n": true,
    "ype\":\"Light\",\"state\":false}\r\n\r\n": "\":\"Light\",\"state\":false}\r\n\r\n\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000"
  },
  {
    "�q�?s": true,
    "": "\u0000�?\u0000�?_lnu986tv_ioi7ok0r17ugdw1bul3jc"        
  }
]
```

What can be the problem here?
A the first serializeJsonPretty, the document looks good, as intended.
When i start to loop over it, the first object in the array is also good.
After that, everything goes wrong.

closed 2023-10-18 by zekageri

bblanchon, 2023-10-18:
Hi @zekageri,

I see you closed this issue, so I assume you found the problem.
Please let me know if there is something I can add to the ArduinoJson Troubleshooter that could have helped you find the solution faster.

Best regards,
Benoit

zekageri, 2023-10-18:
Thank you for the response. Unfortunately the problem does not solved. Strange because i use similar loops all over my sketch and none have this problem. I ended up writing it in a totally different way.