bblanchon/ArduinoJson #1985: Read ArduinoJson in String pointer

saman-taghipour, 2023-10-21:
Hi
I want to read .txt file and assign its value in String pointer

String *strVar;
. . . . .
. . . . .
. . . . .
*strVar = json1["Name"] .as<String>();

when I compiling the .ino file, it complies successfully, but ESP8266 resets at this point.

bblanchon, 2023-10-23:
Hi @saman-taghipour,

The pointer `strVar` is dangling.
If you absolutely want to use a pointer, you must initialize it.
For example you could do:

```c++
String* strVar = new String()
```

However, I recommend that you get rid of the pointer and use a `String` value or reference:

```c++
String strVar = json1["Name"] .as<String>(); 
```

Best regards,
Benoit

saman-taghipour, 2023-10-23:
Thank you @bblanchon 
I used String, but I noticed that the String variable uses heap
when I used library for sending email, ESP12 crashed. I think it depends on heap memory.

saman-taghipour, 2023-10-24:
Dear @bblanchon 
I used this:
`String* strVar = new String()`
for all of my String variables, but still I have problem with email sender
It really reduce heap usage? or not different with String?

bblanchon, 2023-10-26:
`String* strVar = new String()` uses more heap than `String strVar` (and is an abomination, if you ask me)

closed 2023-11-02 by bblanchon