bblanchon/ArduinoJson #1986: uncomplete json

Dome2075, 2023-10-22:
**Description**
I like output like this 
"{"header":["180B1254A963","ABCDA8F8ABCD"],"1":["device_query",],"2":["opened","true"],"3":["trigger,"4":["voltage","NaN"]}"
but get this
"{"header":["180B1254A963","ABCDA8F8ABCD"],"1":["device_query",],"2":["opened","true"],"3":["t,"4":["voltage","NaN"]}"

the key "trigger" ist not complete only "t,

**Troubleshooter's report**
1. The program uses ArduinoJson 6
2. The issue happens at run time
3. The issue concerns serialization
4. Output is incomplete
5. Program doesn't call `serializeJson()` in a loop
6. `JsonDocument::overflowed()` returns `false`
7. No string contains a NUL

**Environment**
* Microcontroller: Arduino Pro or Pro Mini ATmega328 (3.3V, 8 MHz)
* Core/Framework: Atmel AVR (4.0.0) >
* IDE: PlatformIO

**Reproduction code**
```c++
void LoRa_send(){
StaticJsonDocument<384> sjsons;
  JsonArray data;
  String output;
  

  data = sjsons.createNestedArray("header");
  data.add(String(SERVERID));data.add(String(DEVICEID));

  for (int i = 0; LoraData_callback[i][0] != ""; i++){
    data = sjsons.createNestedArray(String(i+1));
    data.add(LoraData_callback[i][0]);data.add(LoraData_callback[i][1]);
    Serial.println(LoraData_callback[i][0] + " : " + LoraData_callback[i][1]);

    LoraData_callback[i][0] = "";LoraData_callback[i][1] = ""; // clear array
  }
 
  serializeJson(sjsons, output);
    
  //Lora send
  LoRa_sendMessage(output); // send a message
}
```

**Remarks**
in this void i serialize my states to send with Lora. (fail) the key "trigger" ist not complete only "t,
in sevond void i deserialize incoming (no fail)

bblanchon, 2023-10-25:
Hi @Dome2075,

I suspect that the strings are not copied as you expect.
Unlike `add(String)`, `add(const char*)` stores the string by address.
Please try to cast to pass `String`s instead:

```c++
data.add(String(LoraData_callback[i][0]));
data.add(String(LoraData_callback[i][1]));
```

Best regards,
Benoit

Dome2075, 2023-10-28:
Hello,

`LoraData_callback[i][0]` is already a String

i have random fails like this 

`{"header":["180B1254A963","ABCDA8F8ABCD"],"1":["device_query","],"2":["opened","true"],"3":["trigger","true"],"4":["vol,"NaN"]}
or
{"header":["180B1254A963","ABCDA8F8ABCD"],"1":["device_query","true"],"2":["opened","true"],"3":["t,"true"],"4":["vol,"NaN"]}`

at the moment i used size `StaticJsonDocument<300>` and seams to be ok

but i down't understand why its random

Dome2075, 2023-10-28:
Hello,

i have found my problem.
i get out of memmory.

I save incoming Lorastream as String an copy it.
So a bigger tream is incomming and the serializeJson also set a string an then run out of memmory.

so its my fault.

Thanks for helping.

mfg 

Dominik

closed 2023-10-28 by Dome2075

bblanchon, 2023-10-29:
You're welcome, Dominik!
Thank you for using ArduinoJson.
Don't forget to [cast a star](https://github.com/bblanchon/ArduinoJson/stargazers) to support the project :wink: