bblanchon/ArduinoJson #1987: ESP32 crash after serialization into Stream

flofricke, 2023-10-25:
**Describe the issue**  
Hi, I'm trying to access an SD card and convert the contents of a given directory into a JSON structure in an HTTP (via ESPAsyncWebServer) request. This works sometimes, very rarely. Most of the time the ESP32 crashes. Creating works but serialising to the stream crashes the program. Above a certain size, however. And as I said, it also goes through sometimes. Unfortunately not very deterministic. I can't do anything with the backtrace, maybe someone can help?

**Troubleshooter report**  
Here is the report generated by the [ArduinoJson Troubleshooter](https://arduinojson.org/v6/troubleshooter/):  
1. The program uses ArduinoJson 6
2. The issue happens at run time
3. The issue concerns serialization
4. Program crashes
5. Program uses `DynamicJsonDocument`
6. Program doesn't use `PROGMEM`
7. Program doesn't insert string pointers into the `JsonDocument`
8. Output type is `Stream`

**Environment**  
Here is the environment that I'm using':
* Microconroller: ESP32-WROOM-32E
* Core/runtime: arduino-esp32 (2.0.14)
* IDE: vscode / platformio

**Reproduction**  
Here is a small snippet that demonstrate the problem.               

```
AsyncResponseStream *response = request->beginResponseStream("application/json");

                DynamicJsonDocument root(24000);
                // StaticJsonDocument<32000> root;

                if (url.length() > 1)
                {

                  root["url"] = url + "/";
                }
                else
                {

                  root["url"] = url;
                }

                JsonArray filesystem = root.createNestedArray("fs");

                File file = DirHandle.openNextFile();

                while (file)
                {

                  JsonObject fs_entry = filesystem.createNestedObject();

                  // Serial.println(file.name());

                  fs_entry["name"] = String(file.name());

                  if (file.isDirectory())
                  {

                    fs_entry["type"] = "d";
                  }
                  else
                  {

                    fs_entry["type"] = "f";
                    fs_entry["size"] = String(file.size());
                  }

                  file = DirHandle.openNextFile();
                }

                serializeJson(root, *response);

[... 29 lines cut ...]
      (inlined by) ArduinoJson::V6213PB2::detail::JsonSerializer<ArduinoJson::V6213PB2::detail::Writer<AsyncResponseStream, void> >::result_type ArduinoJson::V6213PB2::detail::VariantData::accept<ArduinoJson::V6213PB2::detail::JsonSerializer<ArduinoJson::V6213PB2::detail::Writer<AsyncResponseStream, void> > >(ArduinoJson::V6213PB2::detail::JsonSerializer<ArduinoJson::V6213PB2::detail::Writer<AsyncResponseStream, void> >&) const at .pio/libdeps/esp32-esp32-wroom-32e/ArduinoJson/src/ArduinoJson/Variant/VariantData.hpp:43
  #14 0x400d4756:0x3ffe81a0 in ArduinoJson::V6213PB2::detail::JsonSerializer<ArduinoJson::V6213PB2::detail::Writer<AsyncResponseStream, void> >::visitObject(ArduinoJson::V6213PB2::detail::CollectionData const&) at .pio/libdeps/esp32-esp32-wroom-32e/ArduinoJson/src/ArduinoJson/Json/JsonSerializer.hpp:48
      (inlined by) ArduinoJson::V6213PB2::detail::JsonSerializer<ArduinoJson::V6213PB2::detail::Writer<AsyncResponseStream, void> >::result_type ArduinoJson::V6213PB2::detail::VariantData::accept<ArduinoJson::V6213PB2::detail::JsonSerializer<ArduinoJson::V6213PB2::detail::Writer<AsyncResponseStream, void> > >(ArduinoJson::V6213PB2::detail::JsonSerializer<ArduinoJson::V6213PB2::detail::Writer<AsyncResponseStream, void> >&) const at .pio/libdeps/esp32-esp32-wroom-32e/ArduinoJson/src/ArduinoJson/Variant/VariantData.hpp:38
  #15 0x400d4679:0x3ffe8200 in ArduinoJson::V6213PB2::detail::JsonSerializer<ArduinoJson::V6213PB2::detail::Writer<AsyncResponseStream, void> >::visitArray(ArduinoJson::V6213PB2::detail::CollectionData const&) at .pio/libdeps/esp32-esp32-wroom-32e/ArduinoJson/src/ArduinoJson/Json/JsonSerializer.hpp:27
      (inlined by) ArduinoJson::V6213PB2::detail::JsonSerializer<ArduinoJson::V6213PB2::detail::Writer<AsyncResponseStream, void> >::result_type ArduinoJson::V6213PB2::detail::VariantData::accept<ArduinoJson::V6213PB2::detail::JsonSerializer<ArduinoJson::V6213PB2::detail::Writer<AsyncResponseStream, void> > >(ArduinoJson::V6213PB2::detail::JsonSerializer<ArduinoJson::V6213PB2::detail::Writer<AsyncResponseStream, void> >&) const at .pio/libdeps/esp32-esp32-wroom-32e/ArduinoJson/src/ArduinoJson/Variant/VariantData.hpp:35
  #16 0x400d4756:0x3ffe8260 in ArduinoJson::V6213PB2::detail::JsonSerializer<ArduinoJson::V6213PB2::detail::Writer<AsyncResponseStream, void> >::visitObject(ArduinoJson::V6213PB2::detail::CollectionData const&) at .pio/libdeps/esp32-esp32-wroom-32e/ArduinoJson/src/ArduinoJson/Json/JsonSerializer.hpp:48
      (inlined by) ArduinoJson::V6213PB2::detail::JsonSerializer<ArduinoJson::V6213PB2::detail::Writer<AsyncResponseStream, void> >::result_type ArduinoJson::V6213PB2::detail::VariantData::accept<ArduinoJson::V6213PB2::detail::JsonSerializer<ArduinoJson::V6213PB2::detail::Writer<AsyncResponseStream, void> > >(ArduinoJson::V6213PB2::detail::JsonSerializer<ArduinoJson::V6213PB2::detail::Writer<AsyncResponseStream, void> >&) const at .pio/libdeps/esp32-esp32-wroom-32e/ArduinoJson/src/ArduinoJson/Variant/VariantData.hpp:38
  #17 0x400d499f:0x3ffe82c0 in ArduinoJson::V6213PB2::detail::JsonSerializer<ArduinoJson::V6213PB2::detail::Writer<AsyncResponseStream, void> >::result_type ArduinoJson::V6213PB2::detail::variantAccept<ArduinoJson::V6213PB2::detail::JsonSerializer<ArduinoJson::V6213PB2::detail::Writer<AsyncResponseStream, void> > >(ArduinoJson::V6213PB2::detail::VariantData const*, ArduinoJson::V6213PB2::detail::JsonSerializer<ArduinoJson::V6213PB2::detail::Writer<AsyncResponseStream, void> >&) at .pio/libdeps/esp32-esp32-wroom-32e/ArduinoJson/src/ArduinoJson/Variant/VariantFunctions.hpp:18
      (inlined by) unsigned int ArduinoJson::V6213PB2::detail::doSerialize<ArduinoJson::V6213PB2::detail::JsonSerializer, ArduinoJson::V6213PB2::detail::Writer<AsyncResponseStream, void> >(ArduinoJson::V6213PB2::JsonVariantConst, ArduinoJson::V6213PB2::detail::Writer<AsyncResponseStream, void>) at .pio/libdeps/esp32-esp32-wroom-32e/ArduinoJson/src/ArduinoJson/Serialization/serialize.hpp:15
  #18 0x400d49c5:0x3ffe82f0 in unsigned int ArduinoJson::V6213PB2::detail::serialize<ArduinoJson::V6213PB2::detail::JsonSerializer, AsyncResponseStream>(ArduinoJson::V6213PB2::JsonVariantConst, AsyncResponseStream&) at .pio/libdeps/esp32-esp32-wroom-32e/ArduinoJson/src/ArduinoJson/Serialization/serialize.hpp:22
  #19 0x400d53bf:0x3ffe8310 in unsigned int ArduinoJson::V6213PB2::serializeJson<AsyncResponseStream>(ArduinoJson::V6213PB2::JsonVariantConst, AsyncResponseStream&) at .pio/libdeps/esp32-esp32-wroom-32e/ArduinoJson/src/ArduinoJson/Json/JsonSerializer.hpp:127
      (inlined by) operator() at src/main.cpp:839
  #20 0x400d5521:0x3ffe8430 in std::_Function_handler<void (AsyncWebServerRequest*), setup()::{lambda(AsyncWebServerRequest*)#5}>::_M_invoke(std::_Any_data const&, AsyncWebServerRequest*&&) at c:\users\azriel\.platformio\packages\toolchain-xtensa-esp32\xtensa-esp32-elf\include\c++\8.4.0\bits/std_function.h:297
  #21 0x400e0443:0x3ffe8450 in std::function<void (AsyncWebServerRequest*)>::operator()(AsyncWebServerRequest*) const at c:\users\azriel\.platformio\packages\toolchain-xtensa-esp32\xtensa-esp32-elf\include\c++\8.4.0\bits/std_function.h:687
  #22 0x400e047e:0x3ffe8480 in AsyncCallbackWebHandler::handleRequest(AsyncWebServerRequest*) at .pio/libdeps/esp32-esp32-wroom-32e/ESPAsyncWebServer-esphome/src/WebHandlerImpl.h:132
  #23 0x400dd235:0x3ffe84d0 in AsyncWebServerRequest::_parseLine() at .pio/libdeps/esp32-esp32-wroom-32e/ESPAsyncWebServer-esphome/src/WebRequest.cpp:581 (discriminator 1)
  #24 0x400dd2f3:0x3ffe8520 in AsyncWebServerRequest::_onData(void*, unsigned int) at .pio/libdeps/esp32-esp32-wroom-32e/ESPAsyncWebServer-esphome/src/WebRequest.cpp:123
  #25 0x400dd5a9:0x3ffe8580 in std::_Function_handler<void (void*, AsyncClient*, void*, unsigned int), AsyncWebServerRequest::AsyncWebServerRequest(AsyncWebServer*, AsyncClient*)::{lambda(void*, AsyncClient*, void*, unsigned int)#8}>::_M_invoke(std::_Any_data const&, void*&&, AsyncClient*&&, std::_Any_data const&, unsigned int&&) at .pio/libdeps/esp32-esp32-wroom-32e/ESPAsyncWebServer-esphome/src/WebRequest.cpp:76
[... cut ...]
  #27 0x401966f2:0x3ffe85d0 in AsyncClient::_recv(tcp_pcb*, pbuf*, signed char) at .pio/libdeps/esp32-esp32-wroom-32e/AsyncTCP-esphome/src/AsyncTCP.cpp:961
  #28 0x40196735:0x3ffe85f0 in AsyncClient::_s_recv(void*, tcp_pcb*, pbuf*, signed char) at .pio/libdeps/esp32-esp32-wroom-32e/AsyncTCP-esphome/src/AsyncTCP.cpp:1253
  #29 0x4019690e:0x3ffe8610 in _handle_async_event(lwip_event_packet_t*) at .pio/libdeps/esp32-esp32-wroom-32e/AsyncTCP-esphome/src/AsyncTCP.cpp:164
  #30 0x4019698b:0x3ffe8630 in _async_service_task(void*) at .pio/libdeps/esp32-esp32-wroom-32e/AsyncTCP-esphome/src/AsyncTCP.cpp:199
```

bblanchon, 2023-10-26:
Hi @flofricke,

I don't see anything wrong regarding your use of ArduinoJson.
Looking a the stack trace, we can see that `std::terminate` is called when the program tries to allocate memory for a buffer, which suggests you're running out of memory.

Indeed, your `JsonDocument` is already really big, and would get even bigger if there are more files. I suggest you send one JSON object for each file one by one, so as to decouple the size of the `JsonDocument` from the number of files.

You could keep the original format by sending `{"fs":[` at the beginning of the response, and `]}`at the end, but you could also decide to use a JSONLines/NDJSON format.

Best regards,
Benoit

flofricke, 2023-10-26:
Hi @bblanchon, thanks for you answer. I have stopped everything else to get my biggest directory. It is working reliably now. So it is a memory problem. Does ArduinoJson uses PSRAM? With another devboard (esp32-s3, 8mb psram) i had absolutly never a problem with that amount of data.

Keep on the good work :-)

Kind regards,
Florian

bblanchon, 2023-10-27:
Hi Florian,

ArduinoJson can use the PSRAM explicitly or implicitly, depending on your configuration.
See: [How to use external RAM on ESP32?](https://arduinojson.org/v6/how-to/use-external-ram-on-esp32/)

Best regards,
Benoit

closed 2023-10-27 by flofricke