bblanchon/ArduinoJson #1992: FQBN: esp8266:esp8266:generic:xtal=160

alyaparan, 2023-11-09:
FQBN: esp8266:esp8266:generic:xtal=160
Using board 'generic' from platform in folder: C:\Users\alikp\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.2
Using core 'esp8266' from platform in folder: C:\Users\alikp\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.2

"C:\\Users\\alikp\\AppData\\Local\\Arduino15\\packages\\esp8266\\tools\\python3\\3.7.2-post1/python3" -I "C:\\Users\\alikp\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\3.1.2/tools/mkbuildoptglobals.py" "C:\\Users\\alikp\\AppData\\Local\\Programs\\Arduino IDE\\resources\\app\\lib\\backend\\resources" 10607 "C:\\Users\\alikp\\AppData\\Local\\Temp\\arduino\\sketches\\908D72A54E8908A5F6D654D252D7F653" "C:\\Users\\alikp\\AppData\\Local\\Temp\\arduino\\sketches\\908D72A54E8908A5F6D654D252D7F653/core/build.opt" "C:\\Users\\alikp\\OneDrive\\Документы\\GitHub\\duino-coin\\ESP8266_Code/ESP8266_Code.ino.globals.h" "C:\\Users\\alikp\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\3.1.2\\cores\\esp8266/CommonHFile.h"
default_encoding:       cp1252
Assume aggressive 'core.a' caching enabled.
Detecting libraries used...
C:\Users\alikp\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\3.1.0-gcc10.3-e5f9fec/bin/xtensa-lx106-elf-g++ -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ -D_GNU_SOURCE -DESP8266 @C:\Users\alikp\AppData\Local\Temp\arduino\sketches\908D72A54E8908A5F6D654D252D7F653/core/build.opt -IC:\Users\alikp\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.2/tools/sdk/include -IC:\Users\alikp\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.2/tools/sdk/lwip2/include -IC:\Users\alikp\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.2/tools/sdk/libc/xtensa-lx106-elf/include -IC:\Users\alikp\AppData\Local\Temp\arduino\sketches\908D72A54E8908A5F6D654D252D7F653/core -c @C:\Users\alikp\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.2/tools/warnings/none-cppflags -Os -g -free -fipa-pta -Werror=return-type -mlongcalls -mtext-section-literals -fno-rtti -falign-functions=4 -std=gnu++17 -ffunction-sections -fdata-sections -fno-exceptions -DMMU_IRAM_SIZE=0x8000 -DMMU_ICACHE_SIZE=0x8000 -w -x c++ -E -CC -DNONOSDK22x_190703=1 -DF_CPU=160000000L -DLWIP_OPEN_SRC -DTCP_MSS=536 -DLWIP_FEATURES=1 -DLWIP_IPV6=0 -DARDUINO=10607 -DARDUINO_ESP8266_GENERIC -DARDUINO_ARCH_ESP8266 -DARDUINO_BOARD="ESP8266_GENERIC" -DARDUINO_BOARD_ID="generic" -DLED_BUILTIN=2 -DFLASHMODE_DOUT -IC:\Users\alikp\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.2\cores\esp8266 -IC:\Users\alikp\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.2\variants\generic C:\Users\alikp\AppData\Local\Temp\arduino\sketches\908D72A54E8908A5F6D654D252D7F653\sketch\ESP8266_Code.ino.cpp -o nul
Alternatives for ArduinoJson.h: [ArduinoJson@6.21.3]
ResolveLibrary(ArduinoJson.h)
  -> candidates: [ArduinoJson@6.21.3]
C:\Users\alikp\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\3.1.0-gcc10.3-e5f9fec/bin/xtensa-lx106-elf-g++ -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ -D_GNU_SOURCE -DESP8266 @C:\Users\alikp\AppData\Local\Temp\arduino\sketches\908D72A54E8908A5F6D654D252D7F653/core/build.opt -IC:\Users\alikp\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.2/tools/sdk/include -IC:\Users\alikp\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.2/tools/sdk/lwip2/include -IC:\Users\alikp\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.2/tools/sdk/libc/xtensa-lx106-elf/include -IC:\Users\alikp\AppData\Local\Temp\arduino\sketches\908D72A54E8908A5F6D654D252D7F653/core -c @C:\Users\alikp\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.2/tools/warnings/none-cppflags -Os -g -free -fipa-pta -Werror=return-type -mlongcalls -mtext-section-literals -fno-rtti -falign-functions=4 -std=gnu++17 -ffunction-sections -fdata-sections -fno-exceptions -DMMU_IRAM_SIZE=0x8000 -DMMU_ICACHE_SIZE=0x8000 -w -x c++ -E -CC -DNONOSDK22x_190703=1 -DF_CPU=160000000L -DLWIP_OPEN_SRC -DTCP_MSS=536 -DLWIP_FEATURES=1 -DLWIP_IPV6=0 -DARDUINO=10607 -DARDUINO_ESP8266_GENERIC -DARDUINO_ARCH_ESP8266 -DARDUINO_BOARD="ESP8266_GENERIC" -DARDUINO_BOARD_ID="generic" -DLED_BUILTIN=2 -DFLASHMODE_DOUT -IC:\Users\alikp\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.2\cores\esp8266 -IC:\Users\alikp\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.2\variants\generic -Ic:\Users\alikp\OneDrive\Документы\Arduino\libraries\ArduinoJson\src C:\Users\alikp\AppData\Local\Temp\arduino\sketches\908D72A54E8908A5F6D654D252D7F653\sketch\ESP8266_Code.ino.cpp -o nul
C:\Users\alikp\OneDrive\Документы\GitHub\duino-coin\ESP8266_Code\ESP8266_Code.ino:28:10: fatal error: ArduinoJson.h: No such file or directory
compilation terminated.

Alternatives for ArduinoJson.h: [ArduinoJson@6.21.3]
ResolveLibrary(ArduinoJson.h)
  -> candidates: [ArduinoJson@6.21.3]
Using library ArduinoJson at version 6.21.3 in folder: C:\Users\alikp\OneDrive\Документы\Arduino\libraries\ArduinoJson 
exit status 1

Compilation error: ArduinoJson.h: No such file or directory

bblanchon, 2023-11-10:
Hi @alyaparan,

> ArduinoJson.h: No such file or directory

It looks like the library isn't installed correctly.
Can you try to reinstall it from the Library Manager?

Best regards,
Benoit

alyaparan, 2023-11-10:
> Hi @alyaparan,
> 
> > ArduinoJson.h: No such file or directory
> 
> It looks like the library isn't installed correctly. Can you try to reinstall it from the Library Manager?
> 
> Best regards, Benoit

Imma tried to install old versions of ArduinoJson, imma installed ArduinoJson from ZIP file, it's not worked. Same issues showed anytime when I probe all of things.

bblanchon, 2023-11-13:
Can you remove and reinstall the library using the Library Manager (the bookshelf tab in the IDE)?

closed 2023-11-28 by bblanchon