bblanchon/ArduinoJson #2000: Exception or heap memory corruption when deserializing a large (e.g., 900k) JSON string

bergdevCopeland, 2023-11-22:
**Description**
When I call deserializeJson(), passing a null-terminated 904,624 byte null-terminated string, an exception is thrown. The exception is from ArduinoJson code trying to read an invalid memory location, typically a string variable(?).

The stack is likely being corrupted as the call-stack in my IDE is inconsistent and often not available. I have reason to suspect that heap memory may be getting corrupted as well.

This is a really big JSON string. ArduinoJson Assistant indicates at least 3145728 (~3MB) to be passed to the StaticJsonDocument declaration. Whether I use that number or a much larger number to the StaticJsonDocument, it still throws an exception.

I can't define this as a stack variable (stack overflow exception). So, I'm going with a global for now in hopes of learning more.

I took a subset of the problematic code, only changing how the JSON input is read (from a file as opposed to production code reading from an HTTP transaction) and set up a test, passing the same exact input. The JSON is valid as confirmed by Arduino's website and others. It contains no PII or other sensitive information.

Likely because it is called from higher up in the stack, the test doesn't throw an exception. However, the memory holding JSON string input (second arg to the deserializeJson call) is altered after returning from the deserializeJson call. This buffer for that input string is allocated on the heap by a call to new char[nSize]. The file contents were built using the HTTP response payload.

Yesterday, I purchased the book, ArduinoJson.epub. But, it talks a lot about JsonBuffer and not about the newer version 6 JsonDocument. I may be missing a fundamental thing here. I have read the migration notes on the website.

Finally, our work requires repeated use of a JsonDocument for hundreds of requests. It does look like a call to deserializeJson "clears the JSON document (per the migration webpage). Is this equivalent to making a call to JsonDocument.clear()? I have tried adding that, too and subsequent calls to memoryUsage() confirm this.

**Troubleshooter's report**
1. The program uses ArduinoJson 6
2. The issue happens at run time
3. The issue concerns deserialization
4. The program crashes
5. Program crashes inside `deserializeJson()`
6. Removing dangling pointers doesn't prevent the crash
7. Program doesn't use `StaticJsonDocument`
8. Reducing stack usage doesn't prevent the crash

**Environment**
* Microcontroller: x64
* Core/Framework: ArduinoJson-v6.21.3.h
* IDE: Visual Studio 2019, Visual C++ 

**Reproduction code**
```c++
StaticJsonDocument<8192000>   docTest; // ArduinoJson Assistant says 3145728 for this particular JSON

void testFunc()
{
  static const char* c_pszKeyStatus = "status";
  static const char* c_pszKeyStatusDetail = "statusDetail";
  static const char* c_pszKeyPoints = "points";
  static const char* c_pszKeyErr = "err";
  static const char* c_pszKeyError = "ERROR";

  bool              bResult = true;
  JsonObjectConst   sRoot;
  char*             pszJson = NULL;
  CFile             fileJson;
  UINT              nFileSize = 0;
  UINT              nMemoryUsed = 0;
  CString           strTemp;
  CString           strError;

  if (fileJson.Open("C:\\LCB\\TestJson.json", CFile::modeRead ))
  {
	 nFileSize = fileJson.GetLength();
	 pszJson = new char[nFileSize + 1];
	 memset(pszJson, 0, nFileSize + 1);
	 fileJson.Read(pszJson, nFileSize);

	 fileJson.Close();
  }

  // the contents (not the variable's address value) are altered after the next call
  DeserializationError error = deserializeJson(docTest, pszJson);
  nMemoryUsed = docTest.memoryUsage();
  if (error)
  {
	 strTemp.Format("JSON DeSerialize error: %s", error.c_str());
  }
  else
  {
	 if (sRoot.isNull())
	 {
		sRoot = docTest.as<JsonObject>();
	 }

	 if (sRoot.containsKey(c_pszKeyStatus))
	 {
		if (sRoot[c_pszKeyStatus] == c_pszKeyError)
		{
		   bResult = FALSE;
		   if (sRoot.containsKey(c_pszKeyStatusDetail))
		   {
			  const char* pszStatusDetail = sRoot[c_pszKeyStatusDetail];
			  strError = pszStatusDetail;
		   }
		   else
		   {
			  strError.Format("Unknown error: Missing error details key [%s]", c_pszKeyStatusDetail);
		   }
		}
	 }
  }
  docTest.clear();
}

```

**Remarks**
Apologies for the long code. The test JSON can be found at http://bergdev.com/TestJson.json.

Thank you for your consideration.

bblanchon, 2023-11-23:
Hi @bergdevCopeland,

Congratulations on getting issue number 2000 🎉 

I could not remember how to set up Visual Studio to use MFC classes, so I rewrote your program with STL classes:

```c++
#include <iostream>
#include <sstream>
#include <fstream>
#include "ArduinoJson-v6.21.3.h"

StaticJsonDocument<8192000>   docTest;

int main()
{
  std::ifstream file("TestJson.json");
  std::stringstream buffer;
  buffer << file.rdbuf();
  deserializeJson(docTest, buffer.str());
  serializeJsonPretty(docTest, std::cout);
}
```

The program works fine, which confirms there is nothing wrong with `deserializeJson()`.

I suspect the issue is that your program uses the zero-copy mode and, therefore, stores pointers to the input buffer instead of copying the strings. To disable the zero-copy mode, you must pass a `const char*` instead of a `char*`.

> Yesterday, I purchased the book, ArduinoJson.epub. But, it talks a lot about JsonBuffer and not about the newer version 6 JsonDocument.

Thank you very much for purchasing my book ❤️ 
You'll find two epubs in the download links: `arduinojson5.epub` and `arduinojson6.epub`.
The former is for ArduinoJson 5 (`JsonBuffer`) and the latter for ArduinoJson 6 (`JsonDocument`)

Best regards,
Benoit

bergdevCopeland, 2023-11-24:
Thank you very much, Benoit!

Would zero-copy mode result in a lot of stack usage for storing the pointers?

I will try both the const (disable zero-copy mode) and streaming approach.

Again, thanks!

- Lee

Hi @bergdevCopeland [github.com]<https://urldefense.com/v3/__https:/github.com/bergdevCopeland__;!!OkyjEqkqqe9UhNQ!wXfQukjYvXqL6amAoNnB_5lhtWjySR_pVQoch0Qs4DsosnXK6XPjk0JD85mGwhMhrMQy_-CSGv_x-JPsr--njoVTXA$>,

bergdevCopeland, 2023-11-27:
Hi Benoit.

I didn't realize it was a "Y2K" issue - LOL.

Before you start reading this, I would like to thank you for your time and patience dealing with my issues. As you know, this isn't exactly a small embedded application. So, if you don't think it's worth chasing down this issue, I completely understand. Otherwise, read on.

Per your suggestions I corrected my "const" inconsistencies. But, I'm still seeing some "access violation" exceptions. I always suspect my code first, especially the newer parts.

To review, I am parsing a JSON string coming back from the response of an HTTP transaction. That's what's in the JSON file I shared.

The code does two things.
1. Parse the response payload to see if the transaction was even successful.
2. If successful, parse the same payload (same string!) and extract all the good stuff.

Here is the weird part. The second part always works. If I skip over the high-level error check and go straight into the serious parsing of the parsing, everything works great, even for responses which cause problems in the higher-level error check.

The high-level error check is done in a called function, deeper in the stack. So, I tried a few things to eliminate the stack issue as a cause.

1. Set my linker stack parameters to stupid large values.
2. Just prior to the JSON deserialize calls, call a function which allocates a large array on the stack and memsets it to something known (e.g,. '~'). Because it's done in a separate function call, the stack memory gets put into a known state. I wanted to verify that I had plenty of stack and see if some of those bytes show up on the exception analysis.

None of this made a difference in the behavior. And I saw no signs of the memory values in the exception analysis.

The higher-level error checking code and heavy parsing code are using separate, but equally defined StaticJsonDocument<8192000> global variables. The complexity of the code following the deserializeJson() call doesn't matter, the exception occurs inside the deserializeJson() call.

I think the test works great because it's being called at a higher point in the stack. So, I declared a large stack variable and tried to run the test again. It worked perfectly. This is what leads me to consider my code as the source of the problem.

I can say that there is a correlation between the size of the JSON string and the failure (but, only when the deserializeJson() call is made deeper in the stack). Some data that I gathered. "skip" means that the deserializeJson() call will have an "access violation" exception. "OK" means that the call works perfectly.

 904,624: skip
 201,072: OK
  25,009: OK
 231,291: OK
 499,051: OK
 641,912: skip
 622,173: skip
 680,279: skip
 885,103: skip
1,885,995: skip

Finally, I was able to see many exceptions and [maybe] narrow down the location of the failure.

It appears to be related to CollectionData::getSlot(), specifically, when stringEquals() is called, passing the string being evaluated and the current "slot". It's always slot->key that points to bad memory.

Again, thank you for your patience and understanding.

bblanchon, 2023-11-29:
It's impossible to give you any advice without looking at the code.
I do live mentoring over Zoom at affordable prices; please email me if you're interested.

bergdevCopeland, 2023-11-29:
Thank you, Benoît.

What are the terms of your mentoring? Hourly? Minimum charge?

Even though I’m doing work for Copeland, I am an independent contractor and would pay for it myself.

bblanchon, 2023-11-29:
I charge by the minute, $80/hr, no minimum.
You can find my email on the contact page of arduinojson.org

closed 2023-12-13 by bblanchon