bblanchon/ArduinoJson #2001: IAR Embedded Workbench Compiler Error

thomas-bernhard, 2023-11-28:
**Describe the bug**  
The ArduinoJson Library does not compile using the IAR Embedded Workbench for versions 20.0 and newer. Version 19.4 compiles without any errors or warnings. We already reached out to the support, who could reproduce the error with the current version (EWARM 9.40.2) and with version EWSYN 8.23.3. Both show the same error.

**Troubleshooter report**  
Here is the report generated by the [ArduinoJson Troubleshooter](https://arduinojson.org/v6/troubleshooter/):  
1. The program uses ArduinoJson 21.3
2. The issue happens at compile time
3. The error is not in the list

**Environment**  
Here is the environment that I used:
* EWSYN 8.23.3
* Renesas Synergy R7FS5

**Reproduction**  
Here is a small snippet that reproduces the issue.

```c++
char json[] = "{\"sensor\":\"gps\",\"time\":1351824120,\"data\":[48.756080,2.302038]}";

DynamicJsonDocument doc(1024);
deserializeJson(doc, json);

const char* sensor = doc["sensor"];
long time          = doc["time"];
double latitude    = doc["data"][0];
double longitude   = doc["data"][1];
```

**Compiler output**    
Building configuration: GPT_HAL_MG_AP - Debug 
Updating build tree... 
json_test.cpp  
Warning[Pe1346]: function returns incomplete class type "ArduinoJson::V6213DB2::JsonVariant" C:\Users\...\src\ArduinoJson-v6.21.3.hpp 2551 
Warning[Pe1346]: function returns incomplete class type "ArduinoJson::V6213DB2::JsonVariant" C:\Users\...\src\ArduinoJson-v6.21.3.hpp 2569 
Warning[Pe1346]: function returns incomplete class type "ArduinoJson::V6213DB2::JsonVariant" C:\Users\...\src\ArduinoJson-v6.21.3.hpp 2591 
Warning[Pe1346]: function returns incomplete class type "ArduinoJson::V6213DB2::JsonVariant" C:\Users\...\src\ArduinoJson-v6.21.3.hpp 2597 
Error[Pe304]: no instance of overloaded function "ArduinoJson::V6213DB2::detail::VariantRefBase<TDerived>::as [with TDerived=ArduinoJson::V6213DB2::JsonVariant]" matches the argument list C:\Users\...\src\ArduinoJson-v6.21.3.hpp 2555 
          detected during: 
            instantiation of "inline ArduinoJson::V6213DB2::detail::VariantRefBase<TDerived>::operator T() const [with TDerived=ArduinoJson::V6213DB2::JsonVariant, T=ArduinoJson::V6213DB2::JsonVariant]" at line 2669 
            instantiation of "inline TDerived const &ArduinoJson::V6213DB2::detail::VariantRefBase<TDerived>::derived() const [with TDerived=ArduinoJson::V6213DB2::JsonVariant]" at line 2672 
            instantiation of "inline __interwork __softfp ArduinoJson::V6213DB2::detail::MemoryPool *ArduinoJson::V6213DB2::detail::VariantRefBase<TDerived>::getPool() const [with  
TDerived=ArduinoJson::V6213DB2::JsonVariant]" at line 2592 
            instantiation of "inline bool ArduinoJson::V6213DB2::detail::VariantRefBase<TDerived>::set(T const &) const [with TDerived=ArduinoJson::V6213DB2::JsonVariant, T=ArduinoJson::V6213DB2::JsonVariantConst]" at  
line 3546 
Warning[Pe430]: returning reference to local temporary C:\Users\...\src\ArduinoJson-v6.21.3.hpp 2669 
          detected during: 
            instantiation of "inline TDerived const &ArduinoJson::V6213DB2::detail::VariantRefBase<TDerived>::derived() const [with TDerived=ArduinoJson::V6213DB2::JsonVariant]" at line 2672 
            instantiation of "inline __interwork __softfp ArduinoJson::V6213DB2::detail::MemoryPool *ArduinoJson::V6213DB2::detail::VariantRefBase<TDerived>::getPool() const [with  
TDerived=ArduinoJson::V6213DB2::JsonVariant]" at line 2592 
            instantiation of "inline bool ArduinoJson::V6213DB2::detail::VariantRefBase<TDerived>::set(T const &) const [with TDerived=ArduinoJson::V6213DB2::JsonVariant, T=ArduinoJson::V6213DB2::JsonVariantConst]" at  
line 3546 
Warning[Pe540]: support for exception handling is disabled C:\Users\...\src\ArduinoJson-v6.21.3.hpp 5496 
Error[Pe291]: no default constructor exists for class "ArduinoJson::V6213DB2::detail::MemberProxy<ArduinoJson::V6213DB2::JsonDocument &, char const *>" C:\Users\...\src\ArduinoJson-v6.21.3.hpp 4687 
          detected during: 
            instantiation of "inline T ArduinoJson::V6213DB2::Converter<T, Enable>::fromJson(ArduinoJson::V6213DB2::JsonVariantConst) [with  
T=ArduinoJson::V6213DB2::detail::MemberProxy<ArduinoJson::V6213DB2::JsonDocument &, char const *>, Enable=void]" at line 2546 
            instantiation of "inline ArduinoJson::V6213DB2::detail::enable_if<<expression>, T>::type ArduinoJson::V6213DB2::detail::VariantRefBase<TDerived>::as<T>() const [with  
TDerived=ArduinoJson::V6213DB2::detail::MemberProxy<ArduinoJson::V6213DB2::JsonDocument &, char const *>, T=ArduinoJson::V6213DB2::detail::MemberProxy<ArduinoJson::V6213DB2::JsonDocument &, char  
const *>]" at line 2555 
            instantiation of "inline ArduinoJson::V6213DB2::detail::VariantRefBase<TDerived>::operator T() const [with TDerived=ArduinoJson::V6213DB2::detail::MemberProxy<ArduinoJson::V6213DB2::JsonDocument &, char  
const *>, T=ArduinoJson::V6213DB2::detail::MemberProxy<ArduinoJson::V6213DB2::JsonDocument &, char const *>]" at line 2669 
            instantiation of "inline TDerived const &ArduinoJson::V6213DB2::detail::VariantRefBase<TDerived>::derived() const [with  
TDerived=ArduinoJson::V6213DB2::detail::MemberProxy<ArduinoJson::V6213DB2::JsonDocument &, char const *>]" at line 2675 
            instantiation of "inline __interwork __softfp ArduinoJson::V6213DB2::detail::VariantData *ArduinoJson::V6213DB2::detail::VariantRefBase<TDerived>::getData() const [with  
TDerived=ArduinoJson::V6213DB2::detail::MemberProxy<ArduinoJson::V6213DB2::JsonDocument &, char const *>]" at line 2683 
            instantiation of "inline __interwork __softfp ArduinoJson::V6213DB2::JsonVariantConst ArduinoJson::V6213DB2::detail::VariantRefBase<TDerived>::getVariantConst() const [with  
TDerived=ArduinoJson::V6213DB2::detail::MemberProxy<ArduinoJson::V6213DB2::JsonDocument &, char const *>]" at line 2546 
            instantiation of "inline ArduinoJson::V6213DB2::detail::enable_if<<expression>, T>::type ArduinoJson::V6213DB2::detail::VariantRefBase<TDerived>::as<T>() const [with  
TDerived=ArduinoJson::V6213DB2::detail::MemberProxy<ArduinoJson::V6213DB2::JsonDocument &, char const *>, T=char const *]" at line 2555 
[... 65 lines cut ...]
TDerived=ArduinoJson::V6213DB2::detail::ElementProxy<ArduinoJson::V6213DB2::detail::MemberProxy<ArduinoJson::V6213DB2::JsonDocument &, char const *>>]" at line 2675 
            instantiation of "inline __interwork __softfp ArduinoJson::V6213DB2::detail::VariantData *ArduinoJson::V6213DB2::detail::VariantRefBase<TDerived>::getData() const [with  
TDerived=ArduinoJson::V6213DB2::detail::ElementProxy<ArduinoJson::V6213DB2::detail::MemberProxy<ArduinoJson::V6213DB2::JsonDocument &, char const *>>]" at line 2683 
            instantiation of "inline __interwork __softfp ArduinoJson::V6213DB2::JsonVariantConst ArduinoJson::V6213DB2::detail::VariantRefBase<TDerived>::getVariantConst() const [with  
TDerived=ArduinoJson::V6213DB2::detail::ElementProxy<ArduinoJson::V6213DB2::detail::MemberProxy<ArduinoJson::V6213DB2::JsonDocument &, char const *>>]" at line 2546 
            instantiation of "inline ArduinoJson::V6213DB2::detail::enable_if<<expression>, T>::type ArduinoJson::V6213DB2::detail::VariantRefBase<TDerived>::as<T>() const [with  
TDerived=ArduinoJson::V6213DB2::detail::ElementProxy<ArduinoJson::V6213DB2::detail::MemberProxy<ArduinoJson::V6213DB2::JsonDocument &, char const *>>, T=double]" at line 2555 
            instantiation of "inline ArduinoJson::V6213DB2::detail::VariantRefBase<TDerived>::operator T() const [with  
TDerived=ArduinoJson::V6213DB2::detail::ElementProxy<ArduinoJson::V6213DB2::detail::MemberProxy<ArduinoJson::V6213DB2::JsonDocument &, char const *>>, T=double]" at line 15 of "C:\Users\...\src\json_test.cpp" 
Warning[Pe430]: returning reference to local temporary C:\Users\...\src\ArduinoJson-v6.21.3.hpp 2669 
          detected during: 
            instantiation of "inline TDerived const &ArduinoJson::V6213DB2::detail::VariantRefBase<TDerived>::derived() const [with  
TDerived=ArduinoJson::V6213DB2::detail::ElementProxy<ArduinoJson::V6213DB2::detail::MemberProxy<ArduinoJson::V6213DB2::JsonDocument &, char const *>>]" at line 2675 
            instantiation of "inline __interwork __softfp ArduinoJson::V6213DB2::detail::VariantData *ArduinoJson::V6213DB2::detail::VariantRefBase<TDerived>::getData() const [with  
[... cut ...]
TDerived=ArduinoJson::V6213DB2::detail::ElementProxy<ArduinoJson::V6213DB2::detail::MemberProxy<ArduinoJson::V6213DB2::JsonDocument &, char const *>>, T=double]" at line 2555 
            instantiation of "inline ArduinoJson::V6213DB2::detail::VariantRefBase<TDerived>::operator T() const [with  
TDerived=ArduinoJson::V6213DB2::detail::ElementProxy<ArduinoJson::V6213DB2::detail::MemberProxy<ArduinoJson::V6213DB2::JsonDocument &, char const *>>, T=double]" at line 15 of "C:\Users\...\src\json_test.cpp" 
Error while running C/C++ Compiler 
 
Total number of errors: 5 
Total number of warnings: 8

bblanchon, 2023-11-28:
Hi @thomas-bernhard,

Thank you very much for reporting this bug.

There seems to be something wrong with the template type deduction, probably with `VariantRefBase::operator T()`.
Since ArduinoJson compiles fine with major compilers and previous versions of IAR, I'm tempted to believe it's a compiler bug.

Still, I'd like to reproduce the error too, so I contacted IAR to renew my partner license.
I'll let you know when I can get my hands on the new version.

Best regards,
Benoit

bblanchon, 2023-12-06:
The IAR client support sent me a workaround, which I implemented in the branch `issues/2001`.
Unfortunately, they still haven't renewed my license, so I cannot test.
Please try the header file from the [artifacts of build 7117460544](https://github.com/bblanchon/ArduinoJson/actions/runs/7117460544#artifacts) and tell me if it fixes the issue.

bblanchon, 2023-12-07:
I'll merge this into the main branch and make a new release.
Please open a new issue for the warning if you do want to see it fixed.

bblanchon, 2023-12-07:
The fix was published with [ArduinoJson 6.21.4](https://github.com/bblanchon/ArduinoJson/releases/tag/v6.21.4).

bblanchon, 2023-12-07:
I finally received my license and managed to fix the warning.
The fix will be included in the next release of ArduinoJson.

bblanchon, 2024-01-10:
The fix for the warning was published in [ArduinoJson 6.21.5](https://github.com/bblanchon/ArduinoJson/releases/tag/v6.21.5)

commit d981535d (2023-12-07): Fix `no instance of overloaded function...` on IAR (fixes #2001)

Ported from 6ed87029e26efcfdf7904cec5cc053a567416a8e
commit 6ed87029 (2023-12-07): Fix `no instance of overloaded function...` on IAR (fixes #2001)