bblanchon/ArduinoJson #2003: Deserialize/Serialize results in structural differences

bergdevCopeland, 2023-11-29:
**Describe the bug**  
Issues deserializing a large JSON string

**Troubleshooter report**  
Here is the report generated by the [ArduinoJson Troubleshooter](https://arduinojson.org/v6/troubleshooter/):  
1. The program uses ArduinoJson 6
2. The issue happens at run time
3. The issue concerns deserialization
4. `deserializeJson()` returns `Ok`
5. The strings in the `JsonDocument` change for no reason
6. Input type is `const char*`

**Environment**  
Here is the environment that I used:
* Microcontroller: [x86]
* Core/runtime: [Windows 10 C runtime library version not known]
* IDE: [Microsoft Visual Studio/C++ 2019]

**Reproduction**  
#include <ArduinoJson-v6.21.3.h>
const char*	pszFilenameJsonInput = "C:\\LCB\\GetApplicationConfiguration_RESPONSE.json"; // 3318779 bytes, as prettyfied by Notepad++ JSTool plugin
const char*	pszFilenameJsonOutput = "C:\\LCB\\GetApplicationConfiguration_RESPONSE_OUTPUT.json"; // prettyfy, using Notepad++ JSTool plugin before comparing
char		acJsonInput[8 * 1024 * 1024];
char		acJsonOutput[8 * 1024 * 1024];

int main()
{
	struct stat		st;
	char			acErrMessage[500];
	FILE*			fileJsonInput;
	FILE*			fileJsonOutput;
	UINT			nFileSizeInput = 0;
	UINT			nFileSizeOutput = 0;
	UINT			nMemoryUsed = 0;
	JsonObjectConst	Root;

	memset(acJsonInput, 0, sizeof(acJsonInput));
	memset(acJsonOutput, 0, sizeof(acJsonOutput));

	stat(pszFilenameJsonInput, &st);
	nFileSizeInput = st.st_size;

	if (nFileSizeInput < sizeof(acJsonInput)) {
		fileJsonInput = fopen(pszFilenameJsonInput, "r");

		if (fileJsonInput != (FILE*)NULL) {
			fread(acJsonInput, nFileSizeInput, 1, fileJsonInput);
			fclose(fileJsonInput);

			const char* pszJsonInput = acJsonInput;
			DeserializationError error = deserializeJson(docTest, pszJsonInput);
			if (error) {
				sprintf(acErrMessage, "JSON DeSerialize error: %s", error.c_str());
			}
			else {
				Root = docTest.as<JsonObjectConst>();
				nFileSizeOutput = serializeJson(Root, acJsonOutput, sizeof(acJsonOutput));

				fileJsonOutput = fopen(pszFilenameJsonOutput, "w");

				if (fileJsonOutput != (FILE*)NULL) {
					fwrite(acJsonOutput, nFileSizeOutput, 1, fileJsonOutput);
					fclose(fileJsonOutput);
				}
			}
		}
	}
	return(0);
}

After running this program, please prettyfy both input and output files using the same method for easier comparison with your favorite tool. I use BeyondCompare.

NOTE:
The above source code can be found at http://bergdev.com/ArduinoJsonTest.c
The input JSON file is at http://bergdev.com/GetApplicationConfiguration_RESPONSE.json

**Compiler output**    
Not Applicable

**Program output**  
After prettyfying the OUTPUT file, a directory listing shows the following. Note the difference in size.

11/29/2023  02:49 PM         3,318,779 GetApplicationConfiguration_RESPONSE.json
11/29/2023  03:39 PM         3,041,231 GetApplicationConfiguration_RESPONSE_OUTPUT.json

Expected output:

I expected the files to be within a byte or two in size and BeyondCompare to show no strutural differences in the JSON.

Actual output:

At lines 5898 and 5971 the OUTPUT file has a couple of string values which are familiar but incorrect. Then, around line 72358, the differences become significant with entire objects being missing.

FWIW, I am willing to pay for your time to help you look into this. Or, if you feel that ArduinoJson simply isn't suited for such a large use case, just let me know that, too.

Thank you.

bergdevCopeland, 2023-11-29:
[GetApplicationConfiguration_RESPONSE_OUTPUT.json](https://github.com/bblanchon/ArduinoJson/files/13505540/GetApplicationConfiguration_RESPONSE_OUTPUT.json)
[GetApplicationConfiguration_RESPONSE.json](https://github.com/bblanchon/ArduinoJson/files/13505541/GetApplicationConfiguration_RESPONSE.json)
[TestJson.json](https://github.com/bblanchon/ArduinoJson/files/13505542/TestJson.json)

Sorry, not accustomed to working with GitHub issues. Here are all the files.

bblanchon, 2023-11-30:
Hi Lee,

I simplified your program to this:

```c++
#include "ArduinoJson-v6.21.3.h"
#include <stdio.h>

const char* pszFilenameJsonInput = "GetApplicationConfiguration_RESPONSE.json";
const char* pszFilenameJsonOutput = "GetApplicationConfiguration_RESPONSE_OUTPUT.json";
char acJsonInput[8 * 1024 * 1024];
char acJsonOutput[8 * 1024 * 1024];

int main()
{
  FILE* fileJsonInput = fopen(pszFilenameJsonInput, "r");
  size_t nFileSizeInput = fread(acJsonInput, 1, sizeof(acJsonInput), fileJsonInput);
  fclose(fileJsonInput);

  DynamicJsonDocument docTest(4194304);
  deserializeJson(docTest, acJsonInput, nFileSizeInput);
  size_t nFileSizeOutput = serializeJson(docTest, acJsonOutput, sizeof(acJsonOutput));

  FILE* fileJsonOutput = fopen(pszFilenameJsonOutput, "w");
  fwrite(acJsonOutput, 1, nFileSizeOutput, fileJsonOutput);
  fclose(fileJsonOutput);
}
```

I confirm that this program works for x64 and not x86.

Indeed, due to an optimization, a `JsonDocument` is limited in the number of nodes it can contain.
This limit is 64K on x86 and 4M on x64, but you can override the default by adding this line at the top of the file:

```c++
#define ARDUINOJSON_SLOT_OFFSET_SIZE 4
```

Caution: this will significantly increase memory consumption, so you'll need to increase the capacity of your `JsonDocument`. Unfortunately, you cannot count on the ArduinoJson Assistant anymore.

Sure, `deserializeJson()` should have returned an error, but I don't think it's worth fixing it in v6 since I know that it returns `NoMemory` in v7.

Best regards,
Benoit

bergdevCopeland, 2023-11-30:
Thank you, sir.

Your proposed solution has been verified in the application. In fact, once able to consume all of the nodes, deserializing one of the larger documents led to a "TooDeep" error. I added a second line to the top of the file:
`#define ARDUINOJSON_DEFAULT_NESTING_LIMIT 15 // original default is 10`
The max memory usage for this somewhat large test case is still easily managed in this environment.

Finally, I realize that my usage of ArduinoJson is not typical. However, it appears to work well now. Thank you for all your hard work in making ArduinoJson work well, even at scale.

closed 2023-11-30 by bergdevCopeland

bblanchon, 2023-12-01:
You're welcome, Lee.
Thank you for using ArduinoJson!
Don't forget to [cast a star](https://github.com/bblanchon/ArduinoJson/stargazers) to support the project :wink: