bblanchon/ArduinoJson #2004: Extra square brackets in JsonObject with JsonArray

OleksandrDiachenko, 2023-11-30:
As a result of serializing this code on ESP32, I get the following outcome:
 {"s":[{"pm":"pm","oa":"oa","sl":"sl","cpp":[0,1,2,3,4]}]}

```
JsonObject s = doc["s"].add<JsonObject>(); // s
s["pm"] = "pm";
s["oa"] = "oa";
s["sl"] = "sl";

 JsonArray s_cpp = s["cpp"].to<JsonArray>();
        for (int i = 0; i < 5; i++)
        {
            s_cpp.add(i);
        }
```

bblanchon, 2023-12-01:
Hi @OleksandrDiachenko,

```json
{"s":[{"pm":"pm","oa":"oa","sl":"sl","cpp":[0,1,2,3,4]}]}
```

I don't see anything wrong with the brackets.
What did you expect?

Best regards,
Benoit

OleksandrDiachenko, 2023-12-01:
Thank you for the response. 
Perhaps I'm mistaken, but the value of 's' should be an object, not an array of objects.

ArduinoJson @ 6.21.3
`{"s":{"dm":"dm","oa":"oa","sl":"sl","cpp":[0,1,2,3,4]}}`

```
StaticJsonDocument<256> doc;

JsonObject s = doc.createNestedObject("s");
s["dm"] = "dm";
s["oa"] = "oa";
s["sl"] = "sl";

JsonArray s_cpp = s.createNestedArray("cpp");
  for (int i = 0; i < 5; i++)
  {
    s_cpp.add(i);
  }

String output;
serializeJson(doc, output);
```

ArduinoJson @ 7.0.0-alpha+sha.f7dcea5 

`{"s":[{"pm":"pm","oa":"oa","sl":"sl","cpp":[0,1,2,3,4]}]}`

```
  JsonDocument doc;

  JsonObject s = doc["s"].add<JsonObject>(); // s
  s["pm"] = "pm";
  s["oa"] = "oa";
  s["sl"] = "sl";

JsonArray s_cpp = s["cpp"].to<JsonArray>();
  for (int i = 0; i < 5; i++)
  {
    s_cpp.add(i);
  }
```

bblanchon, 2023-12-03:
`add<JsonObject>()` appends a new object to an array. You should call `to<JsonObject>()` instead.

```diff
- JsonObject s = doc["s"].add<JsonObject>();
+ JsonObject s = doc["s"].to<JsonObject>();
```

BTW, you can use the [ArduinoJson Assistant](https://arduinojson.org/v7/assistant/) to generate scaffolding code.