bblanchon/ArduinoJson #2006: USGS GeoJson Summary API

rjssjs, 2023-12-06:
**Describe the issue**  
A clear and concise description of what you're trying to do.
You don't need to explain every aspect of your project: focus on the problem you're having.

**Troubleshooter report**  
Here is the report generated by the [ArduinoJson Troubleshooter](https://arduinojson.org/v6/troubleshooter/):  
[Paste the report here]

**Environment**  
Here is the environment that I'm using':
* Microconroller: [e.g. ESP8266]
* Core/runtime: [e.g. ESP8266 core for Arduino v3.0.2]
* IDE: [e.g. Arduino IDE 1.8.16]

**Reproduction**  
Here is a small snippet that demonstrate the problem.

```c++
DynamicJsonDocument doc(1024);

DeserializationError error = deserializeJson(doc, "{\"hello\":\"world\"}");

// insert code here
```

**Program output**  
If relevant, include the program output.

Expected output:

```
[insert expected output here]
```

Actual output:

```
[insert actual output here]
```

rjssjs, 2023-12-06:
Each time I try to use the Assistant I get the following error when pasting the  GeoJson summary API into the assistant. I have used this numerous times in the past with version 5 and it worked perfectly and gave me the required sisze for the arrays. For some reason it no longer seems to work. I don't think the USGS code has changed  (https://earthquake.usgs.gov/earthquakes/feed/v1.0/geojson.php)

Expected property name or '}' in JSON at position 4 (line 2 column 3)

Help with this would be greatly apreciated

bblanchon, 2023-12-08:
Hi @rjssjs,

The following text is not a valid JSON document:

```json
{
  type: "FeatureCollection",
  metadata: {
    [generated](https://earthquake.usgs.gov/data/comcat/data-eventterms.php#metadata_generated): Long Integer,
    [url](https://earthquake.usgs.gov/data/comcat/data-eventterms.php#metadata_url): String,
    [title](https://earthquake.usgs.gov/data/comcat/data-eventterms.php#metadata_title): String,
    [api](https://earthquake.usgs.gov/data/comcat/data-eventterms.php#metadata_api): String,
    [count](https://earthquake.usgs.gov/data/comcat/data-eventterms.php#metadata_count): Integer,
    [status](https://earthquake.usgs.gov/data/comcat/data-eventterms.php#metadata_status): Integer
  },
  bbox: [
    [minimum longitude](https://earthquake.usgs.gov/data/comcat/data-eventterms.php#longitude),
    [minimum latitude](https://earthquake.usgs.gov/data/comcat/data-eventterms.php#latitude),
    [minimum depth](https://earthquake.usgs.gov/data/comcat/data-eventterms.php#depth),
    [maximum longitude](https://earthquake.usgs.gov/data/comcat/data-eventterms.php#longitude),
    [maximum latitude](https://earthquake.usgs.gov/data/comcat/data-eventterms.php#latitude),
    [maximum depth](https://earthquake.usgs.gov/data/comcat/data-eventterms.php#depth)
  ],
  features: [
    {
      type: "Feature",
      properties: {
        [mag](https://earthquake.usgs.gov/data/comcat/data-eventterms.php#mag): Decimal,
        [place](https://earthquake.usgs.gov/data/comcat/data-eventterms.php#place): String,
        [time](https://earthquake.usgs.gov/data/comcat/data-eventterms.php#time): Long Integer,
        [updated](https://earthquake.usgs.gov/data/comcat/data-eventterms.php#updated): Long Integer,
        [tz](https://earthquake.usgs.gov/data/comcat/data-eventterms.php#tz): Integer,
        [url](https://earthquake.usgs.gov/data/comcat/data-eventterms.php#url): String,
        [detail](https://earthquake.usgs.gov/data/comcat/data-eventterms.php#detail): String,
        [felt](https://earthquake.usgs.gov/data/comcat/data-eventterms.php#felt):Integer,
        [cdi](https://earthquake.usgs.gov/data/comcat/data-eventterms.php#cdi): Decimal,
        [mmi](https://earthquake.usgs.gov/data/comcat/data-eventterms.php#mmi): Decimal,
        [alert](https://earthquake.usgs.gov/data/comcat/data-eventterms.php#alert): String,
        [status](https://earthquake.usgs.gov/data/comcat/data-eventterms.php#status): String,
        [tsunami](https://earthquake.usgs.gov/data/comcat/data-eventterms.php#tsunami): Integer,
        [sig](https://earthquake.usgs.gov/data/comcat/data-eventterms.php#sig):Integer,
        [net](https://earthquake.usgs.gov/data/comcat/data-eventterms.php#net): String,
        [code](https://earthquake.usgs.gov/data/comcat/data-eventterms.php#code): String,
        [ids](https://earthquake.usgs.gov/data/comcat/data-eventterms.php#ids): String,
        [sources](https://earthquake.usgs.gov/data/comcat/data-eventterms.php#sources): String,
        [types](https://earthquake.usgs.gov/data/comcat/data-eventterms.php#types): String,
        [nst](https://earthquake.usgs.gov/data/comcat/data-eventterms.php#nst): Integer,
        [dmin](https://earthquake.usgs.gov/data/comcat/data-eventterms.php#dmin): Decimal,
        [rms](https://earthquake.usgs.gov/data/comcat/data-eventterms.php#rms): Decimal,
        [gap](https://earthquake.usgs.gov/data/comcat/data-eventterms.php#gap): Decimal,
        [magType](https://earthquake.usgs.gov/data/comcat/data-eventterms.php#magType): String,
        [type](https://earthquake.usgs.gov/data/comcat/data-eventterms.php#type): String
      },
      geometry: {
        type: "Point",
        coordinates: [
          [longitude](https://earthquake.usgs.gov/data/comcat/data-eventterms.php#longitude),
          [latitude](https://earthquake.usgs.gov/data/comcat/data-eventterms.php#latitude),
          [depth](https://earthquake.usgs.gov/data/comcat/data-eventterms.php#depth)
        ]
      },
      [id](https://earthquake.usgs.gov/data/comcat/data-eventterms.php#id): String
    },
    …
  ]
}
```

However, if I click on any on the link on the ride panel, I get a valid document.
For example, `https://earthquake.usgs.gov/earthquakes/feed/v1.0/summary/significant_day.geojson` returns:
[... 67 lines cut ...]
        "code": "7000lgwp",
        "ids": ",at00s5ary7,us7000lgwp,pt23341051,",
        "sources": ",at,us,pt,",
        "types": ",dyfi,finite-fault,general-text,ground-failure,impact-link,losspager,moment-tensor,origin,phase-data,shakemap,",
        "nst": 105,
        "dmin": 2.542,
        "rms": 1.04,
        "gap": 32,
        "magType": "mww",
        "type": "earthquake",
        "title": "M 7.1 - 123 km S of Isangel, Vanuatu"
      },
      "geometry": {
        "type": "Point",
        "coordinates": [169.2064, -20.6585, 48.364]
      },
      "id": "us7000lgwp"
    }
  ],
  "bbox": [-98.5503, -20.6585, 44.388, 169.2064, 18.3866, 48.364]
}
```

Best regards,
Benoit

closed 2023-12-28 by bblanchon