bblanchon/ArduinoJson #2007: Platformio Native custom implementation of Arduino String

gabryelreyes, 2023-12-07:
**Describe the issue**  
Hello, I am working on a project that should compile for Arduino as well as for Platformio's Native environment. For this, we developed a replacement of the String class for the native side, so Strings can be used in the application without having to use `ifdef` or checking which environment is being used at the moment. The native String has the same functionalities as Arduino String, however I run into the "unsuported type" error when trying to insert one into a JsonDocument. 

I know how to solve it implementing `convertToJson(...)` or using `c_str()`, but my question is: 

**what should my implementation of String do/have in order to be recognized as an Arduino String by ArduinoJson?** 

Link to native String implementation: https://github.com/BlueAndi/DroidControlShip/blob/main/lib/ArduinoNative/src/WString.h

**Troubleshooter report**  
Here is the report generated by the [ArduinoJson Troubleshooter](https://arduinojson.org/v6/troubleshooter/):  
1. The program uses ArduinoJson 6
2. The issue happens at compile time
3. Error says "no matching function for call to ..."
4. Error says "no matching function for call to `convertToJson(...)`"
5. Converting the value doesn't fix the issue

**Environment**  
Here is the environment that I'm using':
* Microcontroller: None
* Core/runtime: PlatformIO native @ ~1.2.1
* IDE: VS Code

**Reproduction**  
```c++
// Using native String implementation.
String m_robotName = "foo";
StaticJsonDocument<maxDocSize> doc;
doc[ROBOT_NAME]   = m_robotName;
```

**Program output**  

Expected output:
Succesful compilation.

Actual output:

```
In file included from .pio\libdeps\ConvoyLeaderSim\ArduinoJson\src/ArduinoJson.hpp:42,
                 from .pio\libdeps\ConvoyLeaderSim\ArduinoJson\src/ArduinoJson.h:9,
                 from lib\Service\src\SettingsHandler.cpp:37:
.pio\libdeps\ConvoyLeaderSim\ArduinoJson\src/ArduinoJson/Variant/ConverterImpl.hpp: In instantiation of 'static void ArduinoJson::V6213HB4::Converter<T, Enable>::toJson(const T&, ArduinoJson::V6213HB4::JsonVariant) [with T = String; Enable = void]':
.pio\libdeps\ConvoyLeaderSim\ArduinoJson\src/ArduinoJson/Variant/VariantRefBase.hpp:126:25:   required from 'bool ArduinoJson::V6213HB4::detail::VariantRefBase<TDerived>::set(const T&) const [with T = String; TDerived = ArduinoJson::V6213HB4::detail::MemberProxy<ArduinoJson::V6213HB4::JsonDocument&, const char*>]'
.pio\libdeps\ConvoyLeaderSim\ArduinoJson\src/ArduinoJson/Object/MemberProxy.hpp:33:14:   required from 'ArduinoJson::V6213HB4::detail::MemberProxy< <template-parameter-1-1>, <template-parameter-1-2> >& ArduinoJson::V6213HB4::detail::MemberProxy< <template-parameter-1-1>, <template-parameter-1-2> >::operator=(const T&) [with T = String; TUpstream = ArduinoJson::V6213HB4::JsonDocument&; TStringRef = const char*]'
lib\Service\src\SettingsHandler.cpp:213:56:   required from here
.pio\libdeps\ConvoyLeaderSim\ArduinoJson\src/ArduinoJson/Variant/ConverterImpl.hpp:17:18: error: no matching function for call to 'convertToJson(const String&, ArduinoJson::V6213HB4::JsonVariant&)'
   17 |     convertToJson(src, dst); // Error here? See https://arduinojson.org/v6/unsupported-set/
      |     ~~~~~~~~~~~~~^~~~~~~~~~
In file included from .pio\libdeps\ConvoyLeaderSim\ArduinoJson\src/ArduinoJson.hpp:44:
.pio\libdeps\ConvoyLeaderSim\ArduinoJson\src/ArduinoJson/Variant/VariantImpl.hpp:147:13: note: candidate: 'template<class TDerived> void ArduinoJson::V6213HB4::detail::convertToJson(const VariantRefBase<TDerived>&, ArduinoJson::V6213HB4::JsonVariant)'
  147 | inline void convertToJson(const VariantRefBase<TDerived>& src,
      |             ^~~~~~~~~~~~~
.pio\libdeps\ConvoyLeaderSim\ArduinoJson\src/ArduinoJson/Variant/VariantImpl.hpp:147:13: note:   template argument deduction/substitution failed:
.pio\libdeps\ConvoyLeaderSim\ArduinoJson\src/ArduinoJson/Variant/ConverterImpl.hpp:17:18: note:   'const String' is not derived from 'const ArduinoJson::V6213HB4::detail::VariantRefBase<TDerived>'
   17 |     convertToJson(src, dst); // Error here? See https://arduinojson.org/v6/unsupported-set/
      |     ~~~~~~~~~~~~~^~~~~~~~~~
.pio\libdeps\ConvoyLeaderSim\ArduinoJson\src/ArduinoJson/Variant/ConverterImpl.hpp:156:1: note: candidate: 'template<class T> typename ArduinoJson::V6213HB4::detail::enable_if<ArduinoJson::V6213HB4::detail::IsString<TString>::value, bool>::type ArduinoJson::V6213HB4::convertToJson(const T&, JsonVariant)'
  156 | convertToJson(const T& src, JsonVariant dst) {
      | ^~~~~~~~~~~~~
.pio\libdeps\ConvoyLeaderSim\ArduinoJson\src/ArduinoJson/Variant/ConverterImpl.hpp:156:1: note:   template argument deduction/substitution failed:
.pio\libdeps\ConvoyLeaderSim\ArduinoJson\src/ArduinoJson/Variant/ConverterImpl.hpp: In substitution of 'template<class T> typename ArduinoJson::V6213HB4::detail::enable_if<ArduinoJson::V6213HB4::detail::IsString<TString>::value, bool>::type ArduinoJson::V6213HB4::convertToJson(const T&, JsonVariant) [with T = String]':
.pio\libdeps\ConvoyLeaderSim\ArduinoJson\src/ArduinoJson/Variant/ConverterImpl.hpp:17:18:   required from 'static void ArduinoJson::V6213HB4::Converter<T, Enable>::toJson(const T&, ArduinoJson::V6213HB4::JsonVariant) [with T = String; Enable = void]'
.pio\libdeps\ConvoyLeaderSim\ArduinoJson\src/ArduinoJson/Variant/VariantRefBase.hpp:126:25:   required from 'bool ArduinoJson::V6213HB4::detail::VariantRefBase<TDerived>::set(const T&) const [with T = String; TDerived = ArduinoJson::V6213HB4::detail::MemberProxy<ArduinoJson::V6213HB4::JsonDocument&, const char*>]'
.pio\libdeps\ConvoyLeaderSim\ArduinoJson\src/ArduinoJson/Object/MemberProxy.hpp:33:14:   required from 'ArduinoJson::V6213HB4::detail::MemberProxy< <template-parameter-1-1>, <template-parameter-1-2> >& ArduinoJson::V6213HB4::detail::MemberProxy< <template-parameter-1-1>, <template-parameter-1-2> >::operator=(const T&) [with T = String; TUpstream = ArduinoJson::V6213HB4::JsonDocument&; TStringRef = const char*]'
lib\Service\src\SettingsHandler.cpp:213:56:   required from here
.pio\libdeps\ConvoyLeaderSim\ArduinoJson\src/ArduinoJson/Variant/ConverterImpl.hpp:156:1: error: no type named 'type' in 'struct ArduinoJson::V6213HB4::detail::enable_if<false, bool>'
In file included from .pio\libdeps\ConvoyLeaderSim\ArduinoJson\src/ArduinoJson/Document/BasicJsonDocument.hpp:7,
                 from .pio\libdeps\ConvoyLeaderSim\ArduinoJson\src/ArduinoJson/Document/DynamicJsonDocument.hpp:7,
                 from .pio\libdeps\ConvoyLeaderSim\ArduinoJson\src/ArduinoJson.hpp:33:
.pio\libdeps\ConvoyLeaderSim\ArduinoJson\src/ArduinoJson/Variant/ConverterImpl.hpp: In instantiation of 'static void ArduinoJson::V6213HB4::Converter<T, Enable>::toJson(const T&, ArduinoJson::V6213HB4::JsonVariant) [with T = String; Enable = void]':
.pio\libdeps\ConvoyLeaderSim\ArduinoJson\src/ArduinoJson/Variant/VariantRefBase.hpp:126:25:   required from 'bool ArduinoJson::V6213HB4::detail::VariantRefBase<TDerived>::set(const T&) const [with T = String; TDerived = ArduinoJson::V6213HB4::detail::MemberProxy<ArduinoJson::V6213HB4::JsonDocument&, const char*>]'
.pio\libdeps\ConvoyLeaderSim\ArduinoJson\src/ArduinoJson/Object/MemberProxy.hpp:33:14:   required from 'ArduinoJson::V6213HB4::detail::MemberProxy< <template-parameter-1-1>, <template-parameter-1-2> >& ArduinoJson::V6213HB4::detail::MemberProxy< <template-parameter-1-1>, <template-parameter-1-2> >::operator=(const T&) [with T = String; TUpstream = ArduinoJson::V6213HB4::JsonDocument&; TStringRef = const char*]'
lib\Service\src\SettingsHandler.cpp:213:56:   required from here
.pio\libdeps\ConvoyLeaderSim\ArduinoJson\src/ArduinoJson/Document/JsonDocument.hpp:321:13: note: candidate: 'void ArduinoJson::V6213HB4::convertToJson(const JsonDocument&, JsonVariant)'
  321 | inline void convertToJson(const JsonDocument& src, JsonVariant dst) {
      |             ^~~~~~~~~~~~~
.pio\libdeps\ConvoyLeaderSim\ArduinoJson\src/ArduinoJson/Document/JsonDocument.hpp:321:47: note:   no known conversion for argument 1 from 'const String' to 'const ArduinoJson::V6213HB4::JsonDocument&'
  321 | inline void convertToJson(const JsonDocument& src, JsonVariant dst) {
```

bblanchon, 2023-12-10:
Hi @gabryelreyes,

If I understand your question correctly, you want to create a minimalist clone of `String` that ArduinoJson will treat as the original `String` class.

Here is the mock implementation of `String` that I used in tests:
https://github.com/bblanchon/ArduinoJson/blob/v6.21.4/extras/tests/Helpers/api/String.h
Some parts, like `operator<<` and `_maxCapacity` are only used for testing, so you can get rid of them.

You must also set [`ARDUINOJSON_ENABLE_ARDUINO_STRING`](https://arduinojson.org/v6/api/config/enable_arduino_string/) to `1`.

Best regards,
Benoit

gabryelreyes, 2023-12-10:
Hi @bblanchon,

Thank you for your reply. You understand correctly, we have a clone of String that should be treated as the original `String` class.
I tried setting `ARDUINOJSON_ENABLE_ARDUINO_STRING` to `1` , but it still does not work. The log looks exactly the same as before.

I will make some tests using your version of `String.h`, however it is not much different than our version (other than a missing concat which I now added). Maybe this will shed some light on why it does not work. 

Just to be sure, what does `detail::IsString<T>::value` check for?

Best wishes,
Gabryel

bblanchon, 2023-12-13:
Hi Gabryel,

`IsString<T>` is a meta function that returns `true` if `T` is a string object.
It works by checking if there is a `StringAdapter<T>`.
Your `String` clone should be handled by [this specialization of `StringAdapter`](https://github.com/bblanchon/ArduinoJson/blob/v6.21.4/src/ArduinoJson/Strings/Adapters/StringObject.hpp#L14) which looks for member functions `length()` and `c_str()`.

Best regards,
Benoit

gabryelreyes, 2023-12-15:
Hi Benoit,

thank you for pointing me in the right direction. After looking at the StringAdapter and StringTraits, I noticed that the clone's `length()` method returned `unsigned int` instead of `size_t`. Changing this fixed the problem and the clone is treated as the original Arduino String. Therefore, I will close this Issue.

Again, thank you very much!

Best wishes, 
Gabryel

closed 2023-12-15 by gabryelreyes

bblanchon, 2023-12-15:
You're welcome, Gabryel!
Thank you for using ArduinoJson.
Don't forget to [cast a star](https://github.com/bblanchon/ArduinoJson/stargazers) to support the project :wink: