bblanchon/ArduinoJson #2008: Serialization gives null in output while document seems properly sized

hemeleerse, 2023-12-08:
**Description**
When I run the serialize command, the output lists "null", 
The doc.capacity() reports 0, the doc.memusage() reports 0, the doc.overlfow() reports 1.
The DynamicJsonDocument is sized at 6500 while the Arduino Assistant suggest 6144.

**Troubleshooter's report**
1. The program uses ArduinoJson 6
2. The issue happens at run time
3. The issue concerns serialization
4. Output contains null
5. Program doesn't call `serializeJson()` in a loop
6. `JsonDocument::overflowed()` returns `true`
7. Increasing the capacity of the `JsonDocument` doesn't solve the issue.

**Environment**
* Microcontroller: NodeMCU 1.0 (ESP-12E module)

**Reproduction code**
```c++
server.on("/dayjson", HTTP_GET, [](AsyncWebServerRequest *request){
  int byteswritten;
  AsyncResponseStream *response = request->beginResponseStream("application/json");
  DynamicJsonDocument doc(6500);
  JsonObject root = doc.createNestedObject();
  root["curpowdel"] = curpowdel;
  root["curpowret"] = curpowret; 
  root["powermax"] = powermax;
  root["powermaxtime"] = powermaxtime;
  root["minqpowavg"] = minqpowavg;
  JsonArray Quarters = root.createNestedArray("quarters");
  for (int i=0; i<=23;i++) {     // 24 hours
    for (int j=0; j<=3; j++){    // 4 quarters per hour
    Quarters.add(quarters[i][j][0]); //min
    Quarters.add(quarters[i][j][1]); //max
    Quarters.add(quarters[i][j][2]); //avg
    }
  }
  Serial.print("dayjson doc capacity :");Serial.println(doc.capacity());
  Serial.print("dayjson doc memusage :");Serial.println(doc.memoryUsage());
  Serial.print("dayjson overflowed? "); Serial.println(doc.overflowed());
  byteswritten = serializeJson(root, *response);
  Serial.print("Byteswritten="); Serial.println(byteswritten);
  request->send(response);
 });
```

**Remarks**
Examples JSON string should look like (I test with zero values for all data) 
{"curpowdel":0,"curpowret":0,"powermax":0,"powermaxtime":"","minqpowavg":0,"quarters":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0
]}

bblanchon, 2023-12-11:
Hi @hemeleerse,

`doc.capacity()` returns `0`, which means the memory pool allocation failed.
6KB is not that big for an ESP8266; you should be able to make it fit.
Try to free some memory and reduce heap fragmentation.

See also: [How to reduce memory usage?](https://arduinojson.org/v6/how-to/reduce-memory-usage/)

Best regards,
Benoit

hemeleerse, 2023-12-12:
I decided not to use Arduinojson for this particular project and constructed the JSON myself. All works fine now.

hemeleerse, 2023-12-12:
I stopped using Arduinojson for this particular project and wrote my own code to create the JSON.

closed 2023-12-12 by hemeleerse