bblanchon/ArduinoJson #2010: EmptyInput from SolarEdge API

rberkelm, 2023-12-12:
**Description**
In Postman, the API call:
>"https://monitoringapi.solaredge.com/site/2807673/currentPowerFlow?api_key=XXXXXXX" 
works and produces output that looks like:
>{"siteCurrentPowerFlow":{"updateRefreshRate":3,"unit":"kW","connections":[{"from":"LOAD","to":"Grid"},{"from":"PV","to":"Load"}],"GRID":{"status":"Active","currentPower":2.94},"LOAD":{"status":"Active","currentPower":1.37},"PV":{"status":"Active","currentPower":4.31}}}

But I can't make it work in my code (see below). The equivalent code with an Openweathermap API call works perfectly. Any help greatly appreciated!

Based on my code snippet below, I get the following serial output:
>13:11:56.458 -> Connecting...
13:11:56.827 -> Connected to SE!
13:11:57.457 -> Client response: <html>
13:12:07.480 -> Invalid response. Client can't find endOfHeaders
13:12:17.461 -> deserializeJson() failed: EmptyInput
13:12:17.461 -> Load power is: 0.00 
13:12:17.461 -> PV power is: 0.00 
13:12:17.461 -> We are using from the grid!

**Troubleshooter's report**
1. The program uses ArduinoJson 6
2. The issue happens at run time
3. The issue concerns deserialization
4. `deserializeJson()` returns `EmptyInput`
5. Input comes from an HTTP response
6. The status code is not in the list
7. Increasing the timeout doesn't fix the issue

**Environment**
* Microcontroller: ESP8266
* IDE: Arduino 2.2.1

**Reproduction code**
```c++
// Open connection to the HTTP server
void requestSolarEdge(){
   Serial.println(F("Connecting..."));

  // Connect to HTTP server
  client.setTimeout(10000);
  if (!client.connect("monitoringapi.solaredge.com", 80)) {
    Serial.println(F("Connection failed"));
    return;
  }

  Serial.println(F("Connected to SE!")); 

    
  // Send HTTP request
    client.println("GET /site/2807673/currentPowerFlow?api_key=KXXXXX");  
    client.println("Host: monitoringapi.solaredge.com"); 
    client.println(); // end HTTP request header
 
if(!client.connected())
{
  //if the server's disconnected, stop the client:
  Serial.println("disconnected");
  client.stop();
}
}

void readHttpMsg(){
  // Check HTTP status
  Serial.setTimeout(10000);
  char status[64] = {0};
  client.readBytesUntil('\r', status, sizeof(status));
 
  // It should be "HTTP/1.0 200 OK" or "HTTP/1.1 200 OK"
  if (strcmp(status + 9, "200 OK") != 0) {
    Serial.print(F("Client response: "));
    Serial.println(status);
    client.stop();
  }

  // Skip HTTP headers
  char endOfHeaders[] = "\r\n\r\n";
  if (!client.find(endOfHeaders)) {
    Serial.println(F("Invalid response. Client can't find endOfHeaders"));
    client.stop();
    return;
  }
}

void parseData(){
    // Allocate the JSON document
  // Use https://arduinojson.org/v6/assistant to compute the capacity.
  StaticJsonDocument<1024> doc; 

client.setTimeout(10000);  // 10 seconds

  // Parse JSON object
  DeserializationError error = deserializeJson(doc, client);
  if (error) {
  Serial.print(F("deserializeJson() failed: "));
  Serial.println(error.f_str());
  }

  // Extract values
  JsonObject siteCurrentPowerFlow = doc["siteCurrentPowerFlow"];
  loadPwr = siteCurrentPowerFlow["LOAD"]["currentPower"]; 
  pvPwr = siteCurrentPowerFlow["PV"]["currentPower"];
 
  Serial.printf ("Load power is: %.2f \r\n", loadPwr);
  Serial.printf ("PV power is: %.2f \r\n", pvPwr);
    if (pvPwr>loadPwr){
      Serial.println(F("We are feeding into the grid"));
        }
        else{
          Serial.println(F("We are using from the grid!"));
        }
     
  // Disconnect
  client.stop();
}
```

bblanchon, 2023-12-13:
Hi @rberkelm,

Here here what I get with curl:

```http
HTTP/1.1 301 Moved Permanently
Server: nginx
Date: Wed, 13 Dec 2023 20:05:07 GMT
Content-Type: text/html
Content-Length: 162
Connection: keep-alive
Location: https://monitoringapi.solaredge.com/site/2807673/currentPowerFlow?api_key=XXXXXXX
```

The server redirects you to HTTPS, so you must use `WiFiClientSecure` and port 443.

See:

* [Deserialization tutorial](https://arduinojson.org/v6/doc/deserialization/), which shows how to connect to an HTTPS server using ESP8266HTTPClient.
* The Reddit case study of [Mastering ArduinoJson](https://arduinojson.org/book), which shows how to do the same with `WiFiClientSecure`.

Best regards,
Benoit

rberkelm, 2023-12-14:
Hey Benoit,
Thanks for the great tip and wonderful resources - absolute gold and just
what I need! I'll work through it carefully.
Ray

rberkelm, 2023-12-18:
Reporting success with WiFiClientSecure as per the example in Chapter 8.3 of your book - thank you! However, SolarEdge doesn't like " HTTP/1.1" in the path for some reason... No matter, all good!

bblanchon, 2023-12-18:
You're welcome!
Thank you very much for purchasing my book.

closed 2023-12-18 by bblanchon

rberkelm, 2024-01-13:
hey @bblanchon ,
I know you closed this thread, but I've just hit a snag. The function that worked for me last month, now won't work. My serial array is either empty and I get a "EmptyInput" error or I get a single character (either a "{" or a "H") and I get "IncompleteInput" error. I don't recall making any changes and my cURL in Postman still works.
I've worked my way through your Troubleshooting chapter and can eliminate most potential causes, but some remain, e.g. the "chunking" issue. The SolarEdge API insists on HTTP1.1 and chunked data. Could this be it? I'm not convinced as I'd expect to see at least the first chunk of data? I mostly see nothing. The connection itself to SolarEdge is fine. I've increased the Timeout and buffer size, but no luck. Also slowed down the baudrate to 9600. Could I please impose on your time again? My function below:
```
void getSolarEdge(){
  WiFiClientSecure client;
  client.setInsecure(); // disable certificate check
   Serial.println(F("Connecting..."));

  // Connect to HTTP server
  client.setTimeout(10000);
  if (!client.connect("monitoringapi.solaredge.com", 443)) {
    Serial.println(F("Connection failed"));
    return;
  }

  Serial.println(F("Connected to SE!")); 

  // Send HTTP request
  Serial.println(F("Sending request..."));
  client.println(F("GET monitoringapi.solaredge.com"));
  client.println(F("Host: /site/2807673/currentPowerFlow?api_key=XXXXXX"));
  client.println(F("User-Agent: Arduino")); 
  client.println(F("Accept: application/json"));
  client.println(F("Connection: close"));
  client.println();

  // skip headers
  client.find("\r\n\r\n");

  // Check HTTP status
  char data[512] = {0};
  client.readBytesUntil('\r', data, sizeof(data));
  Serial.printf ("Received data: %s \r\n", data);
  client.stop();
  
  Serial.println(F("Parsing..."));

  // Allocate the JsonDocument in the heap. 
  DynamicJsonDocument doc(512);
  // Deserialize the JSON document in the response
    auto err = deserializeJson(doc, data);
    if (err) {
      Serial.print(F("deserializeJson() failed with code "));
      Serial.println(err.c_str());
      client.stop();
      return;
    }

  // Extract values etc
}
```

Output
```
15:57:48.993 -> Connecting...
15:57:51.713 -> Connected to SE!
15:57:51.713 -> Sending request...
15:58:12.386 -> Received data:  
15:58:12.386 -> Parsing...
15:58:12.386 -> deserializeJson() failed with code EmptyInput
```

rberkelm, 2024-01-14:
@bblanchon 
I picked up a paste error between my Host and GET urls... Also removed the client.find("\r\n\r\n") line and now I get
```
16:32:51.304 -> Connecting...
16:32:53.532 -> Connected to SE!
16:32:53.532 -> Sending request...
16:32:57.162 -> Response: HTTP/1.1 200  
16:32:57.162 -> Parsing...
16:33:07.147 -> deserializeJson() failed with code EmptyInput
```
So, my request is successful! I'm just not capturing it. That will give me something to work on.

rberkelm, 2024-01-15:
I couldn't make it work with the WiFiClientSecure library as I just couldn't work out how to capture the stream. However, success with the ESP8266HTTPClient library, thanks to the http.getStream() function.
```
#define REQUEST_HOST "monitoringapi.solaredge.com"
#define REQUEST_PATH "/site/2807673/currentPowerFlow?api_key=XXXX"

void getSolarEdge(){
  WiFiClientSecure client;
  client.setInsecure(); // disable certificate check
   Serial.println(F("Connecting..."));

  // Connect to HTTP server
  HTTPClient http;
  http.begin(client, F(REQUEST_HOST),443,REQUEST_PATH);
  http.useHTTP10(true);
    // start connection and send HTTP header
    int httpCode = http.GET();
    if (httpCode > 0) {
      Serial.printf("[HTTP] GET... code: %d\n", httpCode);

      // if file found at server
      if (httpCode == HTTP_CODE_OK) {

      // Get a reference to the stream
      Stream &response = http.getStream();
          
     // Allocate the JsonDocument in the heap.
      DynamicJsonDocument doc(512);

      // Parse the JSON document from the HTTP response
      DeserializationError err = deserializeJson(doc, response);
            if (err) {
              Serial.print(F("Parsing failed: "));
              Serial.println(err.c_str());
              return;
            }
        //Extract values etc
}
```
Thanks again for your wonderful libraries and awesome book - I have learnt so much!

bblanchon, 2024-01-15:
You're welcome!
Congratulations on finding a solution!