bblanchon/ArduinoJson #2014: Using ArduinoJSON in ESP-IDF

j54j6, 2023-12-22:
**Describe the issue**  
A clear and concise description of what you're trying to do.
You don't need to explain every aspect of your project: focus on the problem you're having.

=> Currently I try to migrate an old Libary of mine to ESP IDF . I have installed the ArduinoJSON Component from IDF Registry and tried to use the old code (working in Arduino/PlattformIO). If I try to compile my modified function for a simple deserialization I get the following error (Pasted in the corresponding section).

**Troubleshooter report**  
Here is the report generated by the [ArduinoJson Troubleshooter](https://arduinojson.org/v6/troubleshooter/):  
=> Report: 
1. The program uses ArduinoJson 7
2. The issue happens at compile time
3. The error is not in the list

**Environment**  
Here is the environment that I'm using':
* Microconroller: ESP32-C6
* Core/runtime: ESP-IDF
* IDE: VS-Code 1.85 (Rel. 12-06-2023)

**Reproduction**  
Here is a small snippet that demonstrate the problem.

```c++
//DynamicJsonDocument doc(1024);

//DeserializationError error = deserializeJson(doc, "{\"hello\":\"world\"}");

// insert code here
=> Code
//From .h File

//Name and type definitions - for simpler understanding
//Typedef for FS State - used to give an overview over the overall fs state
typedef int fs_mgmt_err_t;

#define FS_MGMT_OK                  1
#define FS_MGMT_ERROR               -1

#define FS_MGMT_NOT_MOUNTED         0x100

#define FS_MGMT_FILE_NOT_FOUND      0x101
#define FS_MGMT_INVALID_ARGUMENT    0x102
#define FS_MGMT_BUSY_RESSOURCE      0x103 //USed if a file or directory is already opened for internal file/dir handling
#define FS_MGMT_INCONSISTENT_STATE  0x104
#define FS_MGMT_FILE_EXIST          0x105
#define FS_MGMT_INVALID_OPERATION   0x106

//Function return template
template <typename returnType> 
struct fs_mgmt_value_carrier_t { 
  fs_mgmt_err_t status; 
  returnType *value; 
}; 

//Type def for fs info
typedef struct {
    size_t total = 0;
    size_t used = 0;
    bool mounted = false;
    const char* partition_base_path = NULL;
    fs_mgmt_err_t status = FS_MGMT_ERROR;
} fs_mgmt_fs_info_t;

//From .cpp File
fs_mgmt_value_carrier_t<DynamicJsonDocument> FS_Mgmt::readJsonFile(const char* filename)
{
    fs_mgmt_value_carrier_t<DynamicJsonDocument> returnVal;
    const size_t capacity = JSON_OBJECT_SIZE(25) + 400;
    DynamicJsonDocument jsonDocument(capacity);
    if(this->fs_check() != FS_MGMT_OK)
    {
        returnVal.status = FS_MGMT_ERROR;
        return returnVal;
    }
    if(this->fs_fExist(filename) != FS_MGMT_FILE_EXIST)
    {
        returnVal.status = FS_MGMT_FILE_NOT_FOUND;
        return returnVal;
    }
    fs_mgmt_value_carrier_t<std::string> content = this->fs_fRead(filename);
    if(content.status != FS_MGMT_OK)
    {
        returnVal.status = FS_MGMT_ERROR;
        return returnVal;
    }

//This line makes trouble - If removed it compiles... - content.value is type std::string
    DeserializationError error = deserializeJson(jsonDocument, content.value);
    if(error)
    {
        returnVal.status = FS_MGMT_ERROR;
        return returnVal;
    }
    
    returnVal.status = FS_MGMT_OK;
    returnVal.value = &jsonDocument;
    return returnVal;
}
```

**Compile Error**
```

E:/justin/PrivatProjekte/CodeStuff/C-C++/ESP-IDF/ESP-C6_OS/managed_components/bblanchon__arduinojson/src/ArduinoJson/Deserialization/Reader.hpp: In instantiation of 'int ArduinoJson::V6214HB2::detail::Reader<TSource, Enable>::read() [with TSource = std::__cxx11::basic_string<char>*; Enable = void]':
E:/justin/PrivatProjekte/CodeStuff/C-C++/ESP-IDF/ESP-C6_OS/managed_components/bblanchon__arduinojson/src/ArduinoJson/Json/Latch.hpp:38:25:   required from 'void ArduinoJson::V6214HB2::detail::Latch<TReader>::load() [with TReader = ArduinoJson::V6214HB2::detail::Reader<std::__cxx11::basic_string<char>*, void>]'
E:/justin/PrivatProjekte/CodeStuff/C-C++/ESP-IDF/ESP-C6_OS/managed_components/bblanchon__arduinojson/src/ArduinoJson/Json/Latch.hpp:30:7:   required from 'char ArduinoJson::V6214HB2::detail::Latch<TReader>::current() [with TReader = ArduinoJson::V6214HB2::detail::Reader<std::__cxx11::basic_string<char>*, void>]'
E:/justin/PrivatProjekte/CodeStuff/C-C++/ESP-IDF/ESP-C6_OS/managed_components/bblanchon__arduinojson/src/ArduinoJson/Json/JsonDeserializer.hpp:48:26:   required from 'char ArduinoJson::V6214HB2::detail::JsonDeserializer<TReader, TStringStorage>::current() [with TReader = ArduinoJson::V6214HB2::detail::Reader<std::__cxx11::basic_string<char>*, void>; TStringStorage = ArduinoJson::V6214HB2::detail::StringMover]'
E:/justin/PrivatProjekte/CodeStuff/C-C++/ESP-IDF/ESP-C6_OS/managed_components/bblanchon__arduinojson/src/ArduinoJson/Json/JsonDeserializer.hpp:72:13:   required from 'ArduinoJson::V6214HB2::DeserializationError::Code ArduinoJson::V6214HB2::detail::JsonDeserializer<TReader, TStringStorage>::parseVariant(ArduinoJson::V6214HB2::detail::VariantData&, TFilter, ArduinoJson::V6214HB2::DeserializationOption::NestingLimit) [with TFilter = ArduinoJson::V6214HB2::detail::AllowAllFilter; TReader = ArduinoJson::V6214HB2::detail::Reader<std::__cxx11::basic_string<char>*, void>; TStringStorage = ArduinoJson::V6214HB2::detail::StringMover]'
E:/justin/PrivatProjekte/CodeStuff/C-C++/ESP-IDF/ESP-C6_OS/managed_components/bblanchon__arduinojson/src/ArduinoJson/Json/JsonDeserializer.hpp:36:23:   required from 'ArduinoJson::V6214HB2::DeserializationError ArduinoJson::V6214HB2::detail::JsonDeserializer<TReader, TStringStorage>::parse(ArduinoJson::V6214HB2::detail::VariantData&, TFilter, ArduinoJson::V6214HB2::DeserializationOption::NestingLimit) [with TFilter = ArduinoJson::V6214HB2::detail::AllowAllFilter; TReader = ArduinoJson::V6214HB2::detail::Reader<std::__cxx11::basic_string<char>*, void>; TStringStorage = ArduinoJson::V6214HB2::detail::StringMover]'
E:/justin/PrivatProjekte/CodeStuff/C-C++/ESP-IDF/ESP-C6_OS/managed_components/bblanchon__arduinojson/src/ArduinoJson/Deserialization/deserialize.hpp:48:13:   required from 'ArduinoJson::V6214HB2::DeserializationError ArduinoJson::V6214HB2::detail::deserialize(ArduinoJson::V6214HB2::JsonDocument&, TStream&&, Args ...) [with TDeserializer = JsonDeserializer; TStream = std::__cxx11::basic_string<char>*&; Args = {}; <template-parameter-1-4> = void]'
E:/justin/PrivatProjekte/CodeStuff/C-C++/ESP-IDF/ESP-C6_OS/managed_components/bblanchon__arduinojson/src/ArduinoJson/Json/JsonDeserializer.hpp:689:39:   required from 'ArduinoJson::V6214HB2::DeserializationError ArduinoJson::V6214HB2::deserializeJson(JsonDocument&, TChar*, Args&& ...) [with TChar = std::__cxx11::basic_string<char>; Args = {}]'
E:/justin/PrivatProjekte/CodeStuff/C-C++/ESP-IDF/ESP-C6_OS/components/FS-Mgmt/FS-Mgmt.cpp:631:49:   required from here
E:/justin/PrivatProjekte/CodeStuff/C-C++/ESP-IDF/ESP-C6_OS/managed_components/bblanchon__arduinojson/src/ArduinoJson/Deserialization/Reader.hpp:22:21: error: request for member 'read' in '*((ArduinoJson::V6214HB2::detail::Reader<std::__cxx11::basic_string<char>*, void>*)this)->ArduinoJson::V6214HB2::detail::Reader<std::__cxx11::basic_string<char>*, void>::source_', which is of pointer type 'std::__cxx11::basic_string<char>*' (maybe you meant to use '->' ?)
   22 |     return source_->read();  // Error here? See https://arduinojson.org/v6/invalid-input/
      |            ~~~~~~~~~^~~~
ninja: build stopped: subcommand failed.
ninja failed with exit code 1, output of the command is in the E:\justin\PrivatProjekte\CodeStuff\C-C++\ESP-IDF\ESP-C6_OS\build\log\idf_py_stderr_output_20048 and E:\justin\PrivatProjekte\CodeStuff\C-C++\ESP-IDF\ESP-C6_OS\build\log\idf_py_stdout_output_20048

```

**Program output**  
Not relevant - Error at compile time

Expected output:

```
Not relevant - Error at compile time
```

Actual output:

```
Not relevant - Error at compile time
```

Thanky for your time and help :)

bblanchon, 2023-12-23:
Hi @j54j6,

```text
error: request for member 'read' in '...', which is of pointer type 'std::__cxx11::basic_string<char>*' 
```

The error says you passed a pointer to `std::string` to `deserializeJson()`.
Indeed, `content.value` is a `std::string*` (despite what the comment says :wink:).

Best regards,
Benoit

j54j6, 2023-12-23:
Hey @bblanchon, 

thank you so much - guess I did not understand the error message quite right ^^ - I dereferenced the value and it works like a charm. Thank you so much for your time and have a nice day :)

Best regards
Justin

closed 2023-12-23 by j54j6