bblanchon/ArduinoJson #2015: Help on optimization

ScreamZ, 2023-12-23:
Hello,

I'm really new to cpp and coming from nodeJS / React background with JavaScript.

Currently, I'm connected to a websocket and I receive the payload as follows with a `uint8_t`

```ino
StaticJsonDocument<1024> stateExtractor; // Used to extract WS data into JSON. Adjust size depending on doc size.
JsonObject deviceState; // The synced device state

void onServerEvent(uint8_t *payload, size_t length) {
  // Ignore non-document notification of device state
  if (strstr((char *)payload, ("\"_id\":\"" + deviceID + "\"").c_str()) == NULL)
    return;

  StaticJsonDocument<48> filter;
  filter["result"]["_source"] = true;
  Serial.println((char *)payload);

  DeserializationError error = deserializeJson(stateExtractor, (const char*)payload, length, DeserializationOption::Filter(filter));
  Serial.println(error.c_str());
  deviceState = stateExtractor["result"]["_source"];
}
```

I've been forced to use `(const char*)payload` otherwise as `payload` is a pointer, the value was changed on every new WebSocket message.

But it's written that this is a bad practice to use char*, what do you suggests? I've read about buffer in your doc ? Not sure how to use it ?

Regards

bblanchon, 2023-12-23:
Hi @ScreamZ,

The documentation doesn't mention it, but [`deserializeJson()`](https://arduinojson.org/v6/api/json/deserializejson/) makes no difference between `uint8_t*` and `char*`.

In ArduinoJson 6, `deserializeJson()` treats `char*` and `const char*` differently.
`char*` enables the zero-copy mode, whereas `const char*` doesn't.
In the zero-copy mode, the `JsonDocument` stores pointers to the input buffer instead of making copies.
In your case, using the zero-copy mode is safe since the `JsonDocument` is scoped to the callback function, so you should use `char*`.

Best regards,
Benoit

ScreamZ, 2023-12-23:
Hi Benoit, thanks for your answer.

For some reason using char* results in a crash, I suspect that's because my `payload` is changing and sometimes it's a different JSON. (see the early return statement). 

Using a const without zero-copy mode makes it work. I'll stick with this for now.

closed 2023-12-23 by ScreamZ

bblanchon, 2023-12-24:
I'm sorry; I focused so much on the original question about `char*` that I completely missed the fact that your document was global. Indeed, in this case, you must **not** use the zero-copy mode.

By the way, I removed the zero-copy mode in ArduinoJson 7, so you can remove the cast after updating.

Thank you very much for purchasing my book; your support is greatly appreciated ❤️

ScreamZ, 2023-12-24:
@bblanchon That's a HUGE change in paradigm (you previously recommended to use stack instead of heap as I'm reading book 6)

I'm starting a new project on ESP32, is it stable or release candidate to use 7?

bblanchon, 2023-12-25:
Yes, I recommend that you start with version 7 right away.