bblanchon/ArduinoJson #2017: DynamicJSONDocument as Template Parameter

j54j6, 2023-12-26:
**Describe the issue**  
Hey, 
I currently want to implement a function that returns a JSON Documents from a file. However If I use Pointer it doesn't work. The moment the function returns to main I get an empty document back :/
For better readability I don't want to use pass by Reference (as parameter of the function). I have a custom return type using a template (shown in the following code section). If I want to use the plain DynamicJSONDocument I can't use it because of a "deleted function". 
If there is a solution to increase the "lifespan" after the function ends so I can use it in my main(), this would also fit but I can't find anything to my problem. In the followingis the version using a pointer - compiles fine but dont work when returned to main.
 
1. Version is using Pointer as return type - Right before return is the JSON Document working as intended. I can get the value for my key "test_key" but after returning to main() the JSON Document is empty (if serialized by ArduinoJSON it shows "{}"). -> **This would be my preferred solution but I can't get it to work**. 

2. Version should use a "copy" of the DynamicJsonDocument but this can not compile  (Error: `the default constructor of "fs_mgmt_value_carrier_t<ArduinoJson::V6214HB2::DynamicJsonDocument>" cannot be referenced -- it is a deleted functionC/C++(1790)`)

**Troubleshooter report**  
Here is the report generated by the [ArduinoJson Troubleshooter](https://arduinojson.org/v6/troubleshooter/):  
1. The program uses ArduinoJson 7
4. The issue happens at compile time
5. The error is not in the list

**Environment**  
Here is the environment that I'm using':
* Microconroller: ESP32c6
* Core/runtime: ESP-IDF
* IDE: VS-Code 1.85 (Rel. 12-06-2023)

**Reproduction**  
Here is a small snippet that demonstrate the problem.

```c++
-> From .h file
//Function return template
template <typename returnType> 
struct fs_mgmt_value_carrier_t { 
  fs_mgmt_err_t status; 
  returnType value; 
}; 
fs_mgmt_value_carrier_t<DynamicJsonDocument*> readJsonFile_pointed(const char* filename);

-> From .cpp file

//This function is used to get an useable JSONFile for further useage. If there is an error the value_carrier_t status is != FS_MGMT_OK
fs_mgmt_value_carrier_t<DynamicJsonDocument*> FS_Mgmt::readJsonFile_pointed(const char* filename)
{
    const size_t capacity = JSON_OBJECT_SIZE(25) + 400;
    DynamicJsonDocument jsonDocument(capacity);
    fs_mgmt_value_carrier_t<DynamicJsonDocument*> returnVal;
    //If I use a non pointer DynmaicJSONDocument I will get the error described in the request and it will not compile.
    if(this->fs_check() != FS_MGMT_OK)
    {
        returnVal.status = FS_MGMT_ERROR;
        return returnVal;
    }
    if(this->fs_fExist(filename) != FS_MGMT_FILE_EXIST)
    {
        returnVal.status = FS_MGMT_FILE_NOT_FOUND;
        return returnVal;
    }
    fs_mgmt_value_carrier_t<std::string*> content = this->fs_fRead(filename);
    if(content.status != FS_MGMT_OK)
    {
        returnVal.status = FS_MGMT_READ_ERROR;
        return returnVal;
    }

    std::string file_content = *content.value;
    DeserializationError error = deserializeJson(jsonDocument, file_content);
    if(error)
    {
        ESP_LOGE(this->logger_tag, "Error while reading JSON File! - Error: %s", error.c_str());
        returnVal.status = FS_MGMT_JSON_DES_ERROR;
        return returnVal;
    }
    
    //If I check here for the key I can get the value without any problems.
    if(jsonDocument.containsKey("test_key"))
    {
        std::cout << "Content of key \"test_key\" is: " << jsonDocument["test_key"] << std::endl;
    }
    
    returnVal.status = FS_MGMT_OK;
    returnVal.value = &jsonDocument;
    //After returning to main I will only get an empty document. 
    return returnVal;
}

=> main.cpp File 
//This file contains the following content: >> {"test_key": "hallo"} <<
   fs_mgmt_value_carrier_t<DynamicJsonDocument*> result = test.readJsonFile_pointed("/littlefs/example.txt");

   DynamicJsonDocument* json_data = result.value;
   //3C - Modified to match for both functions - value_carrier and DynamicJSONDocument as return Type (for dev purposes only)
   if(true)
   {
      ESP_LOGI(TAG, "File readed");
      
      bool exist = json_data->containsKey("test_key");
      if (exist)
      {
         ESP_LOGI("MAIN", "Key exists in file!");
         return;

      }
      else {
         //Key does not exist here! - That's my problem ^^
         ESP_LOGE("MAIN", "The specified key does not exist!");
         std::string doc;
         
         //Is there a way to use this function with a pointer? - and how can I get the values with a pointer without dereferencing? 
         //serializeJsonPretty(json_data, doc);
         std::cout << "Data of DOc: " << doc << std::endl;
      }
       //This don't work with pointers but never the less, the document is empty...
      //std::string res_data = json_data["test_key"];
      //ESP_LOGI(TAG, "The value of key is: %s", res_data.c_str());
   }
   else
   {
      ESP_LOGE(TAG, "Error while reading JSON file!");
   }
```

**Program output**  
If relevant, include the program output.

Expected output:

```
E (518) MAIN: Key exists in file!
[...]
```

Actual output:

```
E (518) MAIN: The specified key does not exist!
```

bblanchon, 2023-12-28:
Hi @j54j6,

`DynamicJsonDocument` is a class from ArduinoJson 6.
On ArduinoJson 7, you should use `JsonDocument`.

Best Regards,
Benoit

j54j6, 2023-12-28:
Hey @bblanchon,

thank you again for your great help. Using jsonDocument allows me to pass the jsonDocument as a copy.

Best regards,
Justin

closed 2023-12-28 by j54j6