bblanchon/ArduinoJson #2019: [7.0.0] Using JsonVariantConst causes "no matching function" error

BlueAndi, 2024-01-03:
Example code for reproduction:
```cpp
    JsonDocument    jsonDoc;
    JsonObject      jsonAbc = jsonDoc["abc"].add<JsonObject>();

    jsonAbc["a"] = "a";
    jsonAbc["b"] = "b";
    jsonAbc["c"] = "c";

    {
        JsonObjectConst     value = jsonDoc["abc"];
        JsonVariantConst    jsonA = value["a"]; /* ArduinoJson/Object/JsonObjectConst.hpp:105:38: error: no matching function for call to 'ArduinoJson::V700HB4::detail::ObjectData::getMember(const ArduinoJson::V700HB4::detail::ObjectData* const&, ArduinoJson::V700HB4::detail::StringAdapter<const char*, void>::AdaptedString)' */
    }
```

bblanchon, 2024-01-03:
Hi @BlueAndi,

Thank you very much for reporting this bug :heart:
I'll publish the fix in the next few days.

Best regards,
Benoit

closed 2024-01-03 by bblanchon

BlueAndi, 2024-01-03:
Hi @bblanchon, you are welcome. Happy new year Benoit and thanks for the great work and your effort!

bblanchon, 2024-01-10:
The fix was published with [ArduinoJson 7.0.1](https://github.com/bblanchon/ArduinoJson/releases/tag/v7.0.1).

commit cc423733 (2024-01-03): Fix "no matching function" with `JsonObjectConst::operator[]`

Fixes #2019