bblanchon/ArduinoJson #2022: CollectionIterator crash

zekageri, 2024-01-05:
So i have a crash most likely when i call this function

```cpp
boolean User::getUserBySession(const char* sessionID, JsonObject& user) {
    JsonDocument users; //(USERS_JSON_SIZE);
    if (db.read(USERS_PATH, users) == -1) {
        return false;
    }
    Serial.println("Iterating over users!");
    serializeJsonPretty(users, Serial); // This line is printed fine
    Serial.println();

    for (JsonPair _user : users.as<JsonObject>()) {
        if (hshSystem.isCharArrEqual(_user.value()["sessionID"], sessionID)) {
            user = _user.value();
            return true;
        }
    }
    return false; // Never returns
}
```

**Serial:**

```cpp
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x400d6b6a  PS      : 0x00060630  A0      : 0x800d7ca0  A1      : 0x3ffdd740
A2      : 0x00000001  A3      : 0x3ffdd768  A4      : 0x000000ff  A5      : 0x0000ff00
A6      : 0x00ff0000  A7      : 0x40404040  A8      : 0x0000ffff  A9      : 0x3ffdd720
A10     : 0x00000010  A11     : 0x00000010  A12     : 0x00000002  A13     : 0x0000ff00
A14     : 0x00ff0000  A15     : 0xff000000  SAR     : 0x00000011  EXCCAUSE: 0x0000001c
EXCVADDR: 0x0000001a  LBEG    : 0x4008c079  LEND    : 0x4008c09d  LCOUNT  : 0x8013de18  

Backtrace: 0x400d6b67:0x3ffdd740 0x400d7c9d:0x3ffdd770 0x400d7c02:0x3ffdd7a0 0x4012c9ea:0x3ffdd7d0 0x4012ca65:0x3ffdd870 0x401518e9:0x3ffdd890 0x40155399:0x3ffdd900 0x4019cbfd:0x3ffdd980 0x4019b9f4:0x3ffdd9b0 0x4019c222:0x3ffdda50 0x4019adc4:0x3ffdda70 0x4023787a:0x3ffdda90 0x4019aee9:0x3ffddab0

  #0  0x400d6b67:0x3ffdd740 in ArduinoJson::V700PB2::detail::CollectionIterator::CollectionIterator(ArduinoJson::V700PB2::detail::VariantSlot*, unsigned short) at lib/ArduinoJson-7.x/src/ArduinoJson/Collection/CollectionImpl.hpp:17
      (inlined by) ArduinoJson::V700PB2::detail::CollectionData::createIterator(ArduinoJson::V700PB2::detail::ResourceManager const*) const at lib/ArduinoJson-7.x/src/ArduinoJson/Collection/CollectionData.hpp:88
  #1  0x400d7c9d:0x3ffdd770 in ArduinoJson::V700PB2::detail::JsonSerializer<ArduinoJson::V700PB2::detail::StaticStringWriter>::visit(ArduinoJson::V700PB2::detail::ObjectData const&) at lib/ArduinoJson-7.x/src/ArduinoJson/Json/JsonSerializer.hpp:44

ELF file SHA256: 86722447681a2dd7

Rebooting...
```

This is how i call it

```cpp
JsonDocument userDoc;
JsonObject userObj = userDoc.to<JsonObject>();
if( !user.getUserBySession(sessionID.c_str(),userObj) ){
    // Not authenticated...
}
```

zekageri, 2024-01-05:
It does not crash if i remove the for loop

```cpp
for (JsonPair _user : users.as<JsonObject>()) {
    if (hshSystem.isCharArrEqual(_user.value()["sessionID"], sessionID)) {
        user = _user.value();
        return true;
    }
}
```

zekageri, 2024-01-05:
Trouble shooter wont help

[image]

It is not serialization and not deserialization

zekageri, 2024-01-05:
The problem is with this line

`user = _user.value();`

If i comment it out the program does not crash.

zekageri, 2024-01-05:
It works if i start to copy the properties one by one

```cpp
for (JsonPair _user : users.as<JsonObject>()) {
    if (hshSystem.isCharArrEqual(_user.value()["sessionID"].as<const char*>(), sessionID)) {
        //user = _user.value();
        user["sessionID"] = _user.value()["sessionID"];
        user["registered"] = _user.value()["registered"];
        return true;
    }
}
```

bblanchon, 2024-01-06:
Hi @zekageri,

The problem is that the `userObj` points to a destructed document.
This is because the function takes `user` by reference and then assigns it to a member of the temporary document.

To fix this issue, you must call [`JsonObject::set()`](https://arduinojson.org/v7/api/jsonobject/set/) instead.

```diff
- user = _user.value();
+ user.set(_user.value());
```

Never pass `JsonArray`, `JsonObject`, and `JsonVariant` by reference. They already have reference semantics; passing them by reference only adds confusion.

If it was working with ArduinoJson 6, it was by a happy accident.

Best regards,
Benoit

zekageri, 2024-01-06:
Well, I pass by ref in my project all the time. But good to know ;D will test it and report back! Thanks

closed 2024-01-19 by bblanchon

zekageri, 2024-01-19:
Sorry, it is working properly! Thank you very much.