bblanchon/ArduinoJson #2024: Error with timestamp

ScreamZ, 2024-01-05:
**Describe the issue**  
I'm trying to extract a timestamp from a JSON document, but it doesn't work.

Added 
```
#define ARDUINOJSON_USE_LONG_LONG 1
#include <ArduinoJson.h>
```

above my lib

Here is the environment that I'm using':
* ArduinoJson 7
* Microconroller: ESP32
* Core/runtime: [e.g. ESP32 core for Arduino]
* IDE: VSCode

```cpp
   String test;
   serializeJson(content["result"]["_source"]["_kuzzle_info"]["updatedAt"], test);
   Serial.println(test);
   Serial.println(content["result"]["_source"]["_kuzzle_info"]["updatedAt"].as<unsigned long>());
   latestStateSync = content["result"]["_source"]["_kuzzle_info"]["updatedAt"].as<unsigned long>();
```

Output
```
1704484728221
0
```

Looks like the same happen when you want to serialize millis

like 

```c++
JsonDocument test
test["value"] = millis();
```

At some point, it doesn't update.

Do you have any intel on that ?

Thanks

bblanchon, 2024-01-08:
Hi @ScreamZ,

On ESP32, `unsigned long` is 32 bits wide, so it can only go up to  4,294,967,295.
Please try with `unsigned long long`, which is 64 bits wide.

`millis()` returns an `unsigned long`, so it should be fine.

Best regards,
Benoit

ScreamZ, 2024-01-08:
Works better, nice catch, thanks.

Any way to use `time_t now;` from `#include <time.h>` from time library with arduinoJson casting ?

bblanchon, 2024-01-09:
`time_t` is a typedef of `long`, so it should not require special treatment.

closed 2024-01-09 by ScreamZ