bblanchon/ArduinoJson #2027: ArduinoJson 7 uses ~30 kb more stack on Teensy 4.1

Greg-R, 2024-01-08:
By upgrading to verson 7.0, about 30 kB more stack (RAM1) is used.  I am working with a Teensy 4.1 in Arduino IDE 2.1.1.

**Environment**  
Here is the environment that I'm using':
* Microconroller: Teensy 4.1
* Core/runtime: Teensyduino 1.58
* IDE: Arduino IDE 2.2.1

Compilation report using version 6.21.4:

Memory Usage on Teensy 4.1:
  FLASH: code:244616, data:123488, headers:8724   free for files:7749636
   RAM1: variables:202912, code:228792, padding:584   free for local variables:92000   <---
   RAM2: variables:401920  free for malloc/new:122368

Now install version 7.0 ArduinoJson and re-compile:

Memory Usage on Teensy 4.1:
  FLASH: code:260088, data:123488, headers:8612   free for files:7734276
   RAM1: variables:202912, code:229944, padding:32200   free for local variables:59232  <---
   RAM2: variables:401920  free for malloc/new:122368

Perhaps this is normal for version 7 to consume more stack?  Note that the "padding" seems to be the source of the increase.
My project is here:
https://github.com/Greg-R/T41EEE
https://github.com/Greg-R/T41EEE/tree/main/T41EEE

The code is in the file JSON.cpp.  The code is derived from the configuration file example.

Regards,
Greg

bblanchon, 2024-01-09:
Hi Greg,

No, ArduinoJson 7 doesn't require more stack memory but has a bigger code.
As you can see, the amount of code in RAM1 increased by 1152B.

The hardware imposes that data starts on a 32K boundary, so there must be some unused bytes between code and data.
That's the padding.
By definition, it ranges from 0 to 32,767.

You were initially very lucky because the padding was very small.
But then you were very _unlucky_ because you got a padding close to the maximum.
The extra 1152B made the code cross the 32KB boundary, requiring another 32KB block.

If you need, you can recover these 32KB by saving 567B of code.

See also: [what is "padding"?](https://forum.pjrc.com/index.php?threads/code-compilation-what-is-padding.68168//)

Best regards,
Benoit

Greg-R, 2024-01-09:
Thank you for the detailed explanation!  This is good news, because I know where I can save at least 567B of code.
It is also very interesting and a good thing to understand this detail of memory allocation.

Thank you,
Greg

bblanchon, 2024-01-09:
You're welcome, Greg!
Thank you for using ArduinoJson.
Don't forget to [cast a star](https://github.com/bblanchon/ArduinoJson/stargazers) to support the project :wink:

Greg-R, 2024-01-09:
The library is essential to my project.  It is very easy to use, and now version 7 is even easier.
Super good library!

Thank you,
Greg