bblanchon/ArduinoJson #2030: Losing const char* data while using doc["fileName"].set(cFileName);

luukjansen, 2024-01-09:
**Description**
I pass a const char* to a JsonDocument (that becomes part of a larger JsonDocument) and thought that the documentation said that if I user .set() the const char* is copied rather than using the reference. However, while the value is there (I print it to test) when added to the JsonDocument, a little down the line I get garbage for that value. When I create a char[], copy the content in, use .set() then it works, but it seems like an extra copy operation not needed.

**Troubleshooter's report**
1. The program uses ArduinoJson 7
2. The issue happens at run time
3. The issue concerns serialization
4. Output contains garbage
5. `serializeJson()` produces garbage
6. Program calls `serializeJson(const JsonDocument&, ...)`
7. Program doesn't use `String`

**Environment**
* Microcontroller: ESP32
* Core/Framework: ESP32-D0WD-V3 (revision v3.0) Sdk version: v4.4.5
* IDE: Arduino IDE 2.2.1

bblanchon, 2024-01-11:
Hi @luukjansen,

> the documentation said that if I user .set() the const char* is copied rather than using the reference.

No.
[JsonVariant::set()](https://arduinojson.org/v7/api/jsonvariant/set/) stores `const char*` by pointer and other strings by copy.
If you see some documentation that says otherwise, please tell me the URL so I can fix it immediately.

If you need further assistance, please provide an [MCVE](https://stackoverflow.com/help/mcve).

Best regards,
Benoit

luukjansen, 2024-01-11:
Thank you. I got it from the page you reference above. I interpreted the text under "String Duplication" to mean it will 'copy' rather than reference: _ArduinoJson makes a copy of the string when you call this function with one of the following types... etc._

However, I missed the point earlier: _"As usual, ArduinoJson makes a copy of a string in the JsonDocument, except if it’s a const char*."_

At the moment I create a char[] and then strcpy to populate the char from the char*.
Researching the above I saw that in your book you do sat explicitly that .set() references, and show using  _doc.set(String("hello"))_ (p177) as an option if I understand it correctly. 

Would there be a big performance hit by moving towards using that?

bblanchon, 2024-01-13:
Thank you very much for the feedback.
I updated the documentation; it should be less confusing now.

If you have a `const char*` but need to store a copy, you should cast the string pointer to `char*`:

```c++
var.set(const_cast<char*>(str));
```

This is unpleasant; hopefully, you won't have to do that too often.

luukjansen, 2024-01-13:
@bblanchon perfect, thank you. The documentation is very clear now. I only have to do it on 5 lines in my code, so no sweat.

closed 2024-01-14 by bblanchon