bblanchon/ArduinoJson #2033: [v7] `CappedAllocator` for ArduinoJson 7

mathieucarbou, 2024-01-10:
The switch to a dynamically allocated buffer bothers us because we have pretty big use case (streaming) where functions are populating a json document to be sent to Web Socket.

The problem is that the whole Json document can take up to 16-32k so we have to split it in batches sent consecutively.

So we are relying on the capacity of a Json document to know when it overflows, to send the current json document in the web socket channel, then clear the document and continue the loop so that the application can continue populating the next json objects in the document.

This "streaming" approach allows us to limit the memory usage while not knowing in advance the complete size of the document and number of batches.

Switching to v7 does not allow such approach: we would need to re-implement an allocator, but the current API is incomplete for he reallocate part:

```c++
namespace ArduinoJson {
  class CappedAllocator : Allocator {
  
  public:
    explicit CappedAllocator(size_t capacity, Allocator *delegate = detail::DefaultAllocator::instance()) : _capacity(capacity), _delegate(delegate) {}

    void *allocate(size_t size) {
      if(_allocated + size > _capacity)
        return nullptr;
      void *p = _delegate->allocate(size);
      if(p)
        _allocated+=size;
      return p;
    }

    void deallocate(void *pointer) {
      _delegate->deallocate(pointer);
    }

    void *reallocate(void *ptr, size_t new_size) {
      // ???
    }

  private:
    const size_t _capacity;
    Allocator *_delegate;
    size_t _allocated = 0;
  };
}
```

What we would need is something like this in order to keep track of the capacity:

```c++
virtual void deallocate(void* ptr, size_t old_size) = 0;
virtual void* reallocate(void* ptr, size_t old_size, size_t new_size) = 0;
```

Looking at the ArduinoJson code base, each time `allocate` is called, there is a variable on caller side for the allocated capacity, so it should be possible to provide such interfaces, and even keep backward compatibility by providing stubs.

What do you thing ?

The current unacceptable and slow alternative I see would be to call `measureJson()` instead of `overflowed()`, and make sure the json document has enough capacity, and we send the batches once our json document reaches a size limit.

Is there otherwise another (fast) way to implement such steaming approach ?

Thanks a lot for this awesome library :-)

mathieucarbou, 2024-01-11:
I gave up on the allocator and implemented the `measureJson` way.

```
T [DASH] client=2, measureJson=612
T [DASH] client=2, measureJson=1047
T [DASH] client=2, measureJson=1067
T [DASH] client=2, measureJson=1031
T [DASH] client=2, measureJson=1078
T [DASH] client=2, measureJson=1063
T [DASH] client=2, measureJson=1040
T [DASH] client=2, measureJson=1069
T [DASH] client=2, measureJson=1082
T [DASH] client=2, measureJson=1079
T [DASH] client=2, measureJson=1059
T [DASH] client=2, measureJson=1050
T [DASH] client=2, measureJson=1030
T [DASH] client=2, measureJson=824
T [DASH] client=2, measureJson=1040
T [DASH] client=2, measureJson=159
```

Using ` measureJson` increases the publish time to websocket by a few ms so this is acceptable for a big use case like that.

bblanchon, 2024-01-12:
Hi @mathieucarbou,

I considered many times changing the allocator interface to include the old size as in `std::allocator`, but never found a real use case. I'm afraid "relying on the capacity of a Json document to know when it overflows, to send the current json document" isn't a strong use case either.

> Using  measureJson increases the publish time to websocket by a few ms so this is acceptable for a big use case like that.

The JSON document is fairly small.
Why don't you use a buffer and send it when it reaches a certain level?

```c++
char buffer[2048];
size_t len = serializeJson(doc, buffer);
if (len > 1024) {
  sendJson(buffer, len);
  doc.clear();
}
```

Best regards,
Benoit

mathieucarbou, 2024-01-12:
Actually, I am serialising into `AsyncWebSocketMessageBuffer* buffer = _ws->makeBuffer(size);` to avoid the memory copy happening (that's why I am measuring). That's always a tradeoff between speed and memory.  I will try allocating a buffer like that and see what happens. Thanks for the suggestion!

closed 2024-02-01 by bblanchon