bblanchon/ArduinoJson #2034: Switching from ArduinoJson 6 to 7 leads to a random assertion failure at runtime: `const VariantPoolList.hpp:98 (poolIndex < count_`

mathieucarbou, 2024-01-11:
I cannot use the the Arduino Json Troubleshooter since the issue happens when building a json document, when allocating memory.

I do not use any custom allocator.

Board is the standard ESP32 NodePCU Dev Kit from AZ-Delivery.

Stack trace:

```

Backtrace: 0x400838e5:0x3ffe9ae0 0x4008dbcd:0x3ffe9b00 0x40093719:0x3ffe9b20 0x400dd240:0x3ffe9c50 0x40124079:0x3ffe9cf0 0x40126375:0x3ffe9d10 0x4012880e:0x3ffea410 0x40128d31:0x3ffea490 0x401c579d:0x3ffea4c0 0x401c57c2:0x3ffea510 0x401c609d:0x3ffea540 0x401c6211:0x3ffea580 0x401c2d17:0x3ffea5a0 0x401c2d66:0x3ffea5d0 0x401c2da9:0x3ffea5f0 0x401c2f82:0x3ffea610 0x401c2ff5:0x3ffea630

  #0  0x400838e5:0x3ffe9ae0 in panic_abort at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/panic.c:408
  #1  0x4008dbcd:0x3ffe9b00 in esp_system_abort at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/esp_system.c:137
  #2  0x40093719:0x3ffe9b20 in __assert_func at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/newlib/assert.c:85
  #3  0x400dd240:0x3ffe9c50 in ArduinoJson::V701PB2::detail::VariantPoolList::getSlot(unsigned short) const at .pio/libdeps/pro-esp32-debug/ArduinoJson/src/ArduinoJson/Memory/VariantPoolList.hpp:98
      (inlined by) ArduinoJson::V701PB2::detail::VariantPoolList::allocFromFreeList() at .pio/libdeps/pro-esp32-debug/ArduinoJson/src/ArduinoJson/Memory/VariantPoolImpl.hpp:71
      (inlined by) ArduinoJson::V701PB2::detail::VariantPoolList::allocSlot(ArduinoJson::V701PB2::Allocator*) at .pio/libdeps/pro-esp32-debug/ArduinoJson/src/ArduinoJson/Memory/VariantPoolList.hpp:73
      (inlined by) ArduinoJson::V701PB2::detail::ResourceManager::allocSlot() at .pio/libdeps/pro-esp32-debug/ArduinoJson/src/ArduinoJson/Memory/ResourceManager.hpp:53
      (inlined by) ArduinoJson::V701PB2::detail::CollectionData::addSlot(ArduinoJson::V701PB2::detail::ResourceManager*) at .pio/libdeps/pro-esp32-debug/ArduinoJson/src/ArduinoJson/Collection/CollectionImpl.hpp:52
      (inlined by) ArduinoJson::V701PB2::detail::VariantData* ArduinoJson::V701PB2::detail::ObjectData::addMember<ArduinoJson::V701PB2::detail::StaticStringAdapter>(ArduinoJson::V701PB2::detail::StaticStringAdapter, ArduinoJson::V701PB2::detail::ResourceManager*) at .pio/libdeps/pro-esp32-debug/ArduinoJson/src/ArduinoJson/Object/ObjectData.hpp:27
      (inlined by) ArduinoJson::V701PB2::detail::VariantData* ArduinoJson::V701PB2::detail::ObjectData::getOrAddMember<ArduinoJson::V701PB2::detail::StaticStringAdapter>(ArduinoJson::V701PB2::detail::StaticStringAdapter, ArduinoJson::V701PB2::detail::ResourceManager*) at .pio/libdeps/pro-esp32-debug/ArduinoJson/src/ArduinoJson/Object/ObjectImpl.hpp:24
  #4  0x40124079:0x3ffe9cf0 in ArduinoJson::V701PB2::detail::VariantData* ArduinoJson::V701PB2::detail::VariantData::getOrAddMember<ArduinoJson::V701PB2::detail::StaticStringAdapter>(ArduinoJson::V701PB2::detail::StaticStringAdapter, ArduinoJson::V701PB2::detail::ResourceManager*) at .pio/libdeps/pro-esp32-debug/ArduinoJson/src/ArduinoJson/Variant/VariantData.hpp:226
  #5  0x40126375:0x3ffe9d10 in ArduinoJson::V701PB2::detail::MemberProxy<ArduinoJson::V701PB2::JsonDocument&, char const*>::getOrCreateData() const at .pio/libdeps/pro-esp32-debug/ArduinoJson/src/ArduinoJson/Object/MemberProxy.hpp:59 (discriminator 4)
      (inlined by) ArduinoJson::V701PB2::detail::VariantData* ArduinoJson::V701PB2::detail::VariantAttorney::getOrCreateData<ArduinoJson::V701PB2::detail::MemberProxy<ArduinoJson::V701PB2::JsonDocument&, char const*> const>(ArduinoJson::V701PB2::detail::MemberProxy<ArduinoJson::V701PB2::JsonDocument&, char const*> const&) at .pio/libdeps/pro-esp32-debug/ArduinoJson/src/ArduinoJson/Variant/VariantAttorney.hpp:32 (discriminator 4)
      (inlined by) ArduinoJson::V701PB2::detail::VariantRefBase<ArduinoJson::V701PB2::detail::MemberProxy<ArduinoJson::V701PB2::JsonDocument&, char const*> >::getOrCreateData() const at .pio/libdeps/pro-esp32-debug/ArduinoJson/src/ArduinoJson/Variant/VariantRefBase.hpp:268 (discriminator 4)
      (inlined by) ArduinoJson::V701PB2::detail::VariantRefBase<ArduinoJson::V701PB2::detail::MemberProxy<ArduinoJson::V701PB2::JsonDocument&, char const*> >::getOrCreateVariant() const at .pio/libdeps/pro-esp32-debug/ArduinoJson/src/ArduinoJson/Variant/VariantRefBaseImpl.hpp:100 (discriminator 4)
      (inlined by) bool ArduinoJson::V701PB2::detail::VariantRefBase<ArduinoJson::V701PB2::detail::MemberProxy<ArduinoJson::V701PB2::JsonDocument&, char const*> >::set<char const>(char const*) const at .pio/libdeps/pro-esp32-debug/ArduinoJson/src/ArduinoJson/Variant/VariantRefBaseImpl.hpp:144 (discriminator 4)
      (inlined by) ArduinoJson::V701PB2::detail::MemberProxy<ArduinoJson::V701PB2::JsonDocument&, char const*>& ArduinoJson::V701PB2::detail::MemberProxy<ArduinoJson::V701PB2::JsonDocument&, char const*>::operator=<char const>(char const*) at .pio/libdeps/pro-esp32-debug/ArduinoJson/src/ArduinoJson/Object/MemberProxy.hpp:39 (discriminator 4)
      (inlined by) ESPDash::generateLayoutJSON(AsyncWebSocketClient*, bool, Card*) at lib/ESPDASHPro/src/ESPDashPro.cpp:365 (discriminator 4)
  #6  0x4012880e:0x3ffea410 in ESPDash::ESPDash(AsyncWebServer*, char const*, bool)::{lambda(AsyncWebSocket*, AsyncWebSocketClient*, AwsEventType, void*, unsigned char*, unsigned int)#3}::operator()(AsyncWebSocket*, AsyncWebSocketClient*, AwsEventType, void*, unsigned char*, unsigned int) const at lib/ESPDASHPro/src/ESPDashPro.cpp:91
  #7  0x40128d31:0x3ffea490 in std::_Function_handler<void (AsyncWebSocket*, AsyncWebSocketClient*, AwsEventType, void*, unsigned char*, unsigned int), ESPDash::ESPDash(AsyncWebServer*, char const*, bool)::{lambda(AsyncWebSocket*, AsyncWebSocketClient*, AwsEventType, void*, unsigned char*, unsigned int)#3}>::_M_invoke(std::_Any_data const&, AsyncWebSocket*&&, AsyncWebSocketClient*&&, AwsEventType&&, void*&&, unsigned char*&&, unsigned int&&) at /Users/mat/.platformio/packages/toolchain-xtensa-esp32@8.4.0+2021r2-patch5/xtensa-esp32-elf/include/c++/8.4.0/bits/std_function.h:297
  #8  0x401c579d:0x3ffea4c0 in std::function<void (AsyncWebSocket*, AsyncWebSocketClient*, AwsEventType, void*, unsigned char*, unsigned int)>::operator()(AsyncWebSocket*, AsyncWebSocketClient*, AwsEventType, void*, unsigned char*, unsigned int) const at /Users/mat/.platformio/packages/toolchain-xtensa-esp32@8.4.0+2021r2-patch5/xtensa-esp32-elf/include/c++/8.4.0/bits/std_function.h:687
  #9  0x401c57c2:0x3ffea510 in AsyncWebSocket::_handleEvent(AsyncWebSocketClient*, AwsEventType, void*, unsigned char*, unsigned int) at .pio/libdeps/pro-esp32-debug/ESPAsyncWebServer-esphome/src/AsyncWebSocket.cpp:862
  #10 0x401c609d:0x3ffea540 in AsyncWebSocketClient::_onData(void*, unsigned int) at .pio/libdeps/pro-esp32-debug/ESPAsyncWebServer-esphome/src/AsyncWebSocket.cpp:682
  #11 0x401c6211:0x3ffea580 in std::_Function_handler<void (void*, AsyncClient*, void*, unsigned int), AsyncWebSocketClient::AsyncWebSocketClient(AsyncWebServerRequest*, AsyncWebSocket*)::{lambda(void*, AsyncClient*, void*, unsigned int)#7}>::_M_invoke(std::_Any_data const&, void*&&, AsyncClient*&&, std::_Any_data const&, unsigned int&&) at .pio/libdeps/pro-esp32-debug/ESPAsyncWebServer-esphome/src/AsyncWebSocket.cpp:481
      (inlined by) _M_invoke at /Users/mat/.platformio/packages/toolchain-xtensa-esp32@8.4.0+2021r2-patch5/xtensa-esp32-elf/include/c++/8.4.0/bits/std_function.h:297
  #12 0x401c2d17:0x3ffea5a0 in std::function<void (void*, AsyncClient*, void*, unsigned int)>::operator()(void*, AsyncClient*, void*, unsigned int) const at /Users/mat/.platformio/packages/toolchain-xtensa-esp32@8.4.0+2021r2-patch5/xtensa-esp32-elf/include/c++/8.4.0/bits/std_function.h:687
  #13 0x401c2d66:0x3ffea5d0 in AsyncClient::_recv(tcp_pcb*, pbuf*, signed char) at .pio/libdeps/pro-esp32-debug/AsyncTCP-esphome/src/AsyncTCP.cpp:961
  #14 0x401c2da9:0x3ffea5f0 in AsyncClient::_s_recv(void*, tcp_pcb*, pbuf*, signed char) at .pio/libdeps/pro-esp32-debug/AsyncTCP-esphome/src/AsyncTCP.cpp:1253
  #15 0x401c2f82:0x3ffea610 in _handle_async_event(lwip_event_packet_t*) at .pio/libdeps/pro-esp32-debug/AsyncTCP-esphome/src/AsyncTCP.cpp:164
  #16 0x401c2ff5:0x3ffea630 in _async_service_task(void*) at .pio/libdeps/pro-esp32-debug/AsyncTCP-esphome/src/AsyncTCP.cpp:199
```

bblanchon, 2024-01-12:
Hi @mathieucarbou,

This looks like a corrupted `JsonDocument`.
Are you sure the received reference is pointing to a valid object?

The stack trace shows a `generateLayoutJSON()` called from a lambda, called by a callback handler.
None of this is visible in the code you shared.
Could you reproduce this bug in an MCVE?

Best regards,
Benoit

mathieucarbou, 2024-01-12:
> generateLayoutJSON

Hello,

Yes that's what I am thinking: I discovered yesterday that the library I use is not thread safe and is calling this `generateLayoutJSON` method that is populating a json document both from the async_http task and the loop task. I am digging into the issue and close this one. thanks!

mathieucarbou, 2024-01-12:
@bblanchon : I have to re-open it... I have removed any potential concurrent access: now, the generateLayout method is only called once from the async_http task (websocket handler) and nowhere else.. Also, the json document reads data that is initialised once, stays as references in private fields in a static class, and this data is not modified concrurently.

While the json document is building, I have this error again, and constantly:

```
D         5142 YaSolR-I/O (1) HA-DISCO Publishing HA Discovery configuration for sensor output2_state...
D         5186 YaSolR-I/O (1) HA-DISCO Publishing HA Discovery configuration for sensor output2_routed_power...
D         5290 YaSolR-I/O (1)       HA Published in 2176 ms
W         5293 YaSolR-I/O (1)      I/O Abnormal loop duration: 2179 ms
D         5434 loopTask   (1)  DISPLAY Updated in 49 ms
[  6805][W][AsyncTCP.cpp:950] _poll(): rx timeout 4
D         7610 YaSolR-I/O (1)      I/O Published in 2262 ms
W         7613 YaSolR-I/O (1)      I/O Abnormal loop duration: 2265 ms
T         8130 async_tcp  (1) ESP-DASH client=1, measureJson=612

assert failed: ArduinoJson::V701PB2::detail::VariantSlot* ArduinoJson::V701PB2::detail::VariantPoolList::getSlot(ArduinoJson::V701PB2::detail::SlotId) const VariantPoolList.hpp:98 (poolIndex < count_

Backtrace: 0x400838e5:0x3ffecc70 0x4008dbcd:0x3ffecc90 0x40093719:0x3ffeccb0 0x400dd150:0x3ffecde0 0x40121d5b:0x3ffece80 0x401240ad:0x3ffecea0 0x40126636:0x3ffed5b0 0x40126b31:0x3ffed630 0x401c3bf1:0x3ffed660 0x401c3c16:0x3ffed6b0 0x401c4fd9:0x3ffed6e0 0x401c50b5:0x3ffed720 0x401c0a1f:0x3ffed740 0x401c0a6e:0x3ffed770 0x401c0ab1:0x3ffed790 0x401c0c8d:0x3ffed7b0 0x401c0d09:0x3ffed7d0

  #0  0x400838e5:0x3ffecc70 in panic_abort at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/panic.c:408
  #1  0x4008dbcd:0x3ffecc90 in esp_system_abort at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/esp_system.c:137
  #2  0x40093719:0x3ffeccb0 in __assert_func at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/newlib/assert.c:85
  #3  0x400dd150:0x3ffecde0 in ArduinoJson::V701PB2::detail::VariantPoolList::getSlot(unsigned short) const at .pio/libdeps/pro-esp32-debug/ArduinoJson/src/ArduinoJson/Memory/VariantPoolList.hpp:98
      (inlined by) ArduinoJson::V701PB2::detail::VariantPoolList::allocFromFreeList() at .pio/libdeps/pro-esp32-debug/ArduinoJson/src/ArduinoJson/Memory/VariantPoolImpl.hpp:71
      (inlined by) ArduinoJson::V701PB2::detail::VariantPoolList::allocSlot(ArduinoJson::V701PB2::Allocator*) at .pio/libdeps/pro-esp32-debug/ArduinoJson/src/ArduinoJson/Memory/VariantPoolList.hpp:73
      (inlined by) ArduinoJson::V701PB2::detail::ResourceManager::allocSlot() at .pio/libdeps/pro-esp32-debug/ArduinoJson/src/ArduinoJson/Memory/ResourceManager.hpp:53
      (inlined by) ArduinoJson::V701PB2::detail::CollectionData::addSlot(ArduinoJson::V701PB2::detail::ResourceManager*) at .pio/libdeps/pro-esp32-debug/ArduinoJson/src/ArduinoJson/Collection/CollectionImpl.hpp:52
      (inlined by) ArduinoJson::V701PB2::detail::VariantData* ArduinoJson::V701PB2::detail::ObjectData::addMember<ArduinoJson::V701PB2::detail::StaticStringAdapter>(ArduinoJson::V701PB2::detail::StaticStringAdapter, ArduinoJson::V701PB2::detail::ResourceManager*) at .pio/libdeps/pro-esp32-debug/ArduinoJson/src/ArduinoJson/Object/ObjectData.hpp:27
      (inlined by) ArduinoJson::V701PB2::detail::VariantData* ArduinoJson::V701PB2::detail::ObjectData::getOrAddMember<ArduinoJson::V701PB2::detail::StaticStringAdapter>(ArduinoJson::V701PB2::detail::StaticStringAdapter, ArduinoJson::V701PB2::detail::ResourceManager*) at .pio/libdeps/pro-esp32-debug/ArduinoJson/src/ArduinoJson/Object/ObjectImpl.hpp:24
  #4  0x40121d5b:0x3ffece80 in ArduinoJson::V701PB2::detail::VariantData* ArduinoJson::V701PB2::detail::VariantData::getOrAddMember<ArduinoJson::V701PB2::detail::StaticStringAdapter>(ArduinoJson::V701PB2::detail::StaticStringAdapter, ArduinoJson::V701PB2::detail::ResourceManager*) at .pio/libdeps/pro-esp32-debug/ArduinoJson/src/ArduinoJson/Variant/VariantData.hpp:226
  #5  0x401240ad:0x3ffecea0 in ArduinoJson::V701PB2::detail::MemberProxy<ArduinoJson::V701PB2::JsonObject, char const*>::getOrCreateData() const at .pio/libdeps/pro-esp32-debug/ArduinoJson/src/ArduinoJson/Object/MemberProxy.hpp:59
      (inlined by) ArduinoJson::V701PB2::detail::VariantData* ArduinoJson::V701PB2::detail::VariantAttorney::getOrCreateData<ArduinoJson::V701PB2::detail::MemberProxy<ArduinoJson::V701PB2::JsonObject, char const*> const>(ArduinoJson::V701PB2::detail::MemberProxy<ArduinoJson::V701PB2::JsonObject, char const*> const&) at .pio/libdeps/pro-esp32-debug/ArduinoJson/src/ArduinoJson/Variant/VariantAttorney.hpp:32
      (inlined by) ArduinoJson::V701PB2::detail::VariantRefBase<ArduinoJson::V701PB2::detail::MemberProxy<ArduinoJson::V701PB2::JsonObject, char const*> >::getOrCreateData() const at .pio/libdeps/pro-esp32-debug/ArduinoJson/src/ArduinoJson/Variant/VariantRefBase.hpp:268
      (inlined by) ArduinoJson::V701PB2::detail::VariantRefBase<ArduinoJson::V701PB2::detail::MemberProxy<ArduinoJson::V701PB2::JsonObject, char const*> >::getOrCreateVariant() const at .pio/libdeps/pro-esp32-debug/ArduinoJson/src/ArduinoJson/Variant/VariantRefBaseImpl.hpp:100
[... cut ...]
  serializeJson(doc, buffer->data(), len);

#ifdef DASH_DEBUG
  Serial.printf("T %12u %-10.10s (%u) ESP-DASH client=%u, measureJson=%u\r\n", static_cast<uint32_t>(esp_timer_get_time() >> 10), pcTaskGetName(NULL), xPortGetCoreID(), (client == nullptr ? -1 : client->id()), len);
#endif

  if (client != nullptr) {
    client->text(buffer);
  } else {
    _ws->textAll(buffer);
  }

  doc.clear();
}

bool ESPDash::overflowed(JsonDocument& json) {
  return json.overflowed() || measureJson(json.as<JsonObject>()) > DASH_JSON_SIZE;
}
```

mathieucarbou, 2024-01-12:
Ok so there is definitely something wrong with reusing a Json doc after a clear().

I've implemented a new version which is creating a new JsonDocument each time, and it works.

bblanchon, 2024-01-14:
Could you provide an [MCVE](https://stackoverflow.com/help/mcve)?

mathieucarbou, 2024-01-14:
We have a big internet outage here but as soon as I can, yes.

mathieucarbou, 2024-01-15:
Here is a simple `.ino` file to reproduce:

```c++
#include <ArduinoJson.h>

void fill(JsonObject &obj) {
  obj["foo-bar-baz"] = "foo-bar-baz-foo-bar-baz";
}

void send(JsonObject &obj) {
  serializeJson(obj, Serial);
  Serial.println("");
  obj.clear();
}

void setup() {
  Serial.begin(115200);

  // put your setup code here, to run once:
  JsonDocument doc;

  {
    JsonObject obj = doc.to<JsonObject>();
    obj["command"] = "update:components";
    obj["tabs"][0]["id"] = "0";
    send(obj);
  }

  {
    JsonObject obj = doc.to<JsonObject>();
    obj["command"] = "update:components";
    JsonObject card = doc["cards"].add<JsonObject>();
    fill(card);
    send(obj);
  }
}

void loop() { }
```

Please noe that if I comment out the `obj.clear()` line, it works:

```
void send(JsonObject &obj) {
  serializeJson(obj, Serial);
  Serial.println("");
  //obj.clear();
}
```

closed 2024-01-16 by bblanchon

bblanchon, 2024-01-16:
Thank you very much for this code. :heart:
It's now fixed in the main branch, and I'll publish a new revision soon.

> What is the recommandation for such usage: use only one JsonDocument and scope the construction of inner batches to send through websocket, or constructing a completely new JsonDocument for each batch, scoped by brackets ?

I recommend short lifetimes to release memory as soon as possible and reduce the cognitive load.
In your case, this would mean creating a new `JsonDocument` for each batch.

> My assumption was that destroying and creating a new JsonDocument each time would be slower, so that is why I've tried to reuse it with clear() and .to() calls.

Calling `JsonDocument::clear()` and destructing a `JsonDocument` are roughly the same; I don't expect any measurable difference.
Choose the one that makes the code easier to read and reason about.

mathieucarbou, 2024-01-16:
Thanks for the advices, and glad I was able to help :-)

bblanchon, 2024-01-19:
The fix was published in [ArduinoJson 7.0.2](https://github.com/bblanchon/ArduinoJson/releases/tag/v7.0.2).

commit 1b143d38 (2024-01-16): Fix assertion `poolIndex < count_` after `JsonDocument::clear()`

Fixes #2034