bblanchon/ArduinoJson #2038: How to get the size of the document in v7?

zekageri, 2024-01-19:
Before V7 i knew the max size of the json document because i had to specify it.
Now, with V7 i don't have a size variable so i can't determine it.

Here is an example before V7:

```cpp
boolean Sram::writeJson(JsonObject doc, size_t size){
    char buff[size];
    if( !serializeJson(doc, buff, size) ){ return false; }
    return SRAM.writeBlock(START_ADDRESS, size, (void*)buff);
}

DynamicJsonDocument doc(1024);
doc["test"] = "test";
writeJson(doc,1024);
```

Here is an example after V7:

```cpp
boolean Sram::writeJson(JsonObject doc, size_t size){
    char buff[size];
    if( !serializeJson(doc, buff, size) ){ return false; }
    return SRAM.writeBlock(START_ADDRESS, size, (void*)buff); // Must specify the size of the buffer.
}

JsonDocument doc;
doc["test"] = "test";
size_t size = 5; // Don't know the size??
writeJson(doc, size);
```

Is there a method to get the size of the doc?

bblanchon, 2024-01-21:
Hi @zekageri,

You confused the capacity of the memory pool with the length of the serialized JSON document.
In ArduinoJson 6 and 7, you can measure the length of the serialized JSON document by calling [`measureJson()`](https://arduinojson.org/v7/api/json/measurejson/).

Best regards,
Benoit

zekageri, 2024-01-21:
Thank you very much. That might be the problem.

closed 2024-01-21 by zekageri