bblanchon/ArduinoJson #2040: V7 : Compilation fail on pure example JsonParserExample.ino and Arduino IDE 2.2.1

Linky13860, 2024-01-20:
**Describe the bug**  
Compilation fail : Arduino IDE / JsonParserExample.ino / Wemos ESP8266

**Troubleshooter report**  
Here is the report generated by the [ArduinoJson Troubleshooter](https://arduinojson.org/v7/troubleshooter/):  
[Paste the report here]

**Environment**  
Here is the environment that I used:
* Microcontroller: ESP8266
* Core/runtime: LOLIN D1 mini Pro
* Arduino IDE 2.2.1

**Compiler output**    
In file included from d:\Documents\ArduinoData\libraries\ArduinoJson\src/ArduinoJson.hpp:33:0,
                 from d:\Documents\ArduinoData\libraries\ArduinoJson\src/ArduinoJson.h:9,
                 from D:\Documents\ArduinoData\libraries\ArduinoJson\examples\JsonParserExample\JsonParserExample.ino:9:
d:\Documents\ArduinoData\libraries\ArduinoJson\src/ArduinoJson/Document/JsonDocument.hpp: In constructor 'ArduinoJson::V702PB2::JsonDocument::JsonDocument(ArduinoJson::V702PB2::JsonDocument&&)':
d:\Documents\ArduinoData\libraries\ArduinoJson\src/ArduinoJson/Document/JsonDocument.hpp:33:51: error: no matching function for call to 'ArduinoJson::V702PB2::JsonDocument::JsonDocument()'
   JsonDocument(JsonDocument&& src) : JsonDocument() {
                                                   ^
d:\Documents\ArduinoData\libraries\ArduinoJson\src/ArduinoJson/Document/JsonDocument.hpp:33:51: note: candidates are:
d:\Documents\ArduinoData\libraries\ArduinoJson\src/ArduinoJson/Document/JsonDocument.hpp:39:3: note: template<class T> ArduinoJson::V702PB2::JsonDocument::JsonDocument(const T&, ArduinoJson::V702PB2::Allocator*, typename ArduinoJson::V702PB2::detail::enable_if<(((((ArduinoJson::V702PB2::detail::is_same<T, ArduinoJson::V702PB2::JsonVariant>::value || ArduinoJson::V702PB2::detail::is_same<T, ArduinoJson::V702PB2::JsonVariantConst>::value) || ArduinoJson::V702PB2::detail::is_same<T, ArduinoJson::V702PB2::JsonArray>::value) || ArduinoJson::V702PB2::detail::is_same<T, ArduinoJson::V702PB2::JsonArrayConst>::value) || ArduinoJson::V702PB2::detail::is_same<T, ArduinoJson::V702PB2::JsonObject>::value) || ArduinoJson::V702PB2::detail::is_same<T, ArduinoJson::V702PB2::JsonObjectConst>::value)>::type*)
   JsonDocument(const T& src,
   ^
d:\Documents\ArduinoData\libraries\ArduinoJson\src/ArduinoJson/Document/JsonDocument.hpp:39:3: note:   template argument deduction/substitution failed:
d:\Documents\ArduinoData\libraries\ArduinoJson\src/ArduinoJson/Document/JsonDocument.hpp:33:51: note:   candidate expects 3 arguments, 0 provided
   JsonDocument(JsonDocument&& src) : JsonDocument() {
                                                   ^
d:\Documents\ArduinoData\libraries\ArduinoJson\src/ArduinoJson/Document/JsonDocument.hpp:33:3: note: ArduinoJson::V702PB2::JsonDocument::JsonDocument(ArduinoJson::V702PB2::JsonDocument&&)
   JsonDocument(JsonDocument&& src) : JsonDocument() {
   ^
d:\Documents\ArduinoData\libraries\ArduinoJson\src/ArduinoJson/Document/JsonDocument.hpp:33:3: note:   candidate expects 1 argument, 0 provided
d:\Documents\ArduinoData\libraries\ArduinoJson\src/ArduinoJson/Document/JsonDocument.hpp:28:3: note: ArduinoJson::V702PB2::JsonDocument::JsonDocument(const ArduinoJson::V702PB2::JsonDocument&)
   JsonDocument(const JsonDocument& src) : JsonDocument(src.allocator()) {
   ^
d:\Documents\ArduinoData\libraries\ArduinoJson\src/ArduinoJson/Document/JsonDocument.hpp:28:3: note:   candidate expects 1 argument, 0 provided

exit status 1

Compilation error: exit status 1

**Program output**  
If relevant, include the repro program output.

bblanchon, 2024-01-21:
Hi @Linky13860,

> error: no matching function for call to 'ArduinoJson::V702PB2::JsonDocument::JsonDocument()'

The last time I saw a similar error, it was with GCC 4.9.
Are you sure that your Arduino Core for ESP8266 is up to date?

Best regards,
Benoit

bblanchon, 2024-02-01:
Closing in favor of #2045.