bblanchon/ArduinoJson #2041: conflicting Zero-copy mode documentation

EvEggelen, 2024-01-20:
On https://arduinojson.org/news/2024/01/03/arduinojson-7/ it is described that deserializeJson() will never use zero-copy mode.

However according to https://arduinojson.org/v7/api/json/deserializejson/ the zero copy mode is still there. I assume this is a copy-past mistake from v6, but it confused me.

bblanchon, 2024-01-21:
Hi @EvEggelen,

Thank you very much for reporting this problem ❤️ 

This was indeed an awful copy and paste from the old documentation, and I'm very sorry for that.
It should be fixed now.

Best regards,
Benoit

closed 2024-01-21 by bblanchon

EvEggelen, 2024-01-21:
Not a problem. Thanks a lot for this great library ! It must be a lot of work maintaining one version with this quality, let a lone supporting multiple versions...

Would you even consider bringing back the zero-copy with an explicit interface ? Meaning, that users who did not fully read the manual use the current interface ( no zero copy) , but the ones who do know the details can still use it when explicitly enabled / selected.

E.g. deserializeJson(doc, input, true); ( e.g. true=enable zero-copy and use input by reference/alter the input). Or even a complete new name. : deserializeJsonByReference ( doc,input ).  

Anyway, just my thoughts...

bblanchon, 2024-01-22:
I don't plan on bringing back the zero-copy mode, as it was misused 90% of the time and was very difficult to implement.

EvEggelen, 2024-01-22:
I can imagine the complexity. I am I really big fan of the efficiency when parsing JSONs into C/C++ structs. This is most of my use-cases..... Hoping a explicit interface could drastically reduced the misuse. If you every change your mind, let me know, I am happy to test the pre-releases.