bblanchon/ArduinoJson #2043: Adding a char as a member of a JsonDocument produces a convertToJson error

smaslincosi, 2024-01-25:
**Description**
I am in the process of migrating existing code to ArduinoJson v7, and I have had issues with compilation that I believe stem from changes to the library. In previous versions of the library, I was able to add a char to a StaticJsonDocument without issue. In v7 this produces an error with the convertToJson function.
My intended target is Teensy 3.2, but I have been able to reproduce this issue with the provided code on other platforms such as Arduino Uno.
My provided reproduction code works with version 6.19.4, but not with 6.20.0 or thereafter.

**Troubleshooter's report**
1. The program uses ArduinoJson 7
2. The issue happens at compile time
3. Error says "no matching function for call to ..."
4. Error says "no matching function for call to `convertToJson(...)`"
5. Converting the value doesn't fix the issue

**Environment**
* Microcontroller: Teensy 3.2
* Core/Framework: Teensyduino
* IDE: Arduino IDE 2.2.1

**Reproduction code**
```c++
#include <ArduinoJson.h>

void setup() {

  JsonDocument doc;

  //This will compile:
  //const char* cs = "A";
  //doc["val"] = cs;

  //This will not compile - error: no matching function for call to 'convertToJson(const char&, ArduinoJson::V702L1::JsonVariant&)'
  char c = 'A';
  doc["val"] = c;
}

void loop() {
  // put your main code here, to run repeatedly:

}
```

bblanchon, 2024-01-26:
Hi @smaslincosi,

[ArduinoJson 6.18](https://arduinojson.org/news/2021/05/04/version-6-18-0/#support-for-naked-char-removed) deprecated the support for "naked" `char`, then [ArduinoJson 6.20](https://arduinojson.org/news/2022/12/26/arduinojson-6-20-0/#removed-support-for-char-and-char) removed it entirely.

JSON supports numbers and strings but has no "character" type.
If you want to store an integer, you can use `signed char`, `unsigned char`, or another integer type.
If you want to store a string, you can use `const char*`, `String`, or any supported string type.

If you cannot live without `char` support, you must create a custom converter.
For example, the following converter stores `char`s as one-character strings.

```c++
namespace ArduinoJson {
template <>
struct Converter<char> {
  static void toJson(char c, JsonVariant var) {
    char buf[] = {c, 0};
    var.set(buf);
  }

  static char fromJson(JsonVariantConst src) {
    auto p = src.as<const char*>();
    return p ? p[0] : 0;
  }
};
}
```

Best regards,
Benoit

smaslincosi, 2024-01-26:
Thank you for the incredibly helpful information!

closed 2024-01-26 by smaslincosi