bblanchon/ArduinoJson #2045: error: no matching function for call to 'ArduinoJson::V702PB2::JsonDocument::JsonDocument()' when building example code

evries, 2024-01-31:
**Description**
When I build the example code I get the following error: 
error: no matching function for call to 'ArduinoJson::V702PB2::JsonDocument::JsonDocument()'

**Troubleshooter's report**
1. The program uses ArduinoJson 7
2. The issue happens at compile time
3. Error says "no matching function for call to ..."
4. The error is not in the list

**Environment**
* Microcontroller: ESP8266
* Core/Framework: framework-arduinoespressif8266 @ 3.20704.0 (2.7.4)
* IDE: VS code

**Reproduction code**
```c++
#include <ArduinoJson.h>

void setup() {
  // Initialize Serial port
  Serial.begin(9600);
  while (!Serial)
    continue;

  // Allocate the JSON document
  JsonDocument doc;

  // Add values in the document
  doc["sensor"] = "gps";
  doc["time"] = 1351824120;

  // Add an array
  JsonArray data = doc["data"].to<JsonArray>();
  data.add(48.756080);
  data.add(2.302038);

  // Generate the minified JSON and send it to the Serial port
  serializeJson(doc, Serial);
  // The above line prints:
  // {"sensor":"gps","time":1351824120,"data":[48.756080,2.302038]}

  // Start a new line
  Serial.println();

  // Generate the prettified JSON and send it to the Serial port
  serializeJsonPretty(doc, Serial);
  // The above line prints:
  // {
  //   "sensor": "gps",
  //   "time": 1351824120,
  //   "data": [
  //     48.756080,
  //     2.302038
  //   ]
  // }
}

void loop() {
  // not used in this example
}
```

**Remarks**
Complete compiler output

Processing nodemcuv2 (platform: espressif8266; board: nodemcuv2; framework: arduino)
-----------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/nodemcuv2.html
PLATFORM: Espressif 8266 (2.6.2) > NodeMCU 1.0 (ESP-12E Module)
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
PACKAGES:
 - framework-arduinoespressif8266 @ 3.20704.0 (2.7.4)
 - tool-esptool @ 1.413.0 (4.13)
 - tool-esptoolpy @ 1.20800.0 (2.8.0)
 - toolchain-xtensa @ 2.40802.200502 (4.8.2)
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 30 compatible libraries
Scanning dependencies...
Dependency Graph
|-- ArduinoJson @ 7.0.2
Building in release mode
Compiling .pio\build\nodemcuv2\src\main.cpp.o
Compiling .pio\build\nodemcuv2\FrameworkArduino\Esp.cpp.o
Compiling .pio\build\nodemcuv2\FrameworkArduino\FS.cpp.o
Compiling .pio\build\nodemcuv2\FrameworkArduino\FSnoop.cpp.o
In file included from .pio\libdeps\nodemcuv2\ArduinoJson\src/ArduinoJson.hpp:33:0,
                 from .pio\libdeps\nodemcuv2\ArduinoJson\src/ArduinoJson.h:9,
                 from src\main.cpp:10:
.pio\libdeps\nodemcuv2\ArduinoJson\src/ArduinoJson/Document/JsonDocument.hpp: In constructor 'ArduinoJson::V702PB2::JsonDocument::JsonDocument(ArduinoJson::V702PB2::JsonDocument&&)':
.pio\libdeps\nodemcuv2\ArduinoJson\src/ArduinoJson/Document/JsonDocument.hpp:33:51: error: no matching function for call to 'ArduinoJson::V702PB2::JsonDocument::JsonDocument()'
   JsonDocument(JsonDocument&& src) : JsonDocument() {
                                                   ^
.pio\libdeps\nodemcuv2\ArduinoJson\src/ArduinoJson/Document/JsonDocument.hpp:33:51: note: candidates are:
Compiling .pio\build\nodemcuv2\FrameworkArduino\FunctionalInterrupt.cpp.o
.pio\libdeps\nodemcuv2\ArduinoJson\src/ArduinoJson/Document/JsonDocument.hpp:39:3: note: template<class T> ArduinoJson::V702PB2::JsonDocument::JsonDocument(const T&, ArduinoJson::V702PB2::Allocator*, typename ArduinoJson::V702PB2::detail::enable_if<(((((ArduinoJson::V702PB2::detail::is_same<T, ArduinoJson::V702PB2::JsonVariant>::value || ArduinoJson::V702PB2::detail::is_same<T, ArduinoJson::V702PB2::JsonVariantConst>::value) || ArduinoJson::V702PB2::detail::is_same<T, ArduinoJson::V702PB2::JsonArray>::value) || ArduinoJson::V702PB2::detail::is_same<T, ArduinoJson::V702PB2::JsonArrayConst>::value) || ArduinoJson::V702PB2::detail::is_same<T, ArduinoJson::V702PB2::JsonObject>::value) || ArduinoJson::V702PB2::detail::is_same<T, ArduinoJson::V702PB2::JsonObjectConst>::value)>::type*)
   JsonDocument(const T& src,
   ^
.pio\libdeps\nodemcuv2\ArduinoJson\src/ArduinoJson/Document/JsonDocument.hpp:39:3: note:   template argument deduction/substitution failed:
.pio\libdeps\nodemcuv2\ArduinoJson\src/ArduinoJson/Document/JsonDocument.hpp:33:51: note:   candidate expects 3 arguments, 0 provided
   JsonDocument(JsonDocument&& src) : JsonDocument() {
                                                   ^
.pio\libdeps\nodemcuv2\ArduinoJson\src/ArduinoJson/Document/JsonDocument.hpp:33:3: note: ArduinoJson::V702PB2::JsonDocument::JsonDocument(ArduinoJson::V702PB2::JsonDocument&&)
   JsonDocument(JsonDocument&& src) : JsonDocument() {
   ^
.pio\libdeps\nodemcuv2\ArduinoJson\src/ArduinoJson/Document/JsonDocument.hpp:33:3: note:   candidate expects 1 argument, 0 provided
.pio\libdeps\nodemcuv2\ArduinoJson\src/ArduinoJson/Document/JsonDocument.hpp:28:3: note: ArduinoJson::V702PB2::JsonDocument::JsonDocument(const ArduinoJson::V702PB2::JsonDocument&)
   JsonDocument(const JsonDocument& src) : JsonDocument(src.allocator()) {
   ^
.pio\libdeps\nodemcuv2\ArduinoJson\src/ArduinoJson/Document/JsonDocument.hpp:28:3: note:   candidate expects 1 argument, 0 provided

bblanchon, 2024-02-01:
Hi @evries,

Thank you very much for reporting this issue.

Indeed, the Arduino Core for ESP8266 version 2.7.4 (released in 2020) comes with GCC 4.8.2, an old buggy version of GCC released in 2013, more than ten years ago!

I will try to work around the issues of GCC 4.8 so that ArduinoJson 6.20+ and 7+ build, but I cannot make any promises.

You should upgrade the PlatformIO espressif8266 platform to version 3+.
It will install a newer version of the Arduino Core for ESP8266, bringing a recent version of GCC (currently 10.3.0).

Best regards,
Benoit

closed 2024-02-01 by bblanchon

peault, 2024-02-02:
même erreur avec arduino due  ide 2.21 , carte 1.6.12

bblanchon, 2024-02-05:
The fix is available in [ArduinoJson 7.0.3](https://github.com/bblanchon/ArduinoJson/releases/tag/v7.0.3).

peault, 2024-02-05:
ok avec 7.0.3 merci

commit 72642e30 (2024-02-01): Fix compatibility with GCC 4.8

Closes #2045